libretro/Lakka-LibreELEC #243: Blue turns to red

masid101, 2017-10-12:
- Which version of Lakka are you using? 2.1 RC5
- What system hardware are you using? RPi3
- What did you do? Set Menu Shader Pipeline to off
- What did you expect to happen? The ribbon to disappear
- What happened instead? 

The ribbon disappeared but also the background turned to red while the Color Theme still points to Electric Blue. I don't know if this is a feature or a bug, but I like the blue background.

Kivutar, 2017-10-13:
Disabling the ribbon fallbacks to displaying an image wallpaper. It was the easiest way to please both the users who wanted a ribbon, and the users who wanted wallpapers.

masid101, 2017-10-13:
Ah, okay! But why RED? ;-)

closed 2017-10-13 by masid101

Kivutar, 2017-10-13:
It was the default wallpaper some time ago. The one you can see on our website frontpage :)