libretro/Lakka-LibreELEC #244: Rockchip.TinkerBoard.arm build no Audio over 3.5 mm headphone jack

ShilohS, 2017-10-12:
Rockchip.TinkerBoard.arm build no Audio over 3.5 mm headphone jack

First of all I understand its a alpha and everything inst supposed to work yet, just making sure this is recorded.

Version: Lakka-Rockchip.TinkerBoard.arm-2.1-rc5.img
Asus Tinkerboard Rev 1.2

When you boot into the image you get no audio over  3.5 mm headphone jack, I haven't been able to test if the audio is coming out the HDMI port, don't have a way of testing that right now.

Seems to be a common issue with the tinker board, The way it innumerate's the audio is as a usb device and I think it is not the first endpoint i think its the 3rd. 

Thank you!

ShilohS

gouchi, 2017-10-12:
Did you try to [switching audio device](http://www.lakka.tv/doc/Audio-settings/#switching-audio-device-from-the-gui) ?

You should have for audio device 
```
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=HDMI
    HDMI, 
    Default Audio Device
sysdefault:CARD=HDMI
    HDMI, 
    Default Audio Device
hdmi:CARD=HDMI,DEV=0
    HDMI, 
    HDMI Audio Output
default:CARD=ALC4040
    ALC4040, USB Audio #2
    Default Audio Device
sysdefault:CARD=ALC4040
    ALC4040, USB Audio #2
    Default Audio Device
front:CARD=ALC4040,DEV=0
    ALC4040, USB Audio #2
    Front speakers
```

ShilohS, 2017-10-12:
wow, I'm sorry, I'm new to Lakka, I should of just posted in the forms or something. I Tried to change audio device but I kept selecting it and not knowing what to type into the field. I didn't realize you could push left or right to change audio.

It is working great now, can you let me know how to delete this bug???

closed 2017-10-12 by ShilohS

gouchi, 2017-10-12:
No problem, glad it works.