libretro/Lakka-LibreELEC #245: Orange Pi PC non bootable

Tom-Neverwinter, 2017-10-13:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
Lakka-H3.opipc.arm-2.0, Lakka-H3.opipc.arm-2.1-rc5
- What system hardware are you using?
orangepipc
- What did you do?
instll image using etcher.... wait.... wait..... nothing
- What did you expect to happen?
os to boot properly
- What happened instead?
black screen, nothing, no activity leds

Kivutar, 2017-10-15:
I downloaded the image, and flashed it. It boots fine with a green led, and after the first automatic reboot RetroArch shows up.

So I think your image got corrupted during download.

Tom-Neverwinter, 2017-10-15:
off to work, I will check the md5 and sha1 when I get home tonight or early tomorrow. archive integrity appeared to be good. May also be a issue with a hdmi to vga adapter?

Kivutar, 2017-10-17:
I'm testing on a Orange Pi PC yes.

With the power adapter of an iPad. Using this image http://le.builds.lakka.tv/H3.opipc.arm/Lakka-H3.opipc.arm-2.1-rc5.img.gz and Etcher.

Kivutar, 2017-10-23:
We're working on a completely new image for the Orange Pi. Do you want to try it?

Tom-Neverwinter, 2017-10-23:
I would love to try it

Ntemis, 2017-10-24:
Here you go:
https://androidfilehost.com/?fid=673791459329055943
Enjoy

Tom-Neverwinter, 2017-10-24:
Confirmed new version fixes the issue. I am assuming it was a uboot resolution issue?

Kivutar, 2017-10-24:
It could be. This image is using a completely different u-boot and kernel (mainline)

Message envoyé de mon iPhone ce qui explique son caractère rudimentaire.

closed 2017-10-24 by Kivutar