libretro/Lakka-LibreELEC #249: Unable to access samba shares on latest RC5

arcooke, 2017-10-15:
- Version 2.1 RC5 (Oct 11)
- Raspberry Pi 3
- Installed Lakka exactly as instructed on the step-by-step wizard
- Client system: Solus linux

Please let me know if I need to include anything else, I always forget something :)

I am having a lot of issues getting Lakka up and running on my Pi.  Originally, I installed the public stable version from the site yesterday.  All of my samba shares showed up (roms/assets/cheats/etc). 
 The ROM share only said there was 25MB of storage available and I couldn't copy files, so I installed latest nightly instead.

I can still connect to smb://192.168.1.171, but no shares appear anymore.

[image]

I can't find any errors and the samba config looks fine.. it has not been modified from defaults.

**log.smbd**
```
[2017/10/08 10:40:42.405843,  0] ../source3/passdb/pdb_smbpasswd.c:249(startsmbfilepwent)
  startsmbfilepwent_internal: file /run/samba/smbpasswd did not exist. File successfully created.
[2017/10/08 10:40:42.550031,  0] ../lib/util/become_daemon.c:124(daemon_ready)
  STATUS=daemon 'smbd' finished starting up and ready to serve connections
[2017/10/15 21:10:27.961660,  0] ../lib/util/become_daemon.c:124(daemon_ready)
  STATUS=daemon 'smbd' finished starting up and ready to serve connections
```

**lob.nmbd**
```
[2017/10/08 10:40:40.971309,  0] ../lib/util/become_daemon.c:135(daemon_status)
  STATUS=daemon 'nmbd' : No local IPv4 non-loopback interfaces available, waiting for interface ...NOTE: NetBIOS name resolution$
[2017/10/15 21:00:46.970292,  0] ../lib/util/become_daemon.c:124(daemon_ready)
  STATUS=daemon 'nmbd' finished starting up and ready to serve connections
```

**systemctl status smbd**
```
● smbd.service - Samba SMB Daemon
   Loaded: loaded (/usr/lib/systemd/system/smbd.service; disabled; vendor preset: disabled)
   Active: active (running) since Sun 2017-10-15 21:10:27 UTC; 13min ago
  Process: 585 ExecStart=/usr/sbin/smbd (code=exited, status=0/SUCCESS)
  Process: 574 ExecStartPre=/usr/lib/samba/smbd-config (code=exited, status=0/SUCCESS)
 Main PID: 588 (smbd)
   CGroup: /system.slice/smbd.service
           ├─588 /usr/sbin/smbd
           ├─589 /usr/sbin/smbd
           ├─590 /usr/sbin/smbd
           └─592 /usr/sbin/smbd

Oct 15 21:10:27 Lakka systemd[1]: Starting Samba SMB Daemon...
Oct 15 21:10:27 Lakka systemd[1]: Started Samba SMB Daemon.
Oct 15 21:10:27 Lakka smbd[588]: [2017/10/15 21:10:27.961660,  0] ../lib/util/become_daemon.c:124(daemon_ready)
Oct 15 21:10:27 Lakka smbd[588]:   STATUS=daemon 'smbd' finished starting up and ready to serve connections
```

**Folders _DO_ exist in /storage/**

```
Lakka:/ # cd storage/
Lakka:~ # ls -l
total 26
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 assets
drwxr-xr-x    2 root     root          1024 Oct  8 10:40 backup
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 cores
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 database
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 joypads
drwx------    2 root     root         12288 Oct 11 05:07 lost+found
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 playlists
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 remappings
drwxrwxrwx    3 root     root          1024 Oct  8 10:40 roms
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 savefiles
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 savestates
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 screenshots
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 shaders
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 system
drwxrwxrwx    2 root     root          1024 Oct  8 10:40 thumbnails
```

```
Lakka (community): 2.1-rc5 (RPi2.arm)
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.6.7 (Git 28397b3)
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Oct 10 2017
[INFO] Version: 1.6.7
[INFO] Git: 28397b3
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 4 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Video]: Graphics driver did not initialize an input driver. Attempting to pick a suitable driver.
[WARN] [udev]: Couldn't open any keyboard, mouse or touchpad. Are permissions set correctly for /dev/input/event*?
[INFO] [udev]: Plugged pad: Logitech Gamepad F710 (1133:49695) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event0) supports 16 force feedback effects.
[INFO] [Autoconf]: 186 profiles found.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] ALSA: Using signed 16-bit format.
[INFO] [Autoconf]: selected configuration: /tmp/joypads/udev/Logitech_Gamepad_F710.cfg
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Menu]: Found menu display driver: "menu_display_gl".
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
```

AnicetusCer, 2018-02-06:
This is still an issue (lakka 2.1) tried install on 32GB Samsung SD HC card, formatted with etcher.
The initial format created the 512 MB lakka partition and a 32Mb second partition (assume boot) 
The final partition never gets created on the initial boot, the system fails to boot as it can not find the storage area.

ToKe79, 2018-02-06:
@AnicetusCer on which system did you try? If on generic, try to switch between Legacy / EFI boot in your BIOS settings. If on Amlogic (e.g. S905), make sure you are using correct dtb.img file for your box. The 32 MB partition is the storage partitions - the image file contains only 32 MB of this partition, this then gets resized during the initial boot to the remaining capacity of the SD card / USB thumb drive.

AnicetusCer, 2018-02-06:
Thanks for the comment, the system I am installing on is a "pi zero w" so the boot loader sits in within the system on chip's ROM and points the system to the SD card for its files, but boot loading is not the problem.
Thank you for pointing out the 32mb partitions use, nice solution to utilise all a systems space. 
So it seems the issue is with the resize, on first boot lakka should resize and mount the "storage" file system, the resize job runs, but does not resize the 32mb partition when I check the SD on my pc, the next message I see on reboot is that lakka failed to boot /dev/mccblk***** , can't remember the whole message, will run again tomorrow to check, if I remember correctly it was the storage FS.

closed 2020-08-18 by Ntemis