libretro/Lakka-LibreELEC #267: Rom scanner is acting weird

Arnochocobo, 2017-11-01:
Just checked on the latest github version on XU4.

Let's say I have 1500 roms to scan. I point to the directory, and then scan in progress values will look like this:

1/1500
2/1501
3/1502

After some time the right number will eventually stop and the left number might catch up with it.
Please note that sometimes the right number won't even match the actual number of roms in the specified folder.

Arnochocobo, 2017-11-01:
More details: Here is what happened with a folder containing 1721 roms.

1/1722
2/1723
....
1722/3442 (once reaching the exact double value of the number of roms, the right number will stop to increase) and then only the left number will increase:
1723/3442
up to 

3442/3442

Sobient, 2017-11-02:
You're probably not using only No-intro ROMs? There are issues as Lakka uses this DB when doing checksums when scanning.

Kivutar, 2017-11-02:
I can reproduce this issue, even with my clean roms.

ToKe79, 2018-01-09:
This is not Lakka specific, I think it is RetroArch specific - when scanning ZIP files - number of files in every ZIP file is added to the total (right) digit after the ZIP is scanned, and then when the file(s) from the ZIP file are scanned, they are added to the progress (left) digit.

ToKe79, 2018-01-31:
I think this should be considered as feature / design change.
https://github.com/libretro/RetroArch/issues/5634#issuecomment-341261880
So content of archive is always added to the total number of files to be scanned and in case the CRC of the archive does not match a ROM the content of archive is scanned.

closed 2018-01-31 by ToKe79