libretro/Lakka-LibreELEC #268: Proper error messages to identify issues

Arnochocobo, 2017-11-01:
Checked on latest github version on XU4.

Been a while I messed around with my XU4, and decided to update to the latest LibreELEC glory.

Some of the games that used to work fine with FBA don't work anymore (street fighter 3 third strike)
Right now, I just have a very generic error message.

It would be great to know what really happened and give me hints how to fix it. "Was expecting rom.rom 5645656456 but couldn't find it" or something like that. Thanks! (and yes, I have the latest proper FBA 029742 set, and tried with the latest Mame 0.191 rom as well, same result)

Kivutar, 2017-11-03:
I think that I found the cause of this. There was a bug in RC5 about default cores in playlists. It was affecting FBA.

ToKe79, 2018-01-31:
Hi, can you, please, re-test your issue with latest [stable](http://www.lakka.tv/get) or [nightly](http://nightly.builds.lakka.tv)? Thank you!

Ntemis, 2018-02-14:
closing due to inactivity, if issue remains please answer back

closed 2018-02-14 by Ntemis