libretro/Lakka-LibreELEC #270: Redundant "run" selection / Flow issue

Arnochocobo, 2017-11-04:
As usual, on Xu4 with the latest and greatest version.

When you select a rom in a playlist, you go through this flow:

Current behavior:
Select a ROM -> Select "Run" -> Select Core -> then by default you go to Information, have to press UP to highlight "Run" -> game starts
Going to Information in the middle is a completely unwanted and unnecessary behavior, and you have to select Run twice to start a game.

Expected behavior:
Select a ROM -> Select "Run" -> Select Core -> game starts
OR
Select a ROM ->  Select Core -> Select "Run" -> game starts  (would also be acceptable)

Ntemis, 2018-02-14:
Is this issue fixed? Let me know if i can close this, thank you

ToKe79, 2018-02-15:
From my observations on the devel builds, the flow is already correct. Please, re-open, in case not true.

closed 2018-02-15 by ToKe79