libretro/Lakka-LibreELEC #271: Steam Controller support in Lakka

jphaskins, 2017-11-04:
I'm using Lakka 2.1-RC5 on a Raspberry Pi 3. I plugged in a paired Steam Controller and dongle but it wasn't detected and I wasn't able to control the user interface. I believe that currently steam controllers only work with SDL since I didn't see a driver for it in /tmp/joypads/udev/.

Retropie and recalbox-os support the steam controller using a stand alone driver https://github.com/ynsta/steamcontroller. Perhaps the same solution could be used?

MasterGroosha, 2017-11-29:
Same here with the latest release :(
Steam Controller isn't detected.

Ntemis, 2017-11-29:
Steam controller isnt supported

MasterGroosha, 2017-12-01:
Any chance to have its support in future? There is a confog for SC in repo, isn't it?

Ntemis, 2017-12-02:
I personally dont have any of those so from my side is a nope.

ckdarby, 2018-01-23:
@Ntemis Yo, if I buy you one can you support it with the wireless features?

Ntemis, 2018-02-13:
I could try yes

Xscratch, 2018-02-21:
@ckdarby We could split the amount if you want.

ckdarby, 2018-02-21:
@Xscratch That would be much appreciated.

@Ntemis Can you pass along a way to send you $ like paypal and the cost of a steam controller in your country?

Ntemis, 2018-02-21:
Sorry guys there is no local store that sells those around here
Best move imo is to sent one from italy by @Xscratch as i live in Cyprus

ckdarby, 2018-02-22:
@Ntemis You can't order directly off steam store or Amazon?

Xscratch, 2018-02-22:
I found some used controllers on a craiglist-like italian site. I think one of those could do the job?

gouchi, 2018-02-23:
If [this driver](https://www.mail-archive.com/linux-kernel@vger.kernel.org/msg1614763.html) is accepted upstream, it will be great.

Ntemis, 2018-02-25:
Guys looks like its  99,49€  on steam store
[image]

FM1337, 2018-12-09:
I know this is an old issue, but I'm curious if there's been any progress made/news on adding support for this controller?

natinusala, 2018-12-10:
Actually I do own it, do you want me to try ?

ckdarby, 2018-12-10:
@natinusala :+1:

natinusala, 2018-12-10:
Not immediately though, I will be quite busy until mid January

jphaskins, 2018-12-10:
On newer kernels the driver I linked when I opened this issue shouldn't be necessary anymore. Starting with Linux 4.18 (and some fixes went into 4.19) there is a kernel driver module called hid_steam.

gouchi, 2018-12-11:
Indeed [hid_steam](https://cateee.net/lkddb/web-lkddb/HID_STEAM.html) has been integrated [from kernel 4.18](https://kernelnewbies.org/Linux_4.18#Tablets.2C_touch_screens.2C_keyboards.2C_mouses).

When this [issue](https://github.com/libretro/Lakka-LibreELEC/issues/556) will be fixed, we should get kernel 4.18 for generic platform.

FM1337, 2018-12-11:
> Indeed [hid_steam](https://cateee.net/lkddb/web-lkddb/HID_STEAM.html) has been integrated [from kernel 4.18](https://kernelnewbies.org/Linux_4.18#Tablets.2C_touch_screens.2C_keyboards.2C_mouses). [...]

That's great, hopefully that issue will be fixed in the near future.

Ntemis, 2018-12-12:
so guys shall we close this since it links to https://github.com/libretro/Lakka-LibreELEC/issues/556 ?

FM1337, 2018-12-14:
I'd probably recommend keeping this open until #556 is resolved. *shrug*

natinusala, 2018-12-14:
Would the driver work on older versions of Linux ? If so we can make it a patch and make it available for every platform at once.

FM1337, 2018-12-14:
It might be possible, if someone could link to the driver and a means of getting it to load I would be more than happy to test it.

gouchi, 2018-12-19:
For the [commit](https://git.kernel.org/linus/c164d6abf3841ffacfdb757c10616f9cb1f67276) if you want to try.

kivutar, 2018-12-25:
I also own this controller. I'm against patching the old kernels to support new controllers because we did it for xbox one and it was too much of a pain.

We can start supporting it for devices that have a recent kernel.

natinusala, 2018-12-25:
Fair enough. My ROCKPro64 will cry in a corner until Rockchip updates the kernel :p

untuned, 2019-01-24:
Happy new year, I also have a steam controller and would like this.

gouchi, 2019-01-26:
@FM1337 @untuned you may try this [testing build](https://drive.google.com/drive/folders/128MaH09bjVcB2pR4Kj2XxdBabR-oDP75) for x86_64 plateform.

[Source](https://github.com/libretro/Lakka-LibreELEC/issues/556#issuecomment-455847385)

ToKe79, 2019-07-24:
HI,
we have new devel builds which should support the steam controller. All feedback is appreciated!
http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/

ToKe79, 2019-07-26:
I created this config file for the Wireless Steam Controller https://github.com/libretro/retroarch-joypad-autoconfig/pull/549
Steam controller confirmed working on Lakka-LE9.2 Generic 64-bit.

PR #727 (2019-07-31): linux: enable Steam Controller
enable CONFIG_HID_STEAM on targets with Kernel 4.18 and newer
fixes #271
closed 2020-08-18 by Ntemis