libretro/Lakka-LibreELEC #272: 2.1 RC5: DragonRise USB Gamepads - ABXY Buttons not working in menu

carlmatthiasbrunogeorg, 2017-11-06:
Thank you so much for Lakka. I'm coming from Retropie and now chose Lakka over it. 

While the Buttons for those cheap usb gamepads work in 2.0 final, they do not in 2.1 RC5.
I'm running Lakka on an RPi2.

Thanks for your efforts.

Best

Matthias

gouchi, 2017-11-06:
Hi,

Is it possible to get the output of `lsusb` ? You will need to get access to [Lakka command line interface](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/).

Thank you.

carlmatthiasbrunogeorg, 2017-11-07:
Gouchi, thanks for your quick and elaborate reply. Sure:

Bus 001 Device 006: ID 0079:0006 DragonRise Inc. PC TWIN SHOCK Gamepad
Bus 001 Device 004: ID 0079:0006 DragonRise Inc. PC TWIN SHOCK Gamepad

Please note that this is from my current setup with Lakka 2.0 final (which should not make a difference).

Kind regards
Matthias

gouchi, 2017-11-17:
Do you reproduce your issue with [Lakka 2.1 RC6](http://www.lakka.tv/articles/2017/11/07/lakka-21-rc6-released-with-new-allwinner-and-rockchip-images-and-kiosk-mode/) ?

Thank you.

ToKe79, 2017-12-09:
Hi, there are many gamepads with that VID/PID and also many autoconfigs which match these.
For example I also own one such gamepad. This gamepad has one button between the start/select buttons. When it is pressed, a led on the gamepad is lit and gamepad works as expected. Otherwise, the actions for the right analog sticks are A/B/X/Y buttons. I don't recall how the gamepad works / what works when the led is not lit.
See this autoconfig: https://github.com/libretro/retroarch-joypad-autoconfig/commit/b5bb02732f7ec4b18284bf43d5b3f8a65270511c

Kivutar, 2017-12-23:
Yes, as @ToKe79 said, there are *many* cheap joypads sharing these VID/PID. It's technically impossible to apply the right profile to each of them.
This regression certainly happened because somebody did override the profile in the repo to get his controller working instead of yours.
We should at least setup a git hook that prevents adding a profile with the same vid:pid:name that an already existing profile.

closed 2018-01-31 by ToKe79