libretro/Lakka-LibreELEC #274: X86_64 PS4 New DualShock 4 Remote Does Not Configure.

AnicetusCer, 2017-11-07:
Lakka 2.1 rc5
X86_64

Got a new PS4 Dual Shock Remote (model CUH-ZCT2E) that failed to map buttons correctly.

Re-mapped them, attached are the working mappings for wireless and cabled, was not sure how to get the touchpad to do something, would be cool to get that working as a mouse for older systems.

[PS4_DualShock4_CUH-ZCT2E_Mappings.zip](https://github.com/libretro/Lakka-LibreELEC/files/1449085/PS4_DualShock4_CUH-ZCT2E_Mappings.zip)

gouchi, 2017-11-17:
Do you reproduce the issue with [Lakka 2.1 RC6 ](http://www.lakka.tv/articles/2017/11/07/lakka-21-rc6-released-with-new-allwinner-and-rockchip-images-and-kiosk-mode/)?

It seems there is already a profile for [DS4 wired](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Sony-PlayStation4-DualShock4v2-Controller.cfg) which matches your configuration file but some buttons are different. 

There are already [two](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Sony-PlayStation4-DualShock4-Controller-Bluetooth.cfg) [profiles](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/udev/Sony-PlayStation4-DualShock4-Controller-Bluez.cfg) for the wireless.

Thank you.

AnicetusCer, 2017-11-18:
Hello , the issue is reproducible.

I updated to RC6 and removed my working config files from /storage/Joypads/udev rebooted and the issue returns, I replace my own good config files and it once again works.

I read another post also mentioning that they were having trouble with the latest model of the PS4 controller, the model I have mentioned above is one of the latest batches

See this release from sony 

https://blog.eu.playstation.com/2017/09/01/introducing-the-crystal-dualshock-4-wireless-controller-range/

I have the crystal controller, this is a special edition and it seems sony may have made some slight modifications to there V2 set up for these controllers (just a guess)

gouchi, 2017-11-22:
You can try to submit your configuration file to [retroarch-joypad-autoconfig](https://github.com/libretro/retroarch-joypad-autoconfig) repository as it is working now.

closed 2017-11-22 by gouchi

AnicetusCer, 2017-11-23:
Config files now submitted,
Not sure the wired version will be accepted as the input vendor and model are the same as the black DS4v2, despite the variation in button mapping..., but the wireless config ids were unique so that should be accepted.