libretro/Lakka-LibreELEC #275: RPi3-optimized build?

ryanfb, 2017-11-07:
Is there any specific reason for not including a RPi3-optimized build (vs. the existing RPi/RPi2 builds)? I've just tested building this by modifying `projects/RPi2/options` to have `TARGET_CPU="cortex-a53"` and `TARGET_FPU="neon-fp-armv8"` and everything seems to build fine. I'd be happy to make a pull request for this as well.

Kivutar, 2017-11-11:
Did you notice a performance improvement?

ryanfb, 2017-11-11:
I haven't switched to 2.1/HEAD on my RPi3 because I need WiFi/SSH for my setup (blocked by #238) - I can try rebuilding a RPi3-optimized version with a checkout of the v2.0 tag and compare to the default RPi2 build of v2.0 and get back to you. Any suggested stress tests or benchmarks?

Kivutar, 2017-12-10:
Yes, you can try the same heavy PSP game on both, and check the FPS.

haydenbjyoung, 2017-12-13:
@ryanfb @Kivutar Perhaps you could install an FPS counter and run simultaneous tests, as Kivutar suggests. However, you would need to play for a set time each time to get accurate results. Perhaps use this table as an example for your findings, that way it'll be easier to see for people.

## Name of Game - FPS Tests
<table>
  <tr><th>System<th>FPS (Test 1)<th>FPS (Test 2)<th>FPS (Test 3)<th>FPS (Avg.)</tr>
  <tr><td>RPi 2 (`TARGET_CPU="cortex_a53"` set)<tr>
  <tr><td>RPi 3 (`TARGET_FPU="neon-fp-armv8"` set)</tr>
</table>

mfuntowicz, 2017-12-15:
I also tried to compile following the TARGET_CPU and TARGET_FPU for RPI3, and I'm facing some N64 games unable to start (seg fault) : 

```
[libretro INFO] EmuThread: M64CMD_ROM_OPEN
[libretro INFO] mupen64plus: Goodname: Mario Kart 64 (E) (V1.1) [!]
[libretro INFO] mupen64plus: Name: MARIOKART64
[libretro INFO] mupen64plus: MD5: 2BB149A583FDEFEA96805F628FE42FD9
[libretro INFO] mupen64plus: CRC: 2577C7D4 D18FAAAE
[libretro INFO] mupen64plus: Imagetype: .z64 (native)
[libretro INFO] mupen64plus: Rom size: 12582912 bytes (or 12 Mb or 96 Megabits)
[libretro INFO] mupen64plus: ClockRate = F
[libretro INFO] mupen64plus: Version: 1446
[libretro INFO] mupen64plus: Manufacturer: Nintendo
[libretro INFO] mupen64plus: Cartridge_ID: 544B
[libretro INFO] mupen64plus: Country: Unknown (0x150)
[libretro INFO] mupen64plus: PC = 80000400
[libretro INFO] mupen64plus: Save type: 0
[libretro INFO] EmuThread: M64CMD_ROM_GET_HEADER
[INFO] [CHEEVOS]: system RAM: 0x6c09e6b0 8388608
[INFO] [CHEEVOS]: save RAM:   0x6d69e6b0 296960
[INFO] [CHEEVOS]: video RAM:  (nil) 0
[INFO] [CHEEVOS]: RTC:        (nil) 0
Failed to open /storage/savefiles/Mario Kart 64 (E) (V1.1) [!].srm: No such file or directory
[INFO] Ignore le chargement de la SRAM..
[INFO] Version de l'API libretro: 1
[INFO] Compilé depuis l'API: 1
[INFO] [Audio]: Set audio input rate to: 44100.00 Hz.
[INFO] [Video]: Timings deviate too much. Will not adjust. (Display = 59.94 Hz, Game = 50.00 Hz)
[INFO] Shader trouvé "/tmp/shaders/bilinear.glslp"
[INFO] Shader trouvé "/tmp/shaders/nearest.glslp"
[INFO] Shader trouvé "/tmp/shaders/retroarch.glslp"
[INFO] Shader trouvé "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Shader trouvé "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Using HW render, OpenGL driver forced.
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL]: eglSwapInterval(0)
[INFO] [EGL]: eglSwapInterval(0)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] Querying EGL extension: KHR_image => exists
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GL]: Using 1 textures.
[INFO] [GL]: Loaded 1 program(s).
[INFO] [GL]: Initializing HW render (512 x 512).
[INFO] [GL]: Max texture size: 2048 px, renderbuffer size: 2048 px.
[INFO] [GL]: Supports FBO (render-to-texture).
[ERROR] [GL]: Failed to create HW render FBO #0, error: 0x36054.
[ERROR] [GL]: Hardware rendering context initialization failed.
[ERROR] [Video]: Cannot open video driver ... Exiting ...
[ERROR] Erreur fatale dans: "init_video()"
[ERROR] This core requires a content file.

[ERROR] Échec du démarrage du pilote audio. Pas de son.
[libretro INFO] EmuThread: M64CMD_EXECUTE.
Segmentation fault (core dumped)
```

ryanfb, 2017-12-22:
Finally had some time to try out various ROMs with Lakka 2.0 RPi2-optimized vs. RPi3-optimized as suggested.

For replicability's sake, here's the resulting RPi3-optimized image I used: <http://cryptosystem.org/projects/lakka/Lakka-RPi2.arm-8.0-devel-20171113163931-r26049-ga2a046c.img.gz>
And the state of the build directory: https://gist.github.com/ryanfb/075e0b57a6f5859fd41b408db7c7d148

Anyway, on to the results! As suggested, I tried to find some PSP games that ran <30fps on the current RPi2-optimized build. I couldn't copy the Markdown out of @haydenbjyoung's comment, but I think this will still suffice:

|                               | RPi2-optimized | RPi3-optimized | Improvement |
|-------------------------------|----------------|----------------|-------------|
| God of War: Chains of Olympus | 10fps          | 11fps          | 1fps        |
| Burnout Legends (car select)  | 27fps          | 30fps          | 3fps        |
| Burnout Legends (gameplay)    | 13.7fps        | 15.6fps        | 1.9fps      |
| FlatOut: Head On              | 10.3fps        | 12.1fps        | 1.8fps      |

So, nothing dramatic, but a slight improvement (for PPSSPP at least).

Also, to @mfuntowicz's concern, I was able to launch Mario Kart 64 and Super Mario 64 under mupen64plus on the RPi3-optimized build. So I'm not sure what's going on there, if you want to re-try with the image I posted above and see if it's still a problem.

mfuntowicz, 2017-12-22:
Hum, thanks @ryanfb for sharing your results.

I'm currently not able to run both N64 or PSP games, but was using some more recent version of mupen64plus and ppsspp (segfault). I've reverted those changes and started rebuilding.

Btw, I spot that mupen64plus has a RPi3 specific Makefile rule : https://github.com/libretro/mupen64plus-libretro/blob/master/Makefile#L87

I've made a RPi3 projects, and included the platform=rpi3 in the mupen64plus package.md. If possible @ryanfb, could it be possible for you to make a FPS test running a N64 game (Mario Kart 64 ?), to see if this specific rule change anything ? 

I've included changes from your diff.
News very soon :)

Kivutar, 2017-12-23:
Ok thanks a lot for these tests. Things like 1 or 3 FPS gains can also be obtained by many other ways, small improvements in the code, a little bit of overclocking, a great power supply like Aukrus, etc.

So yes, nothing dramatic so far.

kivutar, 2018-01-31:
PSP and N64 have been fixed, if you want to do more tests on this

closed 2019-07-02 by ToKe79