libretro/Lakka-LibreELEC #276: Logitech Rumblepad2 USB: issues in Lakka menu

macmpi, 2017-11-08:
Tried on Pi with Lakka 2.0 and 2.1rc5

Although detected fine (yellow message confirming connection), the controller can not operate the menu properly.
Neither DPAD or Analog R/L are working.
2-3-10 eventually have some effects, but menu is not usable.

Thanks for any tip.

```
Lakka:~ # dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.20 (kivutar@Server01) (gcc version 6.2.0 (GCC) ) #1 Thu Apr 6 03:12:39 CDT 2017
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi Zero Rev 1.3
[    0.000000] cma: Reserved 8 MiB at 0x15400000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 90112
[    0.000000] free_area_init_node: node 0, pgdat 8093b7b8, node_mem_map 95cd4000
[    0.000000]   Normal zone: 792 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 90112 pages, LIFO batch:15
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 89320
[    0.000000] Kernel command line: bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=720 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  boot=/dev/sda1 disk=/dev/sda2 quiet  vt.global_cursor_default=0 loglevel=2 root=/dev/ram0 rdinit=/init BOOT_IMAGE=/kernel.img usbcore.autosuspend=-1
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 338744K/360448K available (5649K kernel code, 226K rwdata, 1252K rodata, 2292K init, 374K bss, 13512K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0x96800000 - 0xff800000   (1680 MB)
[    0.000000]     lowmem  : 0x80000000 - 0x96000000   ( 352 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x8058c760   (5650 kB)
[    0.000000]       .init : 0x806c7000 - 0x80904000   (2292 kB)
[    0.000000]       .data : 0x80904000 - 0x8093c9c8   ( 227 kB)
[    0.000000]        .bss : 0x8093c9c8 - 0x8099a568   ( 375 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000026] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000063] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000139] bcm2835: system timer (irq = 27)
[    0.000513] Console: colour dummy device 80x30
[    0.000533] console [tty0] enabled
[    0.000562] Calibrating delay loop... 697.76 BogoMIPS (lpj=1161216)
[    0.026921] pid_max: default: 32768 minimum: 301
[    0.027121] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.027135] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.028006] Disabling cpuset control group subsystem
[    0.028133] CPU: Testing write buffer coherency: ok
[    0.028517] Setting up static identity map for 0x8200 - 0x8238
[    0.030184] devtmpfs: initialized
[    0.038096] VFP support v0.3: implementor 41 architecture 1 part 20 variant b rev 5
[    0.038473] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6370867519511994 ns
[    0.038497] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.039687] pinctrl core: initialized pinctrl subsystem
[    0.041107] NET: Registered protocol family 16
[    0.043504] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.051813] cpuidle: using governor ladder
[    0.051827] cpuidle: using governor menu
[    0.052411] Serial: AMBA PL011 UART driver
[    0.054954] bcm2835-mbox 2000b880.mailbox: mailbox enabled
[    0.100693] bcm2835-dma 20007000.dma: DMA legacy API manager at 9680d000, dmachans=0x1
[    0.103118] SCSI subsystem initialized
[    0.103570] usbcore: registered new interface driver usbfs
[    0.103694] usbcore: registered new interface driver hub
[    0.103853] usbcore: registered new device driver usb
[... 288 lines cut ...]
RetroArch [INFO] :: [udev] Adding device /dev/input/event0 as type ID_INPUT_KEYBOARD.
RetroArch [INFO] :: [udev]: Plugged pad: Logitech Logitech RumblePad 2 USB (1133:49688) on port #0.
RetroArch [INFO] :: Autodetect: 179 profiles found.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event1) supports force feedback.
RetroArch [INFO] :: [udev]: Pad #0 (/dev/input/event1) supports 16 force feedback effects.
RetroArch [INFO] :: Found joypad driver: "udev".
RetroArch [INFO] :: ALSA: Using signed 16-bit format.
RetroArch [INFO] :: ALSA: Period size: 384 frames
RetroArch [INFO] :: ALSA: Buffer size: 1536 frames
RetroArch [INFO] :: Found menu display driver: "menu_display_gl".
RetroArch [INFO] :: Autodetect: selected configuration: /tmp/joypads/udev/Logitech_RumblePad_2_USB.cfg
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: Using font rendering backend: freetype.
RetroArch [INFO] :: SRAM will not be saved.
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
RetroArch [INFO] :: Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
RetroArch [INFO] :: Saved new config to "/storage/.config/retroarch/retroarch.cfg".
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_video_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_music_history.lpl
RetroArch [INFO] :: Trying to write to playlist file: /storage/.config/retroarch/content_image_history.lpl
RetroArch [INFO] :: Threaded video stats: Frames pushed: 2440, Frames dropped: 0.
```

ukos-git, 2017-12-03:
I can confirm. Unplugging the Rumblepad receiver and I am able to navigate the menu again.

mstroh76, 2018-01-24:
I have the same problem. The default controller configuration is wrong. I reassigned the controls in input menu with help of a keyboard. But later I stopped using it. Someone should provide a correct configuration.

gouchi, 2018-01-25:
Do you reproduce the issue with [Lakka 2.1](http://www.lakka.tv/articles/2017/11/26/lakka-21-stable-release) ?

mstroh76, 2018-02-02:
Lakka-RPi.arm-2.1-devel-r28180-g31ef195 and Pi Zero W:
problem the same , emailing autoconfig of joypad

[Logitech Logitech RumblePad 2 USB.cfg.zip](https://github.com/libretro/Lakka-LibreELEC/files/1691056/Logitech.Logitech.RumblePad.2.USB.cfg.zip)

paalfe, 2018-02-04:
Pull request created for the correct udev cfg from @mstroh76  :+1: 
https://github.com/libretro/retroarch-joypad-autoconfig/pull/428

macmpi, 2018-02-04:
Thanks.
Is `input_device = "Logitech Logitech RumblePad 2 USB"` with _Logitech_ name twice correct?

Shouldn't the [hid cfg](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/hid/Logitech_Logitech_RumblePad_2_USB.cfg) be modified accordingly then?

paalfe, 2018-02-04:
the input_device may be different between hid, udev, android, xinput... You are using udev (linux), for udev it is input_device = "Logitech Logitech RumblePad 2 USB"

**In your dmesg it shows as:**
`[   30.049566] input: Logitech Logitech RumblePad 2 USB as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.4/1-1.4.4/1-1.4.4:1.0/0003:046D:C218.0002/input/input1`

**In your RetroArch info it shows as:**
`RetroArch [INFO] :: [udev]: Plugged pad: Logitech Logitech RumblePad 2 USB (1133:49688) on port #0.`

macmpi, 2018-02-04:
Yes, not too concerned about that naming thing anyway but, just thought repetition was looking award.
Agreed linux is using **udev** and, though I can't check right now, your submitted change is probably fixing the issue for that platform, thanks for that.

Was just wondering if platforms using **hid** would not also experience issues with this joypad, as **hid** config does not seem to have consistent entries with the one you submitted; ex:
`input_up_btn_label = "D-Pad Up"`  in your fix     _vs_
`input_up_btn_label = "Dpad Up"`   (and 17 other unmatching entries) in [hid cfg](https://github.com/libretro/retroarch-joypad-autoconfig/blob/master/hid/Logitech_Logitech_RumblePad_2_USB.cfg)

paalfe, 2018-02-04:
Pull request created.
https://github.com/libretro/retroarch-joypad-autoconfig/pull/429

macmpi, 2018-02-04:
Thanks!
Closing now.

closed 2018-02-04 by macmpi