libretro/Lakka-LibreELEC #277: Generic PS3 Controller Recognized but input is not on S905

zach-morris, 2017-11-09:
If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using? 2.1rc5
- What system hardware are you using? S905
- What did you do? Plugged in the generic controller, retroarch recognizes controller, but input is not recognized except for the center PS button
- What did you expect to happen? Input is recognized
- What happened instead? Input is not recognized

I've got a few Generic PS3 controllers.  These controllers work with Lakka x86 builds.  I'm testing them now with Lakka and an S905 device.  Plugged in USB, and input is not recognized, except for the center 'PS' button.  Same with bluetooth, although I believe when I try to connect it via bluetooth the controller just times out on the connection.  Same exact controller works with no fussing using Lakka on an x86 box, so I'm guessing it's a driver issue of some kind?

Retroarch 'sees' the controller when i plug it in.  But then attempting to map any buttons it doesn't appear to register the inputs.

The controller(s) are the ones pictured [here](https://www.amazon.com/gp/product/B00GQT0YWI).  Although I think the generic PS3 controllers are generally all the same on the inside.

Here's all the info I could muster from ssh (summarized for info on the controller):
```

xxx:~ # dmesg
...
[   81.498330@3] usb 1-1: USB disconnect, device number 2
[   94.617144@3] usb 1-1: new full-speed USB device number 3 using xhci-hcd
[   94.758175@3] usb 1-1: New USB device found, idVendor=054c, idProduct=0268
[   94.758201@3] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   94.758216@3] usb 1-1: Product: PS(R) Gamepad
[   94.758230@3] usb 1-1: Manufacturer: Gasia Co.,Ltd
[   94.763760@3] sony 0003:054C:0268.0002: Fixing up Sony Sixaxis report descriptor
[   94.766170@3] input: Gasia Co.,Ltd PS(R) Gamepad as /devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:054C:0268.0002/input/input5
[   94.782116@1] sony 0003:054C:0268.0002: input,hiddev0,hidraw0: USB HID v1.11 Joystick [Gasia Co.,Ltd PS(R) Gamepad] on usb-xhci-hcd.0.auto-1/input0

[   73.104905@2] sony 0003:054C:0268.0002: Fixing up Sony Sixaxis report descriptor
[   73.149918@2] input: Sony PLAYSTATION(R)3 Controller as /devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:054C:0268.0002/input/input5
[   73.151395@2] sony 0003:054C:0268.0002: input,hiddev0,hidraw1: USB HID v1.11 Joystick [Sony PLAYSTATION(R)3 Controller] on usb-xhci-hcd.0.auto-1/input0

xxx:~ # bluetoothctl
...
[PLAYSTATION(R)3 Controller]# info
Device 00:26:5C:F2:96:B0
	Name: PLAYSTATION(R)3 Controller
	Alias: PLAYSTATION(R)3 Controller
	Class: 0x000540
	Icon: input-keyboard
	Paired: no
	Trusted: yes
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Human Interface Device... (00001124-0000-1000-8000-00805f9b34fb)
	Modalias: usb:v054Cp0268d0000

xxx:~ # udevadm info --attribute-walk /dev/input/js0
...
  looking at device '/devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:054C:0268.0006/input/input9/js0':
    KERNEL=="js0"
    SUBSYSTEM=="input"
    DRIVER==""

  looking at parent device '/devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:054C:0268.0006/input/input9':
    KERNELS=="input9"
    SUBSYSTEMS=="input"
    DRIVERS==""
    ATTRS{name}=="Gasia Co.,Ltd PS(R) Gamepad"
    ATTRS{phys}=="usb-xhci-hcd.0.auto-1/input0"
    ATTRS{properties}=="0"
    ATTRS{uniq}==""

  looking at parent device '/devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0/0003:054C:0268.0006':
    KERNELS=="0003:054C:0268.0006"
    SUBSYSTEMS=="hid"
    DRIVERS=="sony"

  looking at parent device '/devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1/1-1:1.0':
    KERNELS=="1-1:1.0"
    SUBSYSTEMS=="usb"
    DRIVERS=="usbhid"
    ATTRS{bAlternateSetting}==" 0"
    ATTRS{bInterfaceClass}=="03"
    ATTRS{bInterfaceNumber}=="00"
    ATTRS{bInterfaceProtocol}=="00"
    ATTRS{bInterfaceSubClass}=="00"
    ATTRS{bNumEndpoints}=="02"
    ATTRS{supports_autosuspend}=="1"

  looking at parent device '/devices/c9000000.dwc3/xhci-hcd.0.auto/usb1/1-1':
    KERNELS=="1-1"
    SUBSYSTEMS=="usb"
    DRIVERS=="usb"
    ATTRS{authorized}=="1"
    ATTRS{avoid_reset_quirk}=="0"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{bDeviceClass}=="00"
    ATTRS{bDeviceProtocol}=="00"
    ATTRS{bDeviceSubClass}=="00"
    ATTRS{bMaxPacketSize0}=="64"
    ATTRS{bMaxPower}=="500mA"
    ATTRS{bNumConfigurations}=="1"
    ATTRS{bNumInterfaces}==" 1"
    ATTRS{bcdDevice}=="0100"
    ATTRS{bmAttributes}=="80"
    ATTRS{busnum}=="1"
    ATTRS{configuration}==""
    ATTRS{devnum}=="5"
    ATTRS{devpath}=="1"
    ATTRS{idProduct}=="0268"
    ATTRS{idVendor}=="054c"
    ATTRS{ltm_capable}=="no"
    ATTRS{manufacturer}=="Gasia Co.,Ltd"
    ATTRS{maxchild}=="0"
    ATTRS{product}=="PS(R) Gamepad"
    ATTRS{quirks}=="0x0"
    ATTRS{removable}=="unknown"
    ATTRS{speed}=="12"
    ATTRS{urbnum}=="22"
    ATTRS{version}==" 2.00"
```

Tom-Neverwinter, 2017-11-11:
I have the same issue with the ps3 controller on the allwinner opipc build. (switched to this after ps4 controller would not work [known bug])

cbluth, 2017-11-21:
I am experiencing the same issue on an s802 M8 device with v2.1RC6
with a generic chinese 3rd party sixaxis controller that had worked previously on the x64 arch.
and i get the same behaviour from a known working original sony sixaxis controller as well.

Azimarel, 2017-11-26:
I am also using a Gasia PS3 controller on an S802.
Via USB it is not working out of the box, because the sixaxis driver from kernel 3.10 (S802) does not support the Gasia controller. I think this is the same for kernel 3.14 (S905).
I found a patch ( https://patchwork.kernel.org/patch/3394521 ), don't ask me about quality or what it does to other controllers, for me it is working.
For bluetooth is more or less the same I think. Bluez is not able to connect to Gasia controllers.
The only way I found to get this controller working via bluetooth was to add sixad to the image.
( https://github.com/RetroPie/sixad )
This should only be a hint if you compile Lakka by yourself.
Adding sixad to the official Lakka images is not so easy I think, you would need a switch to enable/disable and many more for daily use/users. Also the kernel patch is not for "official" use.
I have no coding experience, therefore I can't provide clean solution.

Tom-Neverwinter, 2017-12-03:
issue is even worse on the new build.... /deletes all controller "autoconfig" especially ps3 and ps4.... inserts manual controls....  

I still want the retropi version over this "autodetect" thing. it cannot read the ps4/ps3 controller versions apart or read them appropriately. my ps4/dongle bundle work like garbage with the default config or at all. as well as the ps3 controller...

Kivutar, 2017-12-23:
Yeah this system was working well before. Then Sony started manufacturing controllers that have the same vid:pid:name but a different binding. Don't ask me why...
I just hope that the kernel module dev is going to find a way to bind them in a consistent way and clean all this mess.

closed 2020-08-18 by Ntemis