libretro/Lakka-LibreELEC #278: can't compile latest

Arnochocobo, 2017-11-11:
I am getting this error

UNPACK Libretro-database
tar: skipping to next header
tar: skipping to next header
gzip: stdin: invalid compressed data--crc error
gzip: stdin: invalid compressed data--lenght error
tar: child returned status 1
tar: error is not recoverable: exiting now

and then image error 2.

NB: 0 issues compiling until this commit. (Ubuntu 16.04 LTS)

Arnochocobo, 2017-11-11:
I can confirm everything is fine rolling back one commit:
77eb774a8e3a4c655af8472856dc146da0f194db

So 441fa3cbd587dea088ab56c271672f93382715be is 99% to blame.

RobLoach, 2017-11-12:
This should be fixed now.

Arnochocobo, 2017-11-13:
It is not, exact same problem when I try to compile latest.

RobLoach, 2017-11-13:
What error do you get?

Arnochocobo, 2017-11-14:
Just tried again with the latest commit. I am trying to make an image for XU4.

UNPACK libretro-database
tar: skipping to next header
tar: skipping to next header
gzip: stdin: invalid compressed data--crc error
gzip: stdin: invalid compressed data--lenght error
tar: child returned status 1
tar: error is not recoverable: exiting now

and then image error 2.

Same problem as earlier.

Kivutar, 2017-11-14:
Try to rm -rf sources/libretro-database*

Arnochocobo, 2017-11-14:
Thanks Kivutar, it worked. I used the usual git pull to get latest and never had a problem with it. What was the issue? Did I mess up anything?

closed 2017-11-17 by Ntemis

Kivutar, 2017-11-18:
A corrupted download was there in the sources cache