libretro/Lakka-LibreELEC #279: MAME 2003 stopped working RPi3 2.1 RC6

masid101, 2017-11-12:
- Which version of Lakka are you using? 2.1 RC6
- What system hardware are you using? RPi3
- What did you do? Started a MAME2003 game
- What did you expect to happen? A working game like in RC5
- What happened instead? Error

Here's the Log ...

Lakka (official): 2.1-rc6 (RPi2.arm)
Lakka:~ # systemctl stop retroarch.service ; retroarch -v
[INFO] RetroArch 1.6.7 (Git 4e46312)
[INFO] === Build =======================================
Capabilities: NEON VFPv3 VFPv4 
Built: Nov  4 2017
[INFO] Version: 1.6.7
[INFO] Git: 4e46312
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [EGL] Falling back to eglGetDisplay
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x1.
[INFO] [GL]: Found GL context: videocore
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [EGL]: eglSwapInterval(1)
[INFO] [GL]: Vendor: Broadcom, Renderer: VideoCore IV HW.
[INFO] [GL]: Version: OpenGL ES 2.0.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] Shader log: Compiled
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] Shader log: Compiled
[... 135 lines cut ...]
[INFO] [udev]: Keyboard #1 (/dev/input/mouse0).
[ERROR] [udev] Failed to open device: /dev/input/event0 (Success).
[INFO] [udev]: Mouse #0 (/dev/input/mouse0).
[INFO] [udev]: Plugged pad: PLAYSTATION(R)3 Controller (0:0) on port #0.
[INFO] [udev]: Pad #0 (/dev/input/event1) supports force feedback.
[INFO] [udev]: Pad #0 (/dev/input/event1) supports 16 force feedback effects.
[INFO] [Autoconf]: 186 profiles found.
[INFO] [Joypad]: Found joypad driver: "udev".
[WARN] [udev]: Full-screen pointer won't be available.
[INFO] ALSA: Using signed 16-bit format.
[INFO] ALSA: Period size: 384 frames
[INFO] ALSA: Buffer size: 1536 frames
[INFO] [Autoconf]: selected configuration: /tmp/joypads/udev/Sony-PlayStation3-DualShock3-Controller-Bluetooth.cfg
[ERROR] Failed to open libretro core: "/tmp/cores/mame2003_libretro.so"
[ERROR] Error(s): /tmp/cores/mame2003_libretro.so: undefined symbol: driver_baryon
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] SRAM will not be saved.
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[ERROR] Failed to open libretro core: "/tmp/cores/mame2003_libretro.so"
[ERROR] Error(s): /tmp/cores/mame2003_libretro.so: undefined symbol: driver_baryon

markwkidd, 2017-11-14:
@Kivutar and @gouchi there is a lot of development work going on with MAME 2003. It seems like the latest Lakka RC is built from a mame2003 commit with an error.

I think this is the PR that fixes this issue with `driver_baryon`: https://github.com/libretro/mame2003-libretro/pull/175

gouchi, 2017-11-17:
We have updated to [this commit](https://github.com/libretro/mame2003-libretro/commit/276f613ce09abb487431bb1569cc70138aea2a77)([Source](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/packages/libretro/mame2003/package.mk#L22)).

So it should be fixed in the next release.

masid101, 2017-11-25:
Fixed with 2.1 stable!

closed 2017-11-25 by masid101