libretro/Lakka-LibreELEC #281: Content updater downloads fails on sizeable archives

macmpi, 2017-11-13:
Issue on Pi 2.1RC6 (even 2.0)

Content download such as thumbnails fails to complete even-though I have sufficient disk space. Download % increases, and then resets to 0.
`Task Failed` message appears on screen after few tries.
Actually I could not find any interim download file anywhere even during download process...(any RAM caching issue? I have a 512MB PiZero)
Same for Lakka updates (download % gets to about 10-15% max and then restarts)

For smaller thumbnails archives (like Atari 2600, Coleco thumbnails), there is no problem: file is downloaded and extracted fine into place.
MAME, N64, SMS thumbnail archives all fail for instance.

RobLoach, 2017-11-13:
http://thumbnailpacks.libretro.com/?C=S&O=D

You can see some of those exceed the 512MB you have. Extracted, the size would be even larger. Consider either getting a larger file storage, or disabling thumbnails.

macmpi, 2017-11-13:
Rob,
I think you  misunderstood my statement: I was referring to 512MB of RAM (questioning memory caching problem in download process).
As mentioned, actual storage capacity is not an issue (about 16Gb)
Thanks for any tip

RobLoach, 2017-11-13:
Ah, cool :+1: ....... You could try downloading manually? Is there anything in the logs?

Arnochocobo, 2017-11-14:
Exact same problem here, been like that forever, and Kivutar told me to download them manually from the source URL. It worked, but the real problem is the actual feature in Lakka. It doesn't work for big files, as mentioned in the original report. Would be good to at least have a resume function and not have to restart from scratch all the time, since the download fails 100%.

metchebe, 2017-11-14:
This has also happened to me ever since I started using Lakka (Pre 2.0). However I don't think this is because of Lakka, if I recall correctly this hasn't worked for me either in Retroarch (Linux AND Windows). I have since given up entirely on the thumbnail download feature.

One way to solve this (at least for thumbnails) is to only download thumbnails for scanned games, this is the way I do it with a bash script in Linux.

macmpi, 2017-11-15:
Thanks for feedback and confirmation.
Indeed [actual download code](https://github.com/libretro/RetroArch/commits/master/libretro-common/net/net_http.c) seems to be in Retroarch if I'm correct.
Maybe @twinaphex & al. may have ideas on this apparently well-known and longstanding issue?

macmpi, 2018-02-04:
Could be that it was fixed in a [Retroarch Jan 2nd commit](https://github.com/libretro/RetroArch/commit/3528923507685512e90d4fb80f506e3ee2fca088#diff-1fa77422340388277921587608f6fe20).
It should then have made it's way into Lakka, with [recently updated RA package since that date](https://github.com/libretro/Lakka-LibreELEC/commits/Lakka-V2.1-dev/packages/libretro/retroarch/package.mk).

Any chance someone tried downloads/updates on current `Lakka-v2.1-dev` nighties to confirm?

macmpi, 2018-03-08:
tried with Lakka-RPi.arm-2.1-devel-20180305012454-r28154-gb552a9a26, and unfortunately issue is still there :(
@twinaphex any idea?
Thanks.

closed 2020-08-18 by Ntemis