libretro/Lakka-LibreELEC #283: TP Link tl-wn823n v2 Odroid XU4

juri74, 2017-11-15:
Hello, i just updated my Lakka machine to Lakka 2.1RC6 running on Odroid XU4.
I was running Lakka 2.0 stable before updating.
My TP Link tl-wn823n v2 wifi dongle is not working 100%, it is recognized and i was able to connect to my router, but connection is very unstable. I get lot of disconnection when using SSH and SAMBA.
The led light of the dongle is ALWAYS OFF, no blinking during network activity.

Ntemis, 2017-11-16:
Please give me lsusb

juri74, 2017-11-17:
hello, here the lsusb result.

`Lakka (official): 2.1-rc6 (OdroidXU3.arm)
Lakka:~ # lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 4971:1025 SimpleTech
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 004: ID 2357:0109
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ #`

thanks

juri74, 2017-11-17:
this is the original link when i had the issue first time, gouchi gave me a working image

https://forums.libretro.com/t/tp-link-tl-wn823n-v2-support/9294

thanks again

juri74, 2017-12-08:
UPDATE:
same problem with Lakka 2.1 Final

ToKe79, 2017-12-09:
[Here](https://vps.vudiq.sk/lakka-builds/OdroidXU3.arm/Lakka-OdroidXU3.arm-2.1-devel-20171209021622-r28009-g60313e988.img.gz) is latest developer build including the `RTL8192EU` driver.

juri74, 2017-12-09:
> Here is latest developer build including the RTL8192EU driver.
thanks for the effort @ToKe79 !
will be included in next official versions (2.2 beta and following) or i have to ask everytime?

ToKe79, 2017-12-09:
@juri74 does it work? I am not a developer, but I don't see any reason why not. but it is not up to my decission. Is there a reason why you are not using the internal wifi of your odroid box?

ToKe79, 2017-12-09:
[Here](https://vps.vudiq.sk/lakka-builds/Lakka-OdroidXU3+RTL8192EU.arm-2.1.img.gz) is also latest stable build with `RTL8192EU` driver.

ToKe79, 2017-12-09:
OK, looked at the specs, it has only ethernet.

ToKe79, 2017-12-10:
@juri74 can you please confirm, that it works, so it can be in later stable/devel. thanks!

juri74, 2017-12-13:
hello, i've already downloaded Lakka 2.1 stable a week ago, but same result :(
sorry for late reply, i'm ill very bad. give me some time do download the fixed Lakka 2.1 stable and try
thanks again for the effort

ToKe79, 2017-12-13:
Hi @juri74 , you can test these images:
[fixed stable](http://nightly.builds.lakka.tv/Lakka-OdroidXU3+RTL8192EU.arm-2.1.img.gz)
or
[latest dev build](http://nightly.builds.lakka.tv/OdroidXU3.arm/Lakka-OdroidXU3.arm-2.1-devel-20171212220828-r28021-g5dc54e46d.img.gz)
Best would be to confirm with the "fixed stable".
Thank you :-)

juri74, 2017-12-13:
hello again,
i downloaded your image and putted in ".update" folder
i rebooted and system did the update, after update interface icons looked different (so something happened). the file in the ".update" folder, disappeared.
unfortunately nothing changed :(
i did a lsusb:
```
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (community): 2.1 (OdroidXU3.arm)
Lakka:~ # lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 4971:1025 SimpleTech
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 004: ID 2357:0109
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ #

```

juri74, 2017-12-13:
will try with the "nightly" one, wait please :)

juri74, 2017-12-13:
ok, tested the "FIXED STABLE" -> didn't work. lsusb:
```
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (community): 2.1 (OdroidXU3.arm)
Lakka:~ # lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 4971:1025 SimpleTech
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 005: ID 2357:0109
Bus 003 Device 004: ID 2dc8:9001
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ #
```
and "LATEST DEV BUILD" (r28021) -> no success :'( lsusb:
```
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (community): devel-20171212220828-r28021-g5dc54e46d (OdroidXU3.arm)
Lakka:~ # lsusb
Bus 006 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 4971:1025 SimpleTech
Bus 004 Device 002: ID 05e3:0616 Genesys Logic, Inc. hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)
Bus 003 Device 005: ID 2357:0109
Bus 003 Device 004: ID 2dc8:9001
Bus 003 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Lakka:~ #
```

thanks again for your effort, fortunately even if wifi dongle do not work @100% i'm able to do some ssh..
maybe i will buy a certified working one in the future
thanks again for the effort @ToKe79 
Juri

ToKe79, 2017-12-13:
can you try `modprobe 8192eu` or plug/unplug the USB wifi?

juri74, 2017-12-13:
hello i typed "modprobe 8192eu" in ssh prompt, but nothing happened:

```
login as: root
root@192.168.1.105's password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (community): devel-20171212220828-r28021-g5dc54e46d (OdroidXU3.arm)
Lakka:~ # modprobe 8192eu
Lakka:~ # modprobe 8192eu
Lakka:~ #
```

for the plug/unplug procedure you have to wait some time because i have to open the case.

juri74, 2017-12-13:
hello even the plug/unplug worked, if it could help when i plugged the dongle the activity led blinked briefly

ToKe79, 2017-12-13:
Hi @juri74 
after `modprobe 8192eu` do a `lsmod` and `lsusb` and `dmesg`.
Maybe @gouchi pulled some other magic tricks out of the sleeves to make it work, I am not that familiar with kernel / modules staff. I just added the driver to the `ADDITIONAL_DRIVERS` of the `projects/OdroidXU3/options` file:
```
  # additional drivers to install:
  # for a list of additinoal drivers see packages/linux-drivers
  # Space separated list is supported,
  # e.g. ADDITIONAL_DRIVERS="DRIVER1 DRIVER2"
  # FIX for juri74
    ADDITIONAL_DRIVERS="$ADDITIONAL_DRIVERS RTL8192EU"
```
as written in the [forum post](https://forums.libretro.com/t/tp-link-tl-wn823n-v2-support/9294).

ToKe79, 2017-12-14:
Hi @juri74 
If you have time, please test the following [nightly](http://nightly.builds.lakka.tv/OdroidXU3.arm/Lakka-OdroidXU3.arm-2.1-devel-20171214003558-r28023-gd3c16e419.img.gz). I have added also `RTL8192CU` and `RTL8192DU` drivers.
After boot, if the wifi does not start, please try following:
```
dmesg -c
lsusb
lsmod
modprobe 8192eu
lsusb
lsmod
dmesg
```
The first `dmesg -c` will clear the messages, the second `dmesg` prints then only new messages.
`modprobe 8192eu` should do no output if the module loading was successful.
Save the output and try to connect to wifi.
Thank you for your time!

juri74, 2017-12-14:
ok installed that nightly but no succes, here we go:
DMESG -C:
```
Lakka (community): devel-20171214003558-r28023-gd3c16e419 (OdroidXU3.arm)
Lakka:~ # dmesg -c
[    0.000000] [c0] Booting Linux on physical CPU 0x100
[    0.000000] [c0] Initializing cgroup subsys cpuset
[    0.000000] [c0] Initializing cgroup subsys cpu
[    0.000000] [c0] Initializing cgroup subsys cpuacct
[    0.000000] [c0] Linux version 3.10.106 (vudiq@vps) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT Mon Dec 11 14:14:39 CET 2017
[    0.000000] [c0] CPU: ARMv7 Processor [410fc073] revision 3 (ARMv7), cr=10c5387d
[    0.000000] [c0] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] [c0] Machine: ODROID-XU3, model: Hardkernel odroid-xu3 board based on EXYNOS5422
[    0.000000] [c0] ION: Contiguous 0x6650000 bytes @ 0x0 defined for 0:common
[    0.000000] [c0] ION: Contiguous 0x400000 bytes @ 0x0 defined for 2:mfc_sh
[    0.000000] [c0] ION: Contiguous 0x800000 bytes @ 0x0 defined for 10:g2d_wfd
[    0.000000] [c0] ION: Contiguous 0x6000000 bytes @ 0x0 defined for 11:video
[    0.000000] [c0] ION: Contiguous 0x1000000 bytes @ 0x0 defined for 7:mfc_input
[    0.000000] [c0] ION: Contiguous 0x400000 bytes @ 0x0 defined for 9:sectbl
[    0.000000] [c0] ION: Contiguous 0x400000 bytes @ 0x0 defined for 8:mfc_fw
[    0.000000] [c0] ION: Contiguous 0x400000 bytes @ 0x0 defined for 12:mfc_nfw
[    0.000000] [c0] ION: Contiguous 0x400000 bytes @ 0x0 defined for 13:secdma
[    0.000000] [c0] cma: CMA: reserved 104 MiB at b8000000
[    0.000000] [c0] cma: CMA: reserved 4 MiB at b7c00000
[    0.000000] [c0] cma: CMA: reserved 8 MiB at b7400000
[    0.000000] [c0] cma: CMA: reserved 96 MiB at b1400000
[    0.000000] [c0] cma: CMA: reserved 16 MiB at b0400000
[    0.000000] [c0] cma: CMA: reserved 4 MiB at b0000000
[    0.000000] [c0] cma: CMA: reserved 4 MiB at afc00000
[    0.000000] [c0] cma: CMA: reserved 4 MiB at af800000
[    0.000000] [c0] cma: CMA: reserved 4 MiB at af400000
[    0.000000] [c0] cma: CMA: reserved 256 MiB at 5f800000
[    0.000000] [c0] Memory policy: ECC disabled, Data cache writealloc
[    0.000000] [c0] create_mapping_memory: section: 40000000 ~ 6f800000
[    0.000000] [c0] CPU EXYNOS5422 (id 0xe5422001)
[    0.000000] [c0] On node 0 totalpages: 518656
[    0.000000] [c0] free_area_init_node: node 0, pgdat c0d49d40, node_mem_map c11bf000
[    0.000000] [c0]   Normal zone: 1520 pages used for memmap
[    0.000000] [c0]   Normal zone: 0 pages reserved
[    0.000000] [c0]   Normal zone: 194560 pages, LIFO batch:31
[    0.000000] [c0]   HighMem zone: 2532 pages used for memmap
[    0.000000] [c0]   HighMem zone: 324096 pages, LIFO batch:31
[    0.000000] [c0] PERCPU: Embedded 9 pages/cpu @c21cd000 s14080 r8192 d14592 u36864
[    0.000000] [c0] pcpu-alloc: s14080 r8192 d14592 u36864 alloc=9*4096
[    0.000000] [c0] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7
[    0.000000] [c0] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 517136
[    0.000000] [c0] Kernel command line: console=ttySAC2,115200n8 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 ro quiet consoleblank=0 loglevel=2 vt.global_cursor_default=0 smsc95xx.macaddr=00:1e:06:61:7a:39 governor=performance hdmi_tx_amp_lvl=31 hdmi_tx_lvl_ch0=3 hdmi_tx_lvl_ch1=3 hdmi_tx_lvl_ch2=3 hdmi_tx_emp_lvl=6 hdmi_clk_amp_lvl=31 hdmi_tx_res=0 HPD=true vout=hdmi usbhid.quirks=0x0eef:0x0005:0x0004
[    0.000000] [c0] hdmi: using HDMI Mode
[    0.000000] [c0] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] [c0] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] [c0] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] [c0] allocated 4149248 bytes of page_cgroup
[    0.000000] [c0] please try 'cgroup_disable=memory' option if you don't want memory cgroups
[    0.000000] [c0] Memory: 2026MB = 2026MB total
[    0.000000] [c0] Memory: 1522268k/1522268k available, 552356k reserved, 1046528K highmem
[    0.000000] [c0] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xfff00000 - 0xfffe0000   ( 896 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc08d5fcc   (9016 kB)
[    0.000000]       .init : 0xc08d6000 - 0xc0cdb700   (4118 kB)
[    0.000000]       .data : 0xc0cdc000 - 0xc0d4e1a0   ( 457 kB)
[    0.000000]        .bss : 0xc0d4e1a0 - 0xc11befe0   (4548 kB)
[    0.000000] [c0] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[    0.000000] [c0] Preemptible hierarchical RCU implementation.
[    0.000000] [c0] NR_IRQS:16 nr_irqs:16 16
[    0.000000] [c0] apll 800000000
[... 1081 lines cut ...]
Module                  Size  Used by
hidp                   13840  0
bnep                   11204  2
8021q                  21789  0
garp                    6230  1 8021q
mrp                     7584  1 8021q
stp                     2028  1 garp
llc                     5089  2 stp,garp
8192eu               1102744  0
btusb                  16511  0
cfg80211              450133  1 8192eu
bluetooth             227932  22 bnep,hidp,btusb
rfkill                 13089  5 cfg80211,bluetooth
r8152                  60855  0
spidev                  5899  0
ina231_sensor           9032  0
autofs4                23436  0
```
and finally DMESG: (it didn't display any info)
```
Lakka:~ # dmesg
Lakka:~ #
```

Thanks..

juri74, 2017-12-14:
could be possible to try official linux drivers?
http://www.tp-link.com/us/download/TL-WN823N.html#Driver
thanks

ToKe79, 2017-12-14:
@juri74 to me it looks like it connected to the wifi:
```
[   18.724977] [c5] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   19.390528] [c0] RTL871X: start auth
[   19.396970] [c0] RTL871X: auth success, start assoc
[   19.409967] [c0] RTL871X: assoc success
[   19.412515] [c0] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   21.415615] [c0] RTL871X: recv eapol packet
[   21.419888] [c7] RTL871X: send eapol packet
[   21.435664] [c0] RTL871X: recv eapol packet
[   21.438816] [c6] RTL871X: send eapol packet
[   21.442740] [c7] RTL871X: set pairwise key camid:4, addr:f0:84:2f:a3:08:83, kid:0, type:AES
[   21.453216] [c7] RTL871X: set group key camid:5, addr:f0:84:2f:a3:08:83, kid:2, type:TKIP
```

juri74, 2017-12-15:
yes it connect from the beginning! problem is that it doesn't work 100%. after updating from lakka v2.0 release candidate provided to me originally from gouchi, i was able to connect but connection is very unstable. I get lot of disconnection when using SSH and SAMBA.
The led light of the dongle is ALWAYS OFF, no blinking during network activity.
tp-link provide a linux driver, but i do not know how to install it ..

ToKe79, 2017-12-15:
I build an image with the driver version as at gouche build the image:
http://nightly.builds.lakka.tv/Lakka-OdroidXU3+old_RTL8192EU.arm-2.1-devel.img.gz
I will try to build an image with the TP-Link driver.

ToKe79, 2017-12-15:
The driver from TP-Link does not compile for Linux 3.10.106 (according to the download page 2.6.18 ~ 3.10.10 is supported).

ToKe79, 2017-12-15:
So, the version used previously (`0c80da7`) refers to 4.3.1.1 version of the Realtek driver (master branch), current version (`192bad4`) used by Lakka refers to branch based on 4.4.1 version of the Realtek driver (4.4.x branch).
So [here](http://nightly.builds.lakka.tv/Lakka-OdroidXU3.arm-2.1-devel+RTL8192EU_43_driver.img.gz) is an image with the latest driver from [`master`](https://github.com/Mange/rtl8192eu-linux-driver/tree/master) branch.

juri74, 2017-12-15:
ok thanks, i will try both of them now, first the one 0c80da7 then 192bad4.  (oh btw what a funny version 192 "BAD" 4 :D )
thanks again

ToKe79, 2017-12-15:
Try the one I posted as latest.
`192bad4` you tested already with the 2.1-stable
`0c80da7` [image](http://nightly.builds.lakka.tv/Lakka-OdroidXU3+old_RTL8192EU.arm-2.1-devel.img.gz) - is the 2.1-devel with the old 4.3 driver as it was in 2.0-stable
`0af336b` [image](http://nightly.builds.lakka.tv/Lakka-OdroidXU3.arm-2.1-devel+RTL8192EU_43_driver.img.gz) - is the 2.1-devel with the 4.3 driver updated -> my last post

juri74, 2017-12-15:
Hello @ToKe79 
GOOOOOD NEWS :D 
i started testing with image "0af336b" and it WORK!
i'm transferring some files, activity led is blinking and i have not experienced any disconnection with samba until now.

i do not have the words to thank you!!
hope will work in future version :)
thanks again!

juri74, 2017-12-16:
it is working perfectly, i close the issue thanks!

cslam001, 2018-09-24:
I 've got similar issue : 

[ 1405.247092] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247137] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247161] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247184] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247206] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247228] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247250] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247273] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247295] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247316] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.247340] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253733] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253775] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253800] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253940] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253971] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.253995] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254017] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254040] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254062] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254085] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254107] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254130] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254152] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254174] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254198] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254221] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254243] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254266] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254288] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254311] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254333] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254356] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254378] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254401] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254423] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254446] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254468] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254491] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254513] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254536] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254558] RTL871X: Wrong rate section: (CCK,2T)
[ 1405.254653] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254676] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254699] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254722] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254744] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254766] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254789] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254811] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254834] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254856] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254879] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254901] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254924] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254946] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254969] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.254991] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255014] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255037] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255060] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255082] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255105] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255127] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255150] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255172] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255195] RTL871X: Wrong rate section: (OFDM,2T)
[ 1405.255217] RTL871X: Wrong rate section: (OFDM,2T)
[... 58 lines cut ...]
[ 1405.257910] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 04]
[ 1405.257952] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 05]
[ 1405.257994] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 05]
[ 1405.258036] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 05]
[ 1405.258078] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 06]
[ 1405.258120] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 06]
[ 1405.258161] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 06]
[ 1405.263151] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 07]
[ 1405.263217] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 07]
[ 1405.263261] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 07]
[ 1405.263303] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 08]
[ 1405.263345] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 08]
[ 1405.263387] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 08]
[... cut ...]
[ 1405.263680] RTL871X: duplicate tx power limit combination [band 2.4G][regulation FCC][bw 40M][rate section HT][rf path 2T][chnl 11]
[ 1405.263723] RTL871X: duplicate tx power limit combination [band 2.4G][regulation ETSI][bw 40M][rate section HT][rf path 2T][chnl 11]
[ 1405.263765] RTL871X: duplicate tx power limit combination [band 2.4G][regulation MKK][bw 40M][rate section HT][rf path 2T][chnl 11]
[ 1405.265105] RTL871X: rtw_ndev_init(ra0) if1 mac_addr=7c:8b:ca:19:da:36

And I am not able to get the source code for 0af336b image, would anyone provide me some help

ToKe79, 2018-09-24:
My guess is you are testing either official or nightly release. These releases do not have the "older" driver for RTL8192EU - for that I created a branch in my fork, unfortunately apparently I deleted the branch long time ago. I will try to prepare images with this driver. What device (Generic PC - 32/64-bit, Odroid, ...) are you using?

cslam001, 2018-09-27:
Thanks, problem solved !

ToKe79, 2018-09-27:
So were you able to build a custom image with this driver and wifi works as expected? Let me know, so I can send a merge request to the master branch :-)

closed 2019-06-26 by ToKe79

rtaylo, 2019-09-17:
Hi There, I've got the same wireless adapter.
Currently using one of the September nightly builds. My machine is just a generic PC. 
Any ideas on things I can try?

ToKe79, 2019-09-17:
Hi @rtaylo 
I think you should try our devel build based on the LibreELEC 9.2 buildsystem:
http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/
It is the base for the next major release.