libretro/Lakka-LibreELEC #285: Recording function doesn't work in Orange Pi Plus 2e

zarkon0, 2017-11-18:
The Recording function doesn't work in Orange Pi Plus 2e.
Recording doesn't work saving to .mkv file.
Recording doesn't work streaming to twitch.

The file name and other options in the Recoding option menu never save to retroarch.cfg file.
If edit manually retroarch.cfg to include recording options lakka clear this options.

Kivutar, 2017-12-10:
Ok, can you try recording using the command line?

zarkon0, 2017-12-11:
Where I can use the command line?

I connect with putty to the Orange Pi Plus 2e but I don't know the location
of the retroarch executable and the command line that I must be used.

Kivutar, 2017-12-14:
Take a look at our wiki, everything is explained about accessing the command line

closed 2020-08-18 by Ntemis