libretro/Lakka-LibreELEC #286: No Boot Black Screen for M8 s802

cbluth, 2017-11-19:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
I have tried Lakka versions:
Lakka-S802.M8.arm-2.0.img.gz
Lakka-S8X2.M8.arm-2.1-rc6.img.gz
Lakka-S8X2.M8.arm-2.1-rc5.img.gz

I have also tried these SD cards:

[image]

- What system hardware are you using?
M8 s802 

- What did you do?
I first used etcher to write the image, i tried all image files, and all SD cards. The v2.0 image is the only image that produces different behavior. It tries to boot, then it says "cant stat /flash/SYSTEM", and then drops to a prompt. The other RC5-6 image files just sit at a black screen.

- What did you expect to happen?
Lakka to boot

- What happened instead?
Black screen

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Ntemis, 2017-11-20:
Dont bother you cant boot from sdcard on S802 please try with a usb stick

closed 2017-11-20 by Ntemis

cbluth, 2017-11-21:
https://github.com/lakkatv/lakka-website/pull/45

cbluth, 2017-11-21:
@Ntemis https://github.com/lakkatv/lakka-website/pull/45