libretro/Lakka-LibreELEC #287: Can't compile latest (commit 034ddbdabe825cb0de6cceb933d19d1b140362be)

Arnochocobo, 2017-11-20:
Compile starts, and then while downloading retroarch 7ff25fe, I get Read Error at byte 14680064.

Ntemis, 2017-11-20:
Time to change hdd imo, close.

closed 2017-11-20 by Ntemis

Arnochocobo, 2017-11-20:
My SSD is completely fine, and was not the problem for issue #278.

No more issues with retroarch f27574b though.

Arnochocobo, 2017-11-21:
Could compile latest (1f6b4166ccdb4d5267bdd88e352e9679096933c8). No problem, even on my "broken HDD".

Instead of me changing my "HDD", I would recommend stopping uploading broken binaries, or at least checking them (downloading and trying to compile) before accusing people of having faulty hardware, imo.

Ntemis, 2017-11-21:
Then is either your internet connection  or github has some issues on your
part of the world.