libretro/Lakka-LibreELEC #289: Please Include rsync

cbluth, 2017-11-21:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
2.1RC6
- What system hardware are you using?
s802 M8
- What did you do?
tried to use rsync
- What did you expect to happen?
rsync to be available
- What happened instead?
rsync was not available

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

---

Please include rsync, it is very useful for transferring content over ssh. The orignal issue was found here: https://github.com/libretro/Lakka/issues/517

Here is a workaround:
Build a statically linked binary: http://www.virtuallyghetto.com/2011/02/how-to-compile-statically-linked-rsync.html

Put it in ~/.local/usr/bin/
Then I did something like this:
rsync -avPz --rsync-path=/storage/.local/usr/bin/rsync /path/ 192.168.1.12:/storage/roms/

RobLoach, 2017-11-21:
It's included as part of `network-tools`:
https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/packages/addons/tools/network-tools/package.mk

We may need to install network-tools in the OEM package.

closed 2017-11-27 by Ntemis

cplr, 2018-08-10:
How does one install the network-tools package? I can't find any documentation on this.

kadamwolfe, 2019-02-09:
> How does one install the network-tools package? I can't find any documentation on this.

I'm in the same boat.  Having rsync installed/included would be fantastic.

garrison, 2019-03-17:
> How does one install the network-tools package?

From within Kodi: Add-ons -> Install from repository -> LibreELEC Add-ons -> Program add-ons -> Network Tools

natinusala, 2019-03-17:
Kodi isn't included in Lakka though. You need to rebuild the image yourself with this package included

WHK102, 2020-12-09:
The problem: need a package. The solution: Install a package but need install other package for install the packages, but can not install the main package beacuse lakka no have a platform for install packages without recompile all.

Realy Lakka is a linux system with ssh but whitout rsync o package manager? how to install privative drivers for specific old laptop?, how to install specific tools for remote administration?, how configure a custom output audio? realy need remount the filesystem and rebuild all into the root directories?, realy use a root user without unprivilegied process with sudo and sudoers? 

Lakka is a decapited linux system.

gianfma, 2020-12-11:
> The problem: need a package. The solution: Install a package but need install other package for install the packages, but can not install the main package beacuse lakka no have a platform for install packages without recompile all.

I was looking for the same thing - use rsync on Lakka. I can't try it since I use debian, but if somebody using Arch compiles a binary static of Rsync, would it work on Lakka? I think it should.

WHK102, 2020-12-11:
I tried uploading the libraries and the rsync binary from ubuntu but it caused many conflicts with the existing libraries.

gianfma, 2020-12-11:
> I tried uploading the libraries and the rsync binary from ubuntu but it caused many conflicts with the existing libraries.

I think the only way to go is to compile a bianry static. That's meant to avoid the dependencies. I got the idea from here: https://community.broadcom.com/groups/communities/community-home/librarydocuments/viewdocument?DocumentKey=028d2d43-d20d-4a24-af1d-7b346137a72f

I downloaded the binary from there and copied it to /usr/bin, but I didn't manage to get it to run. Maybe you can try yourself and see if you manage to get it to work. Please let me know if you have better results than me.

WHK102, 2020-12-15:
I have surfed the depths of the internet, asked linux elders for advice, and managed to get rsync to work on lakka.

The first thing to consider is that Lakka uses a file system called squashfs in read-only mode, this is mounted in a temporary directory in memory and makes the system work, in this way it maintains its integrity but also makes it difficult to make changes to the file. File System.

I use Ubuntu so the task is much easier for me, we must remember that the core of Lakka is Linux with LibreELEC.

## Step 1: Modify the file system.

We assume that we have a Lakka USB connected to the pc. The USB is mounted on `/media/whk/LAKKA/` and `SYSTEM` file is the squashfs image.

The first thing we need to do is unpack the contents of the file system in any directory on the pc:

```
whk@machine:~$ cd Downloads
whk@machine:Downloads$ mkdir lakka && cd lakka
whk@machine:Downloads/lakka$ sudo su
root@machine:Downloads/lakka# unsquashfs /media/whk/LAKKA/SYSTEM
Parallel unsquashfs: Using 8 processors
30058 inodes (36913 blocks) to write

[===========================================================/] 36913/36913 100%

created 29467 files
created 921 directories
created 591 symlinks
created 0 devices
created 0 fifos
```

We now have the lakka filesystem in our download folder:

```
root@machine:Downloads/lakka# ls -lah squashfs-root/
total 48K
drwxrwxr-x 12 whk whk 4,0K ene 21  2020 .
drwxrwxr-x  3 whk whk 4,0K dic 14 22:25 ..
lrwxrwxrwx  1 whk whk    8 ene 21  2020 bin -> /usr/bin
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 dev
drwxrwxr-x 18 whk whk 4,0K dic 14 22:50 etc
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 flash
lrwxrwxrwx  1 whk whk    8 ene 21  2020 lib -> /usr/lib
lrwxrwxrwx  1 whk whk    8 ene 21  2020 lib64 -> /usr/lib
lrwxrwxrwx  1 whk whk   10 ene 21  2020 media -> /var/media
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 proc
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 run
lrwxrwxrwx  1 whk whk    9 ene 21  2020 sbin -> /usr/sbin
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 storage
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 sys
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 tmp
drwxrwxr-x  9 whk whk 4,0K ene 21  2020 usr
drwxrwxr-x  2 whk whk 4,0K ene 21  2020 var
```

Be careful, you should not use the symbolic links or you will end up in the file system of your pc, you should use the physical files only.

## Step 2: Edit the file system.

Now we need to edit the profiles file that `ash` uses (because lakka doesn't use `bash`):

```
root@machine:Downloads/lakka# cat squashfs-root/etc/profile
...
# read config files
  for config in /etc/profile.d/*; do
    if [ -f "$config" ] ; then
      . $config
    fi
  done
```

[... 167 lines cut ...]
             33 100%   32.23kB/s    0:00:00 (xfr#1, to-chk=524/550)
.cache/connman/ethernet_68b599e59065_cable/
.cache/connman/ethernet_68b599e59065_cable/data
          4,096 100%    3.91MB/s    0:00:00 (xfr#2, to-chk=514/550)
.cache/connman/ethernet_68b599e59065_cable/settings
            185 100%  180.66kB/s    0:00:00 (xfr#3, to-chk=513/550)
.cache/cores/
.cache/cores/core.!usr!bin!retroarch.1607533717.375
     90,550,272 100%   44.77MB/s    0:00:01 (xfr#4, to-chk=509/550)
.cache/cores/core.!usr!bin!retroarch.1607533841.572
     89,432,064 100%   27.40MB/s    0:00:03 (xfr#5, to-chk=508/550)
.cache/cores/core.!usr!bin!retroarch.1607989955.390
...
system/mame2016/ini/ui.ini
          1,701 100%   44.90kB/s    0:00:00 (xfr#364, to-chk=0/550)
thumbnails/

sent 7,515 bytes  received 1,707,051,104 bytes  9,563,353.61 bytes/sec
total size is 2,498,912,969  speedup is 1.46
Finish.
```

Now you can install any binary with its dependencies directly on the filesystem of the writable partition. You can also use filezilla to upload the files, just keep in mind that executable files and libraries must have execution permissions.

Using lakka with rsync, The future is now! https://img.memecdn.com/this-woman-deserves-a-nobel-prize_o_2572665.jpg

cbluth, 2020-12-15:
@WHK102 great work, thanks!

gianfma, 2020-12-15:
> # How to install rsync on Lakka

Awesome! Thanks for sharing. I'm an Ubuntu user as well, goingo to try to set it up soon. Out of curiosity, did you get a chance to try a static version of Rsync first?

WHK102, 2020-12-15:
> > # How to install rsync on Lakka
> 
> Awesome! Thanks for sharing. I'm an Ubuntu user as well, goingo to try to set it up soon. Out of curiosity, did you get a chance to try a static version of Rsync first?

Not because that required recompiling each binary. It must be remembered that Lakka is available for many types of architecture and systems, so trying to recompile each tool for each case and for each architecture can be a very difficult task.

beadon, 2022-10-09:
I understand the need for squashfs, but it would really be nice to have a package manager onboard to add things on-the-fly, rather than rebuilding / reflashing the image each time.  OpenWRT uses a 'thin' package manager 'opkg', perhaps this would be a nice way to add pre-compiled packages to systems on-the-fly ?

yhojann-cl, 2022-10-10:
> I understand the need for squashfs, but it would really be nice to have a package manager onboard to add things on-the-fly, rather than rebuilding / reflashing the image each time. OpenWRT uses a [...]

The problem is that lakka is a linux system that is designed for windows users, so it lacks any basic tools of any gnu/linux distribution.