libretro/Lakka-LibreELEC #290: ODROID XU4: More recent kernel

naude-r, 2017-11-24:
hi,

hardkernel supports both a 4.9 LTS and 4.14 LTS kernel for the XU4. would it be possible to support a newer kernel instead of the 3.10 kernel currently used?

https://github.com/hardkernel/linux/tree/odroidxu4-4.9.y
https://github.com/hardkernel/linux/commits/odroidxu4-4.14.y

Arnochocobo, 2017-11-24:
Those 2 kernels have TONS of issues. Latest doesn't mean it is better, especially with hardkernel products. Get some facts before requesting to implement broken stuff, thanks.

naude-r, 2017-11-24:
thank you for the feedback.

closed 2017-11-24 by naude-r

Arnochocobo, 2017-11-25:
No problem. I have a XU4 too, and it is just really disappointing to see how bad the support is for this board. Closed source proper MALI drivers, and so on... The list is really long. This is what happens when you are more expensive than the RPI... Less people around to make magic happen, and HK is not helping at all neither. But Kivutar did a great job with Lakka on the XU4, so, at least we have this until the next board they release. (hopefully with good Vulkan support)