libretro/Lakka-LibreELEC #294: Regression: Video/USB issues on Allwinner H3 SoCs

diizzyy, 2017-11-26:
Hardware: OrangePi PC
Lakka: version 2.1-rc6
Random video corruption, doesn't occur on 2.0
USB is twitchy and responding sporadically

dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.13.12 (kivutar@Server01) (gcc version 6.2.0 (GCC)) #1 SMP Wed Nov 8 19:30:49 CST 2017
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Xunlong Orange Pi PC
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x79c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c0d62600, node_mem_map ef7fa000
[    0.000000]   Normal zone: 1536 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:31
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 16 pages/cpu @ef7a2000 s35084 r8192 d22260 u65536
[    0.000000] pcpu-alloc: s35084 r8192 d22260 u65536 alloc=16*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260608
[    0.000000] Kernel command line: console=ttyS0,115200 console=tty0 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 consoleblank=0 quiet root=/dev/ram0 rdinit=/init
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 960668K/1048576K available (7168K kernel code, 397K rwdata, 1632K rodata, 3072K init, 275K bss, 22372K reserved, 65536K cma-reserved, 196600K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc0800000   (8160 kB)
[    0.000000]       .init : 0xc0a00000 - 0xc0d00000   (3072 kB)
[    0.000000]       .data : 0xc0d00000 - 0xc0d63500   ( 398 kB)
[    0.000000]        .bss : 0xc0d65000 - 0xc0da9e34   ( 276 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[    0.000007] sched_clock: 56 bits at 24MHz, resolution 41ns, wraps every 4398046511097ns
[    0.000016] Switching to timer-based delay loop, resolution 41ns
[    0.000188] Console: colour dummy device 80x30
[    0.000202] console [tty0] enabled
[    0.000224] Calibrating delay loop (skipped), value calculated using timer frequency.. 48.00 BogoMIPS (lpj=240000)
[    0.000234] pid_max: default: 32768 minimum: 301
[    0.000351] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000362] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000856] CPU: Testing write buffer coherency: ok
[    0.001149] /cpus/cpu@0 missing clock-frequency property
[    0.001184] /cpus/cpu@1 missing clock-frequency property
[    0.001210] /cpus/cpu@2 missing clock-frequency property
[    0.001235] /cpus/cpu@3 missing clock-frequency property
[    0.001251] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.001544] Setting up static identity map for 0x40100000 - 0x40100060
[    0.001639] Hierarchical SRCU implementation.
[    0.002106] smp: Bringing up secondary CPUs ...
[... 363 lines cut ...]
[   18.803691] Mali:
[   18.803695] Mali device driver loaded
[   18.965723] systemd-journald[174]: Received request to flush runtime journal from PID 1
[   20.258746] Bluetooth: Core ver 2.22
[   20.258832] NET: Registered protocol family 31
[   20.258838] Bluetooth: HCI device and connection manager initialized
[   20.258852] Bluetooth: HCI socket layer initialized
[   20.258861] Bluetooth: L2CAP socket layer initialized
[   20.258982] Bluetooth: SCO socket layer initialized
[   20.270007] usbcore: registered new interface driver btusb
[   20.405091] Bluetooth: hci0: BCM: chip id 63
[   20.407294] Bluetooth: hci0: BCM: features 0x07
[   20.429105] Bluetooth: hci0: BCM20702A
[   20.431261] Bluetooth: hci0: BCM20702A1 (001.002.014) build 0000
[   20.432496] bluetooth hci0: Direct firmware load for brcm/BCM20702A1-0b05-17cb.hcd failed with error -2
[   20.432512] Bluetooth: hci0: BCM: Patch brcm/BCM20702A1-0b05-17cb.hcd not found
[   20.635855] Generic PHY stmmac-0:01: attached PHY driver [Generic PHY] (mii_bus:phy_addr=stmmac-0:01, irq=-1)
[   20.638077] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   20.638094] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   20.638407] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   23.759608] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   23.759649] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   23.769971] 8021q: 802.1Q VLAN Support v1.8
```
[lakka-regression.zip](https://github.com/libretro/Lakka-LibreELEC/files/1504067/lakka-regression.zip)

diizzyy, 2017-11-27:
Cold boot, same video issue as before... Rebooted (and changed RetroArch resolution to 720p) and it's gone. This issue seems to be intermittent unfortunately.

Tom-Neverwinter, 2017-12-03:
control usage still blows.... the "autodetect" feature is a horrendous feature... /deletes the "autoconfig"... manually sets up.

kivutar, 2018-02-12:
I merged Ntemis PR for H3. My DS4 controller works fine on it.

diizzyy, 2018-06-06:
@kivutar 
I tried the newest nightly build available (http://nightly.builds.lakka.tv/2018-05-28/) and all graphical glitches are still there, something you can verify?

diizzyy, 2018-07-28:
Compatibility issue, closing

closed 2018-07-28 by diizzyy