libretro/Lakka-LibreELEC #295: Lakka 2.1 generi  x86_64 no video image on hdmi

dropy2008, 2017-11-26:
This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

BEFORE you report a bug make sure you have tried the latest nightly version of Lakka at http://le.builds.lakka.tv/. Your bug might be already fixed.

If you are sure that it's a bug in Lakka, please answer the following questions:
- Which version of Lakka are you using?
- What system hardware are you using?
- What did you do?
- What did you expect to happen?
- What happened instead?

It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka/

Please delete this text before posting.

dropy2008, 2017-11-26:
Hp g 60 nf300ea gpu 1gb ram nvidia hdmi is the only device connected but no image.
P.S. LVDS IS MISSING IS JUST THE G60 MOTHERBOARD CONECTED TO TV
PLs
 HELP

gouchi, 2017-12-03:
[Source](https://forums.libretro.com/t/lakka-generic-x86-x64-2-1-rc5-no-hdmi-video-output/13125)

closed 2017-12-04 by Ntemis