libretro/Lakka-LibreELEC #296: Lakka 2.1 x86_64 -> No wifi (Asus VivoPC 1007U or VM40B)

godorowski, 2017-11-27:
In version 2.0 of lakka Wifi works perfectly, but on 2.1 RC6 and 2.1 Stable Wifi cant find any network or find one wifi with random ascii character.  

Wifi card is Realtek RTL8XXX.

godorowski, 2017-11-27:
Ill add that version is updated from 2.0 Stable.

gouchi, 2017-11-29:
Hi,

It is written in the [issue template](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/ISSUE_TEMPLATE.md)

`It is important to provide logs for debugging as part of your report. Instructions for generating logs can be found in the Troubleshooting Lakka documentation at http://www.lakka.tv/doc/Troubleshooting-Lakka`

Thank you.

godorowski, 2017-11-29:
[INFO] RetroArch 1.6.9 (Git c056726)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4 SSE4.2 
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/retroarch.glslp"
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 3 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: yes
[INFO] [DRM]: Connector 1 has 28 modes.
[INFO] [DRM]: Connector 1 assigned to monitor index: #1.
[INFO] [DRM]: Connector 2 connected: no
[INFO] [DRM]: Connector 2 has 0 modes.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 6: (1680x1050) 1680 x 1050, 60 Hz
[INFO] [DRM]: Mode 7: (1400x1050) 1400 x 1050, 60 Hz
[INFO] [DRM]: Mode 8: (1600x900) 1600 x 900, 60 Hz
[INFO] [DRM]: Mode 9: (1280x1024) 1280 x 1024, 75 Hz
[INFO] [DRM]: Mode 10: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 11: (1440x900) 1440 x 900, 60 Hz
[INFO] [DRM]: Mode 12: (1280x800) 1280 x 800, 60 Hz
[INFO] [DRM]: Mode 13: (1152x864) 1152 x 864, 75 Hz
[INFO] [DRM]: Mode 14: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 15: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 16: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 17: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 18: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 19: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 20: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 21: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 22: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 23: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 24: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 25: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 26: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 27: (720x400) 720 x 400, 70 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7f3ac8136160.
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [GL]: Vendor: Intel Open Source Technology Center, Renderer: Mesa DRI Intel(R) Ivybridge Mobile .
[INFO] [GL]: Version: 3.0 Mesa 17.1.3.
[INFO] [GL]: Using resolution 1920x1080
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[INFO] Found #pragma parameter Sharpness Image (SHARPNESS_IMAGE) 1.000000 1.000000 5.000000 1.000000
[... 65 lines cut ...]
[INFO] Found #pragma parameter Brightness (BRIGHTNESS) 1.500000 0.000000 2.000000 0.050000
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] [Font]: Using font rendering backend: freetype.
[INFO] SRAM will not be saved.
[INFO] Found #pragma parameter Sharpness Image (SHARPNESS_IMAGE) 1.000000 1.000000 5.000000 1.000000
[INFO] Found #pragma parameter Sharpness Edges (SHARPNESS_EDGES) 3.000000 1.000000 5.000000 1.000000
[INFO] Found #pragma parameter Glow Width (GLOW_WIDTH) 0.500000 0.050000 0.650000 0.050000
[INFO] Found #pragma parameter Glow Height (GLOW_HEIGHT) 0.500000 0.050000 0.650000 0.050000
[INFO] Found #pragma parameter Glow Halation (GLOW_HALATION) 0.100000 0.000000 1.000000 0.010000
[INFO] Found #pragma parameter Glow Diffusion (GLOW_DIFFUSION) 0.050000 0.000000 1.000000 0.010000
[INFO] Found #pragma parameter Mask Colors (MASK_COLORS) 2.000000 2.000000 3.000000 1.000000
[INFO] Found #pragma parameter Mask Strength (MASK_STRENGTH) 0.300000 0.000000 1.000000 0.050000
[INFO] Found #pragma parameter Mask Size (MASK_SIZE) 1.000000 1.000000 9.000000 1.000000
[INFO] Found #pragma parameter Scanline Size Min. (SCANLINE_SIZE_MIN) 0.500000 0.500000 1.500000 0.050000
[INFO] Found #pragma parameter Scanline Size Max. (SCANLINE_SIZE_MAX) 1.500000 0.500000 1.500000 0.050000
[INFO] Found #pragma parameter Gamma Input (GAMMA_INPUT) 2.400000 1.000000 5.000000 0.100000
[INFO] Found #pragma parameter Gamma Output (GAMMA_OUTPUT) 2.400000 1.000000 5.000000 0.100000
[INFO] Found #pragma parameter Brightness (BRIGHTNESS) 1.500000 0.000000 2.000000 0.050000
[INFO] Loading history file: [/storage/.config/retroarch/content_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_favorites.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_music_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_video_history.lpl].
[INFO] Loading history file: [/storage/.config/retroarch/content_image_history.lpl].
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

godorowski, 2017-11-29:
lscpi gives me this about my wifi

02:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8821AE 802.11ac PCIe Wireless Network Adapter

gouchi, 2017-11-29:
And `dmesg` output ?

Thank you.

godorowski, 2017-11-29:
Got update.

After fiddling in settings I made even lan connection impossible so I made reinstall from fresh 2.1 x86 64 iso. After clean install Wifi is working again.

I think this problem occured during update from 2.0 -> 2.1.  I had then configured WiFi. 

Thanks for your support!

closed 2017-11-30 by gouchi