libretro/Lakka-LibreELEC #299: Banana Pi Lakka 2.1 Problem with update or clear install

isanphone, 2017-11-29:
Lakka-Allwinner.Bananapi.arm-2.1.img.gz

after update log:
```
Lakka-Allwinner.Bananapi.arm-2.1.img.gz is not compatible with bananapi.arm hardware
update system:allwiner.bananapi.arm
```
If install as normal problem with kernel.
Lakka 2 install with no problems.

Ntemis, 2017-11-29:
Yeah there is a change in arch because we switched to mainline kernel and u-boot.
You can update if you place a empty file named .nocompat next to the new lakka image

closed 2017-11-29 by Ntemis

isanphone, 2017-11-30:
Update done, thanks! But now, banana Pi have not load lakka at all. And if I do clear install img at sd card, lakka says about problem with kernel. Maybe fake img or something?