libretro/Lakka-LibreELEC #300: Lakka 2.1 cannot boot on cubietruck

CliffHan, 2017-12-04:
Downloaded lakka 2.1 (Lakka-Allwinner.Cubietruck.arm-2.1.img.gz) from lakka.tv, and extracted without error.
Used etcher to flash it on a 8gb tf (sdhc) card without error. And connected a Logitech F710 usb receiver to the cubietruck. Connected the board to tv with hdmi.
When the board boot, I can see nothing on the screen. I waited a long time and it's keeping black.

Kivutar, 2017-12-10:
Well, I just tested it now, and it boots for me. I can't reproduce.

closed 2017-12-10 by Ntemis

Matt1234321, 2018-12-21:
I have the same issue with 2.1.1

kacherga, 2019-01-05:
I confirm it does not work on the CUBETRUCK! neither 2.1 nor 2.2 - black screen!

CliffHan, 2019-02-22:
I just tested 2.2 and it cannot work either, the phenomenon was same as 1st comment.

LennyLip, 2020-01-31:
it seems it's hdmi/vga issue related.