libretro/Lakka-LibreELEC #307: Segfault in allwinner kernel

Kivutar, 2017-12-10:
> - Which version of Lakka are you using?

Post 2.1

> - What system hardware are you using?

Cubietruck

> - What did you do?

Boot

> - What did you expect to happen?
> - What happened instead?

Kernel or module segfault

Kivutar, 2017-12-14:
The fix doesn't work:

```
Lakka:~ # lsmod
Module                  Size  Used by
8021q                  24576  0
brcmfmac              188416  0
brcmutil               16384  1 brcmfmac
cfg80211              245760  1 brcmfmac
overlay                57344  5
Lakka:~ # modprobe mali
modprobe: ERROR: could not insert 'mali': No such device
Lakka:~ # Connection to 192.168.1.120 closed.
```

ToKe79, 2017-12-15:
Allwinner does not compile now:
```
  HOSTCC  scripts/extract-cert
scripts/extract-cert.c:21:25: fatal error: openssl/bio.h: No such file or directory
 #include <openssl/bio.h>
                         ^
compilation terminated.
make[2]: *** [scripts/Makefile.host:102: scripts/extract-cert] Error 1
make[1]: *** [Makefile:562: scripts] Error 2
```

sybnex, 2018-02-11:
Same on Cubieboard2. It doesn't boot and hangs like in Issue 317.

As I'm using Lakka the first time, is there a repo with the 2.0 images to try them out?

kivutar, 2018-02-12:
Now H3 is fixed

mfwhite, 2018-03-26:
Hello friends, the image for Banana Pi is in trouble, when giving the boot appears the following message "Failed to start load Kernel modules".

Thermionix, 2018-07-20:
the 2.1 images are missing the mali node in the sun7i device trees, the mali DRM module fails to load with `kernel: couldn't find the mali node`

I attempted to backport the patch however the new code utilises sunxi-ng `dt-bindings/clock/sun7i-a20-ccu.h` instead of sunxi `dt-bindings/clock/sun4i-a10-pll2.h` which would require a fair bit of grafting

I've compiled images using the docker image via `docker run -it -v $(pwd):/root -e DISTRO=Lakka -e PROJECT=Allwinner -e SYSTEM=Linksprite_pcDuino3_Nano -e ARCH=arm lakka` and tested on my hardware.

I removed most of the old allwinner linux kernel patches as many have been merged to mainline

Thermionix, 2018-07-20:
I need some help testing the changes on other a20 / H3 boards

kivutar, 2018-07-22:
Please post your image there and somebody will test them. I have some a20 and h3 here.

aymericDD, 2018-11-03:
I have the same issue with my a20... I hope we will find a solution...

roger-, 2019-01-13:
Same problem here with A20.

ezaul, 2019-02-17:
cubieboard2 failed to Start Load Kernel Modules
# lsmod                        
Module          Size      Used  by
 8021q           24576   0
overlay           57344  5

kivutar, 2019-02-18:
Yeah, an issue with the mali node

CliffHan, 2019-02-25:
sirs, any progress?

aymericDD, 2019-02-25:
I have tried all builds for Cubieboard v2 and none works...

ToKe79, 2019-07-06:
New images based on LibreELEC 9.2 (beta) are [here](http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/) for testing.
Source: https://github.com/libretro/Lakka-LibreELEC/tree/Lakka-LE9.2

aymericDD, 2019-07-14:
@ToKe79 I don't find the image for Cubieboard 2. 😕

ToKe79, 2019-07-14:
I added image with cubieboard2 dtb (http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm).

ToKe79, 2019-07-14:
I also added images for:
- bananapi
- cubietruck
- nanopi_m1_plus
- orangepi_lite
- orangepi_one

aymericDD, 2019-07-20:
It does not work on my cubieboard 2. Lakke does not boot. I have the following error messages at Cubieboard:
```
retriving file: /extlinux/pxelinux.cfg/default
ethernet@1c50000 Waiting for PHY auto negotation to complete........ TIMEOUT
Could not initializer PHY ethernet@1c50000
```
The value of `retriving file: /extlinux/pxelinux.cfg/default` change each times. It could be `/extlinux/pxelinux.cfg/default, `/extlinux/pxelinux.cfg/default-arm`, `/extlinux/pxelinux.cfg/default-arm-sunxi`.

And before each error if have this error: `Lakka for failure retrieving fdt`

Ioalevia, 2019-09-29:
banana pi a20. tried all latest versions of images. tried nightly versions. two varians: stucks with kernel fail or has ethernet fail. one year ago i used lakka on my device and it worked perfect. if anybody has lakka image for banana pi from 2016-2017????

Ntemis, 2019-09-29:
@Ioalevia stop already,really!

koozkaa, 2019-10-13:
Hi Guys, @ToKe79 what about cubieboard 1? Could you please also add image for a 10 cubieboard 1?

linnea-s, 2019-10-26:
Hi, I have a Cubieboard3 gathering dust. If one of the Lakka devs would like it to improve support, I would be happy to donate it.

Ntemis, 2019-10-27:
You can install pandora box operating system on it if you want

kindalo, 2019-11-01:
Hi @ToKe79,
> I added image with cubieboard2 dtb (http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm).
```
Hi @aymericDD,

> It does not work on my cubieboard 2. Lakka does not boot. I have the following error messages at boot:
> ![IMG_20190721_134329](https://user-images.githubusercontent.com/8859832/61590778-29818200-abbe-11e9-8daf-a3ce22023bef.jpg)
> ![IMG_20190721_134415](https://user-images.githubusercontent.com/8859832/61590781-2be3dc00-abbe-11e9-97d9-e4925fb8d5eb.jpg)

kindalo, 2019-11-01:
Hi @ToKe79,
> I added image with cubieboard2 dtb (http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm).
I test the image from the link on my cubieboard2 and it looks like the fdt fail (**sun7i-a20-cubieboard2.dtb**) when retrieving. Could you correct that?
```
=> run bootcmd_mmc0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /extlinux/extlinux.conf
Retrieving file: /extlinux/extlinux.conf
177 bytes read in 8 ms (21.5 KiB/s)
1:      Lakka
Retrieving file: /KERNEL
8267752 bytes read in 454 ms (17.4 MiB/s)
append: boot=UUID=1709-4623 disk=UUID=fcc9a14b-b8ed-4a3c-b73b-5788fb6c53a6 quiet console=ttyS0,115200 console=tty1
Retrieving file: /sun7i-a20-cubieboard2.dtb
Skipping Lakka for failure retrieving fdt
SCRIPT FAILED: continuing...
```
Hi @aymericDD,
I try it on my cubieboard2 too and same problem here. Cause the boot image in SD is incorrect and it's trying to try other boot options. In your case, it try to boot from networking.
> It does not work on my cubieboard 2. Lakka does not boot. I have the following error messages at boot:
> [image]
> [image]

kindalo, 2019-11-14:
> Hi @ToKe79,
> 
> > I added image with cubieboard2 dtb (http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm).
> 
> I test the image from the link on my cubieboard2 and it looks like the fdt fail (**sun7i-a20-cubieboard2.dtb**) when retrieving. Could you correct that?
> 
I added the **sun7i-a20-cubieboard2.dtb** in SD card and can't boot after loading kernel in uboot. I used the image here (http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm).

And I used the [commit](https://github.com/libretro/Lakka-LibreELEC/pull/504/commits/63d267f4e405402aae788cce583e47930ca730f8) from @Thermionix and compile in lakka-2.3.1 and I can get into the UI on my **cubieboard2** now.
Now I can move forward in the project. Thanks @Thermionix !

Ntemis, 2019-11-14:
then probably is kernel config related @kindalo 
@ToKe79 can you adjust current kernel config and release an image for him to test?

kindalo, 2019-11-14:
> then probably is kernel config related @kindalo
> @ToKe79 can you adjust current kernel config and release an image for him to test?
For **cubieboard2**
Based on lakka-2.3.1, I back-ported some code in dtb and make mali node could be found in kernel and I got another error when set mali clock defaults and I think that might be other problem in kernel and mali driver.

Ntemis, 2019-11-14:
Most possibly these issues can be resolved when we migrate to lima but we can't support boards we dont own. Feel free to sent a pr and if you need help come and join us on discord channel.

kindalo, 2019-11-18:
Hi @Ntemis,
I am not sure which way you prefer to fix in Allwinner projects. The new one "4.17.8" or old one "4.14.18" (base on lakka-2.3.1 & lakka 2.1)? Let me know and try to provide PR.

johndoe71rus, 2019-12-03:
Yesterday I boot the Lakka-Allwinner.Cubieboard2.arm-2.3.1.img.gz image. An error in the console that the kernel module cannot load. Then successful messages [ok] but does not load into the UI.
on the flash drive did not find boot logs

murakas, 2019-12-21:
Hi, I also own a banana pi. I can also participate in testing.

WizzardSK, 2020-03-23:
Where can we find Pandora Box operating system?

closed 2020-08-18 by Ntemis