libretro/Lakka-LibreELEC #309: Mayflash DolphinBar/wiimote B/A button not firing

jedvargo, 2017-12-13:
Log file: https://pastebin.com/hZSCtbE3
version: Lakka Generic_x86_64-2.1
System Hardware: Toshiba Satellite C55-B5299
Issue: B, A mouse click not firing in MAME2003 or other cores

    After I syncing the controller with the dolphinbar, in mode 1 or mode 2, I can cat the input of the device events from the cmd line and I see the controller movement and the B, A clicks registering, so I know I’m getting input from the controller on the system level. I’ve matched the index of the mouse to the correct device event.
    I start MAME2003 with Crossbow or star wars and the B and A buttons do not fire, but the crosshairs will work a-ok when I move the controller around. When I do push either of the two buttons it will cause the crosshairs to pause momentarily but they don’t do anything else. I’ve tried to configure those button events in the MAME config menu but they aren’t seen there either.
    I removed the usb key and the dolphinbar from the laptop I was using and booted from a different laptop and had the same results. 
    I installed retroarch from the deb repository on a different linux desktop (Kubuntu 17.10) and tested with the same Crossbow rom and mame2003 core. I used the same dolphinbar and wiimote. I noticed the crosshairs also paused when I pushed B, but this time the shot fired.

gouchi, 2017-12-16:
I see that there is a [PR for Mayflash Dolphin bar for RA](https://github.com/libretro/RetroArch/pull/5882) not sure if it will fix your issue.

Did you try to change your joypad driver to hid or linuxraw ?

Also can you provide the output of  
`lsusb`

Thank you.

markwkidd, 2017-12-24:
Have you tried using the Dolphinbar in Mode 3? I don't have mine with me now but if I recall that mode has worked fine for me in the past with MAME 2003.

jedvargo, 2017-12-24:
output from lsusb:

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 006: ID 0079:1802 DragonRise Inc. 
Bus 002 Device 009: ID 0930:0227 Toshiba Corp. 
Bus 002 Device 003: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 007: ID aa55:0101  
Bus 002 Device 005: ID d209:15a2 Ultimarc 
Bus 002 Device 004: ID 1241:1122 Belkin Typhoon Stream Optical Mouse USB+PS/2
Bus 002 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

I've also tried the other joypad drivers with the same result. 

I've discovered an important bit of information. I installed Ubuntu Mate on the laptop, tested RetroArch in X windows and the wiimote A/B buttons work a-ok. When I change grub to boot to console and then run retroarch from the command line I encounter the same problem as before.

gouchi, 2017-12-29:
On Lakka we are not using Xorg for Generic platform. 

Can you check that on your Ubuntu Mate if you have for input driver X org udev ?

Did you try to use Dolphingbar in mode 3 as @markwkidd  suggested ?

Thank you.

ToKe79, 2019-07-02:
no response

closed 2019-07-02 by ToKe79