libretro/Lakka-LibreELEC #312: Lakka bootloop

Alexius77, 2017-12-15:
Hello, I have a bootloop after flashing lakka to an usb-drive.
- Which version of Lakka are you using?
2.1 stable i386 and x64
- What system hardware are you using?
Usb kingston 101, cpu amd athlon II 170u, gpu integrated radeon 2100, 3 gb ddr2 800 mhz
- What did you do?
Nothing, just inserted usb flash drive into pc
- What did you expect to happen?
That Lakka should start
- What happened instead?
Bootloop on lakka berry image

gouchi, 2017-12-16:
Please boot with 
`live textmode`

And provide some [Lakka system and graphic logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

Alexius77, 2017-12-16:
[image]
I've entered the commands, what should I do next?

Alexius77, 2017-12-16:
i don't have internet connection or lan on pc running lakka and none of three linux fs drivers work in my windows 10

Alexius77, 2017-12-16:
at last here it is
[log2.txt](https://github.com/libretro/Lakka-LibreELEC/files/1565177/log2.txt)

gouchi, 2017-12-18:
Same [issue](https://forums.libretro.com/t/stuck-at-first-time-boot-screen-infinite-reboot-loop/10599) with R300 graphic card family. Yours is R400.

`Unable to find target for this triple (no targets are registered)`

It seems an issue related to llvm.

Alexius77, 2017-12-18:
So, what's the solution then?

gouchi, 2017-12-19:
I don't know yet the solution.

gouchi, 2018-06-02:
May you  try to do some test with our [nightly builds](http://nightly.builds.lakka.tv/) and provides some log ?

Alexius77, 2018-06-03:
oh, well, i can try

gouchi, 2018-06-03:
Please check [this comment](https://github.com/libretro/Lakka-LibreELEC/issues/312#issuecomment-352193187).

Alexius77, 2018-06-06:
[log.txt](https://github.com/libretro/Lakka-LibreELEC/files/2078460/log.txt)
here is the log from that build

gouchi, 2018-06-07:
Unfortunately, you got a [segfault](https://en.wikipedia.org/wiki/Segmentation_fault).

Can you try to use the rgui menu driver ? Boot with `live textmode` then

```
sed -i -e "s/menu_driver = \"xmb\"/menu_driver = \"rgui\"/g" .config/retroarch.cfg
retroarch -v
```

Alexius77, 2018-06-07:
sed: .config/retroarch.cfg: No such file or directory

ToKe79, 2018-06-07:
should be
```
.config/retroarch/retroarch.cfg
```

Alexius77, 2018-06-07:
[image]
ls says that there is no config directory

ToKe79, 2018-06-07:
it is a hidden folder (starting with a dot), so you need `ls -a`. absolute path is `/storage/.config/retroarch/retroarch.cfg`. i just checked on my s912 box.

Alexius77, 2018-06-07:
done, again this segmentation fault

Alexius77, 2018-06-07:
by the way, what causes this segfault in general - cpu/ram/hdd?

gouchi, 2018-06-07:
Please provide the log with rgui menu.

Alexius77, 2018-06-08:
[logrgui.txt](https://github.com/libretro/Lakka-LibreELEC/files/2085054/logrgui.txt)

gouchi, 2018-06-09:
Same error with rgui menu driver

```
Unable to find target for this triple (no targets are registered)
Segmentation fault (core dumped)
```
We need to search/find a fix for this issue as I don't know how to fix it currently.

gouchi, 2018-06-16:
Can you try to launch RA with

`DRAW_USE_LLVM=0 retroarch -v`

Thank you.

Alexius77, 2018-06-17:
yes, it worked - the system has started with rgui driver, but i don't know whether it is an error or not - when i press ESC in any screen, the system quits to the textmode, and when I try to connect to wi-fi and enter the password, nothing happens - so I don't know works my connection or not
and my USB flash drive doesn't look like working properly - when I connect it to PC, Windows doesn't allow to format it in any way :( and when I try to view it in Control Panel - Administrative Tools - Computer Management, the system hangs

gouchi, 2018-06-17:
It is normal when you press Esc that RA closed as you started it manually.

You may try to [setup the network with  conmannctl](http://www.lakka.tv/doc/Network-settings/).

It is normal you don't see the usb stick as the partition format is [not recognized natively on Windows](http://www.lakka.tv/doc/Accessing-Lakka-filesystem/#direct-drive-access).

Alexius77, 2018-06-17:
no, I mean it wasn't recognized before, but I could easily format it into NTFS or smth and then flash it with a new build of Lakka, now I can't do it at all

gouchi, 2018-06-18:
I will close this issue as the original issue is fixed.

closed 2018-06-18 by gouchi