libretro/Lakka-LibreELEC #313: Unable to (cross) compile for ROCK64

andrebraga, 2017-12-15:
Hi, I'm attempting to compile Lakka under a standard Debian docker environment.

I'm finiding myself stuck at:

```
/root/Lakka-LibreELEC/build.LibreELEC-ROCK64.arm-8.2-devel/kodi-5bd45ab/xbmc/windowing/egl/EGLWrapper.cpp: In function 'bool {anonymous}::CorrectGuess(CEGLNativeType*, const string&)':
/root/Lakka-LibreELEC/build.LibreELEC-ROCK64.arm-8.2-devel/kodi-5bd45ab/xbmc/windowing/egl/EGLWrapper.cpp:63:13: error: invalid use of incomplete type 'class CEGLNativeType'
```

Any pointers for building? I tried both `arm` and `aarch64` and both break at the same point.

`PROJECT=Rockchip DEVICE=ROCK64 ARCH=arm make image`

ToKe79, 2017-12-16:
seems like you are missing  `DISTRO=Lakka` anywhere before `make`

Kivutar, 2017-12-23:
Yep.

closed 2017-12-23 by Kivutar