libretro/Lakka-LibreELEC #314: PS4 Mini Wired Gamepad not working

catzoo, 2017-12-17:
Thank you for developing Lakka. Currently i'm trying to make a present for my Grandparents, they always talked about miss playing old games (such as Tetris and Doctor Mario)

- Which version of Lakka are you using? 
I believe 2.1    (RPi2.arm-2.1)
- What system hardware are you using? 
Raspberry Pi 3
- What did you do? 
Plugged in the controller and tried to bind all the inputs
- What did you expect to happen?
Bind and Save the inputs
- What happened instead?
Did not recognize the inputs and did nothing

I did test the controllers on other computers making sure the controllers aren't broken.
And i'm using this controller: [Link](https://www.playstation.com/en-us/explore/accessories/gaming-controllers/mini-wired-gamepad/)

Thank you for reading

---Input Logs---

Lakka:~ # lsusb
Bus 001 Device 006: ID 0f0d:00ee Hori Co., Ltd
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast                                                                                                     Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

Lakka:~ # dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.9.59 (kivutar@Server01) (gcc version 6.2.0 (GCC)                                                                                                     ) #1 SMP Sun Nov 19 07:05:16 CST 2017
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructio                                                                                                    n cache
[    0.000000] OF: fdt:Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] cma: Reserved 8 MiB at 0x2f800000
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 196608
[    0.000000] free_area_init_node: node 0, pgdat 80c363c0, node_mem_map af13600                                                                                                    0
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:31
[    0.000000] percpu: Embedded 14 pages/cpu @af0f0000 s25420 r8192 d23732 u5734                                                                                                    4
[    0.000000] pcpu-alloc: s25420 r8192 d23732 u57344 alloc=14*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                                                    es: 194880
[    0.000000] Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2                                                                                                    708_fb.fbheight=1080 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_s                                                                                                    ize=0x40000000  boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 quiet  vt.global_cursor_                                                                                                    default=0 loglevel=2 root=/dev/ram0 rdinit=/init BOOT_IMAGE=/kernel.img usbcore.                                                                                                    autosuspend=-1
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 758304K/786432K available (6144K kernel code, 222K rwdata                                                                                                    , 1376K rodata, 3072K init, 394K bss, 19936K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xb0800000 - 0xff800000   (1264 MB)
[    0.000000]     lowmem  : 0x80000000 - 0xb0000000   ( 768 MB)
[    0.000000]     modules : 0x7f000000 - 0x80000000   (  16 MB)
[    0.000000]       .text : 0x80008000 - 0x80700000   (7136 kB)
[    0.000000]       .init : 0x80900000 - 0x80c00000   (3072 kB)
[    0.000000]       .data : 0x80c00000 - 0x80c378a0   ( 223 kB)
[    0.000000]        .bss : 0x80c39000 - 0x80c9ba6c   ( 395 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000]  Build-time adjustment of leaf fanout to 32.
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] arm_arch_timer: Architected cp15 timer(s) running at 19.20MHz (ph                                                                                                    ys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles:                                                                                                     0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 43980                                                                                                    46511078ns
[    0.000016] Switching to timer-based delay loop, resolution 52ns
[    0.000204] Console: colour dummy device 80x30
[    0.000219] console [tty0] enabled
[    0.000234] Calibrating delay loop (skipped), value calculated using timer fr                                                                                                    equency.. 38.00 BogoMIPS (lpj=64000)
[... 216 lines cut ...]
[    8.441011] Bluetooth: HCI UART protocol Three-wire (H5) registered
[    8.441333] Bluetooth: HCI UART protocol Broadcom registered
[   10.504957] Bluetooth: hci0 command 0x0c14 tx timeout
[   10.505401] Bluetooth: hci0: Frame reassembly failed (-84)
[   10.505413] Bluetooth: hci0: Frame reassembly failed (-84)
[   14.520944] random: crng init done
[  301.481993] usb 1-1.2: USB disconnect, device number 4
[  306.082311] usb 1-1.3: USB disconnect, device number 5
[  310.544891] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
[  310.637889] usb 1-1.2: New USB device found, idVendor=0f0d, idProduct=00ee
[  310.637906] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber                                                                                                    =0
[  310.637913] usb 1-1.2: Product: HORIPAD mini4
[  310.637920] usb 1-1.2: Manufacturer: HORI CO.,LTD.
[  310.649225] input: HORI CO.,LTD. HORIPAD mini4 as /devices/platform/soc/3f980                                                                                                    000.usb/usb1/1-1/1-1.2/1-1.2:1.0/0003:0F0D:00EE.0006/input/input2
[  310.650241] hid-generic 0003:0F0D:00EE.0006: input,hiddev0,hidraw0: USB HID v                                                                                                    1.11 Gamepad [HORI CO.,LTD. HORIPAD mini4] on usb-3f980000.usb-1.2/input0
[  753.938122] usb 1-1.4: new full-speed USB device number 7 using dwc_otg
[  754.033509] usb 1-1.4: New USB device found, idVendor=046d, idProduct=c52b
[  754.033530] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber                                                                                                    =0
[  754.033537] usb 1-1.4: Product: USB Receiver
[  754.033545] usb 1-1.4: Manufacturer: Logitech
[  754.052911] logitech-djreceiver 0003:046D:C52B.0009: hiddev0,hidraw1: USB HID                                                                                                     v1.11 Device [Logitech USB Receiver] on usb-3f980000.usb-1.4/input2
[  754.174337] input: Logitech K360 as /devices/platform/soc/3f980000.usb/usb1/1                                                                                                    -1/1-1.4/1-1.4:1.2/0003:046D:C52B.0009/0003:046D:4004.000A/input/input3
[  754.177194] logitech-hidpp-device 0003:046D:4004.000A: input,hidraw2: USB HID                                                                                                     v1.11 Keyboard [Logitech K360] on usb-3f980000.usb-1.4:1
[  776.808114] usb 1-1.4: reset full-speed USB device number 7 using dwc_otg
[  777.479756] usb 1-1.4: USB disconnect, device number 7

ToKe79, 2017-12-17:
HI,
to me it looks like the gamepad is detected, e.g starting from 301 seconds.
You can list the input devices:
```bash
ls /dev/input
```
joysticks / gamepads should be listed as `jsN`, e.g. `js0`. And then try to test the input:
```bash
jstest /dev/input/jsN
```
where the `N` is replaced with a number, most probably `0`. Pressing the D-Pad, buttons, moving the sticks should trigger either `on` for buttons or axis changes for analog sticks/buttons.

catzoo, 2017-12-17:
Hey, and yes I did have a wireless keyboard during that time, I only have one keyboard so I kept plugging it in and unplugging it. I should've said this in the report, sorry about that.
They keyboard is a old Wireless Logitech keyboard, I forgot what type it is.

During this time I only have the controller connected and nothing else. Its weird that it thinks its a mice and not a controller.

I try to test it out and I pushed everything and moved the analogs, and nothing came up.

```
Lakka:~ # ls /dev/input
mice
Lakka:~ # jstest /dev/input/mice
Joystick (Unknown) has 2 axes and 2 buttons. Driver version is 0.8.0.
Testing ... (interrupt to exit)

^?
^C
Lakka:~ #
```
When I tried to stop it, I missed the Ctrl-C at first.

ToKe79, 2017-12-17:
Hm, the gamepad is not listed in `/dev/input`, but according to `dmesg` it was detected. The `mice` device does not have to be related to the gamepad - you can confirm when you unplug the gamepad and do `ls /dev/input` - if `mice` will be still there -> not related to gamepad.

Have you tried navigating in the RetroArch using the keyboard `cursor keys`, `enter` and `backspace`? If navigation with keyboard works, try to get to `Settings` -> `Input` -> `Input User 1 Binds` and check what you have under the item `User 1 Device Index` and try to configure the controls below.

catzoo, 2017-12-17:
I checked to see if Mice is unrelated, I unplugged and plugged it back in and I found the js0.
```
Lakka:~ # ls /dev/input
by-id    by-path  event0   js0      mice
```
I tested out the input on js0 and it all worked.

When I turn the Raspberry Pi on I always have the controller plugged in before turning it on. Whenever I try to bind it, it doesn't work and I can't find it in `Device 1 Index`. After I found out unplugging the controller and plugging it back in detected it, it all started working. I found it in the `Device 1 Index` and I can bind all of it.

Whats weird is it doesn't detect it when its plugged in on bootup, But if you plug it in after its been turned on and on the main screen it all starts working.

gouchi, 2017-12-18:
Hi,

Can you try to give the output with the block related to your game controller 

`cat /proc/bus/input/devices`

Then provide the output of

`udevadm info --name=input/eventX` where X is the number related to your game controller. You can get it from the line starting with `H: Handlers=`

For example, for this output

I: Bus=0003 Vendor=054c Product=0268 Version=8111
N: Name="**Gasia Co.,Ltd PS(R) Gamepad**"
P: Phys=usb-0000:00:1d.1-2/input0
S: Sysfs=/devices/pci0000:00/0000:00:1d.1/usb3/3-2/3-2:1.0/0003:054C:0268.0004/input/input11
U: Uniq=00:26:5c:00:77:f6
H: Handlers=js1 **event12** 
B: PROP=0
B: EV=20001b
B: KEY=f00000000 0 0 0 7fdb000000000000 0 0 0 0
B: ABS=3f
B: MSC=10
B: FF=107030000 

I should enter udevadm info --name=input/**event12**

Thank you.

catzoo, 2017-12-19:
Hey, and sure
```Lakka:~ # cat /proc/bus/input/devices
I: Bus=0003 Vendor=0f0d Product=00ee Version=0111
N: Name="HORI CO.,LTD. HORIPAD mini4"
P: Phys=usb-3f980000.usb-1.3.2/input0
S: Sysfs=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.0/0003:0F0D:00EE.0001/input/input0
U: Uniq=
H: Handlers=js0 event0
B: PROP=0
B: EV=1b
B: KEY=3fff0000 0 0 0 0 0 0 0 0 0
B: ABS=3003f
B: MSC=10
```

and here is the output:
```Lakka:~ # udevadm info --name=input/event0
P: /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.0/0003:0F0D:00EE.0001/input/input0/event0
N: input/event0
S: input/by-id/usb-HORI_CO._LTD._HORIPAD_mini4-event-joystick
S: input/by-path/platform-3f980000.usb-usb-0:1.3.2:1.0-event-joystick
E: DEVLINKS=/dev/input/by-id/usb-HORI_CO._LTD._HORIPAD_mini4-event-joystick /dev/input/by-path/platform-3f980000.usb-usb-0:1.3.2:1.0-event-joystick
E: DEVNAME=/dev/input/event0
E: DEVPATH=/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.3/1-1.3.2/1-1.3.2:1.0/0003:0F0D:00EE.0001/input/input0/event0
E: ID_BUS=usb
E: ID_FOR_SEAT=input-platform-3f980000_usb-usb-0_1_3_2_1_0
E: ID_INPUT=1
E: ID_INPUT_JOYSTICK=1
E: ID_MODEL=HORIPAD_mini4
E: ID_MODEL_ENC=HORIPAD\x20mini4
E: ID_MODEL_ID=00ee
E: ID_PATH=platform-3f980000.usb-usb-0:1.3.2:1.0
E: ID_PATH_TAG=platform-3f980000_usb-usb-0_1_3_2_1_0
E: ID_REVISION=0100
E: ID_SERIAL=HORI_CO._LTD._HORIPAD_mini4
E: ID_TYPE=hid
E: ID_USB_DRIVER=usbhid
E: ID_USB_INTERFACES=:030000:
E: ID_USB_INTERFACE_NUM=00
E: ID_VENDOR=HORI_CO._LTD.
E: ID_VENDOR_ENC=HORI\x20CO.\x2cLTD.
E: ID_VENDOR_ID=0f0d
E: MAJOR=13
E: MINOR=64
E: SUBSYSTEM=input
E: TAGS=:seat:uaccess:
E: USEC_INITIALIZED=3392365
```

After I found out unplugging it and plugging it back in worked, I binded it all and it started to work.
When I start up the raspberry pi with the controller plugged in it worked.

gouchi, 2017-12-19:
Hi,

It is detected as game controller so it is ok. Did you try to push PS button before using it ?

Thank you.

closed 2018-01-31 by ToKe79