libretro/Lakka-LibreELEC #315: Enhancement: Adding the simplest Web Server possible

duduke, 2017-12-17:
As I was trying to find the simplest lightest Web Server for LAKKA to serve some static pages (especially when LAKKA acts as an Access Point). 
I came across: [https://github.com/avleen/bashttpd](url) which uses plain BASH and NETCAT to run the simplest possible web server.

Some basic instructions:
- Clone the github project listed above.
- Run: `chmod +x ./bashttpd`
- Run `./bashttpd` once to create the `bashttpd.conf` file, or create the bashttpd file manually and insert: `serve_file ./index.html`
- Create a dummy `index.html` file. You can use anything.
- Run `:;while [ $? -eq 0 ];do nc -lp 8888 -e ./bashttpd ;done` (note the port number can be changed "8888")
- That's about it. a further enhancement can be to turn this into a systemd service.

This can be used to host an online guide, links to mobile apps etc.

Thanks.

Kivutar, 2017-12-23:
Interesting find, thank you. I appreciate that the server doesn't add any additional dependancy.

If somebody wants to work on integrating this, I would merge it.

ToKe79, 2018-06-27:
I just found out that busybox is configured to provide simple HTTP daemon (httpd). See https://busybox.net/BusyBox.html#httpd for usage.

closed 2018-06-27 by ToKe79