libretro/Lakka-LibreELEC #316: PSP Games not working in RPi3 Lakka 2.1 stable release build

jds40net, 2017-12-19:
Various PSP games run on my RPi3 using Lakka 2.1-rc5. They don’t in the more recent 2.1 stable release (11/26/2017). Video doesn't render with these games. Here are some troubleshooting results:

"Ok, confirmed that the problem is ppsspp_libretro.so of the stabile.

 I solved extracting ppsspp_libretro.so from the RC5 version, I put it in /storage/cores, and I modified the path inside the playlist Sony - PlayStation Portable.lpl replace it from /tmp/cores to storage/cores.

All works perfectly."

Here's a link to the original thread in the lakka forum:

https://forums.libretro.com/t/psp-games-not-working-in-rpi3-lakka-2-1-stable-release-build/13509

Kivutar, 2017-12-23:
@gouchi tested it on x86_64 and the problem doesn't happen there. It may affect only ARM boards.

Kivutar, 2017-12-23:
I confirm this bug. When launching a psp rom, RA will just exit cleanly, without segfault.

kivutar, 2018-01-31:
Fixed by https://github.com/libretro/Lakka-LibreELEC/commit/583afe6afb7f786863fa2961929a71ddf20d0549

closed 2018-01-31 by kivutar

jds40net, 2018-07-23:
The bug below was marked as fixed and the thread "closed":

https://github.com/libretro/Lakka-LibreELEC/issues/316

I have just loaded Lakka-RPi2.arm-2.1.1.img.gz dated 6/25/2018. I'm afraid the bug persists.  With this build, video does not render on various games that run successfully on my RPi3 using Lakka 2.1-rc5.

Please can you investigate and reopen the bug report if needed.  Thanks.

Fixed by 583afe6