libretro/Lakka-LibreELEC #317: Hangs after second boot banana pi/ pro

pcs3rd, 2017-12-19:
(banana pro) first boot works fine, but afterwards, the pro gets stuck at either starting wpa supplicant  or reached target retroarch

aPollO2k, 2018-01-31:
Can confirm this on PBI M1. I will end in one of this two messages each boot.

ToKe79, 2018-01-31:
probably related to #307

vision-4, 2018-01-31:
got exactly same problem : my bpi m1 stuck in second boot on `[ OK ] Reached target Retroarch.` or `[ OK ] Started WPA supplicant.` or `[ OK ] Started Udevil mount service.`

livanh, 2018-04-06:
Same here. No error message like in #307, it just waits forever

CaptainStouf, 2018-04-09:
Same here using Banana Pi A20. It is stuck on `[ OK ] Reached target Retroarch` after first reboot.

Ioalevia, 2019-09-29:
banana pi a20. tried all latest versions of images. tried nightly versions. two varians: stucks with kernel fail or has ethernet fail. one year ago i used lakka on my device and it worked perfect. if anybody has lakka image for banana pi from 2016-2017????

Ntemis, 2019-09-29:
A20 is EOL and no one i know is working on it.
@natinusala decide to keep ot close this down, your call.

natinusala, 2019-09-30:
As long as the platform is listed on the website we should continue to support it or drop it entirely.

Ioalevia, 2019-09-30:
Ntemis. You don't know me. But i use a20 for 4 years. And i think i'm not alone. lakka was the only variant for a20 to create working retrogaming console. Other OS work not so... ideal.... or don't work at all...
Also natinusala is right. banana pi is listed on the webside. And it is exectly banana pi a20. if it is not more supported- please- show it. or make the same as raspberry. There are a lot of banana models, but only a20 is on the website.
but nobody answer the first question- does anybody have lakka image for banana pi from 2016-2017????

natinusala, 2019-09-30:
No, nobody has it sadly

Ioalevia, 2019-09-30:
Even developers? So stop fooling people and delete banana from the list. I can't believe that nobody has  archive of old images. And why developers create latest and nightly versions? For other models? But there only one on the website.

natinusala, 2019-09-30:
Human error happened and we lost all the old images, and due to the very volatile nature of libretro we can't rebuild them.

However you can help us diagnose the issue and fix it if you want

Ioalevia, 2019-09-30:
Issues #317, 307, 308, 504, 475, 510, 382, 232, 360, 543

In my case - #308 failed to start load kernel modules
Stuck on [ok] reached target retroarch.

Is it enough? Sorry but it is all i can write with my child on my knees.

natinusala, 2019-09-30:
I was more thinking like getting the crash log and send it to me so that I build a fixed image that you can then try

Seeing as nobody in the team owns a Banana Pi anymore this is the only way to go if you want a fix in a reasonable amount of time

Ioalevia, 2019-09-30:
So give me an instruction please of how i can get the crash log. Banana tries to install image. Resizes storage. Reboots. Ànd stucks.

Ntemis, 2019-09-30:
@kivutar afaik has an a20. Maybe he can send it over to @natinusala so we can start supporting it back

natinusala, 2019-09-30:
Can you:

- add `ssh` to the kernel command line (unsure on how to go on the Banana Pi bootloader)
- connect your board to ethernet
- find the IP address of the board somehow
- connect to SSH using PuTTY
- do `systemctl stop retroarch` then `retroarch -v`
- give me the output of that

Ioalevia, 2019-09-30:
It is a perfect instruction for linux user.... But i think it is impossible for me in windows... So the only variant is to talk to kivutar... Or kill banana....

Ioalevia, 2019-09-30:
Oh. i've forgot. There are one more human error... On the website.... Form lakka.tv/contact/ does not work. No sending letter...

natinusala, 2019-09-30:
What? PuTTY is a Windows program

Ioalevia, 2019-09-30:
Add ssh

natinusala, 2019-09-30:
You need to add SSH to the Banana Pi kernel command line

Ioalevia, 2019-09-30:
How?

Ioalevia, 2019-09-30:
Sorry but it is difficult for me to understand some things....

natinusala, 2019-09-30:
I don't know I'm unfamiliar with the Banana Pi bootloader, Google will have
the answer

Ioalevia, 2019-09-30:
Ok. I'l try to do this. Thanks.

ToKe79, 2019-09-30:
According to https://www.lakka.tv/doc/the-bootloader/, you should edit a file named `uEnv.txt`. This file should be placed on the `LAKKA` partition of your SD card / USB thumb drive. Open this file in your beloved text editor (be it Notepad on Windows) and there should be a line containing the text `boot=...` and other stuff - place the cursor at the end of that line, add a space and `ssh`, save, exit. If not sure, just copy/paste the content of that file as a new comment and one of us will help you. Then put the SD/USB back into your Banana Pi A20 and power on. Make sure Ethernet cable is plugged into the device and the other end of the cable is connected to your router. When the boot process gets to the `Reached target retroarch`, ssh service should be running and you can use [PuTTY](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html) to connect to the device.

But first you need to find the IP address of the device. This can be little tricky, but your router should list all of your devices connected to it including the IPv4 address (number dot number dot number dot number).

Start PuTTY, as `Connection type` select `SSH` (should be pre-selected) and in `Host Name (or IP address)` field key in the IP address, or you can bet on your luck and just key in `lakka` (if your router acts also as DNS relay, it just might resolve that host name to an IP address). `Port` should be `22`, if not, change it to `22`. You can also select `Never` in the lower part - `Close window on exit`. Click the `Open` button.

When you get a login prompt `login as:`, key in `root` (you should see as your are typing) and bash the Enter key. Then it will prompt for `Password:`, again, key in `root` (but this time you will not see what you are typing, even no asterisks and the cursor will not move - so don't panic, that's correct) and bash the Enter key again. If all went OK, you should see a shell prompt, soomething like `Lakka:~#` and a cursor.

So now getting the logs. First stop the currently running RetroArch service. Key in
```
systemctl stop retroarch
```
and, yes, bash the Enter key! And now manually start RetroArch in verbose mode, key in
```
retroarch -v
```
and, again, bash the Enter key. It will print bunch of letters and digits. To copy them, just select them with your mouse cursor (do not right click, only select, they will be copied into the "clipboard" automatically) and then paste here as a new comment.

Godspeed!

ToKe79, 2019-09-30:
After reading https://www.lakka.tv/doc/the-bootloader/ thoroughly I found out, that A20 stands also for the CubieBoard2 and CubieTruck?
Doesn't matter, but the line you are probably looking for in `uEnv.txt` (or `.cb2` or `.ct`) is something like:
```
extraargs='console=ttyS0,115200 console=tty0 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 rootwait quiet ssh loglevel=2 hdmi.audio=EDID:0 disp.screen0_output_mode=EDID:1280x1024p60 consoleblank=0'
```
Here the `ssh` is already included, so maybe it is already done also for Banana Pi / all Allwinner A20 variants. So you just need to find the IP address and connect.

ToKe79, 2019-09-30:
One more note: if `ssh` is not in that `extraargs` line, add it within the single quotes and separate from other command with space.

Ioalevia, 2019-10-01:
Thank you for instructions. I'll try it as fast as it possible

Ioalevia, 2019-10-01:
i found lakka in my network. but bootload line... is quite different...
bootargs=console=ttyS0,115200 console=tty0 boot=/dev/mmcblk0p1 disk=/dev/mmcblk0p2 consoleblank=0 quiet
bootcmd=load mmc 0:1 0x43000000 sun7i-a20-bananapi.dtb; load mmc 0:1 0x42000000 KERNEL; bootz 0x42000000 - 0x43000000

Ioalevia, 2019-10-01:
ssh should stay after quiet?

Ioalevia, 2019-10-01:
!!!!!!!!!

login as: root
root@192.168.100.3's password:
###########################################
# Lakka - The DIY retro emulation console #
# ...... visit http://www.lakka.tv ...... #
###########################################

Lakka (official): 2.3.1 (Allwinner.Bananapi.arm)
Lakka:~ # systemctl stop retroarch
Lakka:~ # retroarch -v
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] Capabilities: NEON VFPv3 VFPv4
[INFO] Built: Sep 16 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
Lakka:~ #

ToKe79, 2019-10-01:
So this looks like missing GLES issue to me. According to above statement (https://github.com/libretro/Lakka-LibreELEC/issues/317#issuecomment-379919188), the last working was 2.1-RC5, which means it was before the merge of the Allwinner devices. At that time `sunxi-mali` was used as `OPENGLES`:

https://github.com/libretro/Lakka-LibreELEC/blob/72314011e80e901ecbe6e75aac6546d8d16ecee8/projects/Bananapi/options#L93-L94

Ioalevia, 2019-10-01:
Can it help build a fixed image?

ToKe79, 2019-10-01:
probably, but not sure...
meanwhile you could test out image based on LibreELEC 9.2 beta: http://nightly.builds.lakka.tv/Lakka-LE9.2/latest/H3.arm/Lakka-H3.arm-2.9-devel-20190917204415-de642b3-bananapi.img.gz

Ioalevia, 2019-10-02:
First error.
Could no initialize ghy ethernet@1c50000
Retrieving file: /extlinux/pxelinux.cfg/degault
ethernet@1c50000 waiting for phy auto negotiation to complete..... Timeout!

When connect ethernet cable
Dhcp client bound to adress 192.168.100.3
***Error: `serverip' not set
Cannot autoload with tftpget
Missing environment variable: pxeuuid
Retrieving file: /extlinux/pxelinux.cfg/c0a86403
Retrieving file: /extlinux/pxelinux.cfg/c0a8640
Retrieving file: /extlinux/pxelinux.cfg/c0a864
Retrieving file: /extlinux/pxelinux.cfg/c0a86
Retrieving file: /extlinux/pxelinux.cfg/c0a8
Retrieving file: /extlinux/pxelinux.cfg/c0a
Retrieving file: /extlinux/pxelinux.cfg/c0
Retrieving file: /extlinux/pxelinux.cfg/c
Retrieving file: /extlinux/pxelinux.cfg/default-arm-sunxi
Retrieving file: /extlinux/pxelinux.cfg/default-arm
Retrieving file: /extlinux/pxelinux.cfg/default

Ioalevia, 2019-10-02:
No flower logo.

Ioalevia, 2019-10-09:
Lakka (official): devel-20191005004050-r28345-g286a2a41f7
no ethernet error more. the same error as Lakka (official): 2.3.1

davidKooreman, 2019-10-09:
I can test if needed with a Bananapi M1+ A20 chipset...

davidKooreman, 2019-10-09:
same results as loalevia here with Lakka-H3.arm-2.9-devel-20190917204415-de642b3-bananapi.img.gz image

Ioalevia, 2019-10-12:
Lakka (official): devel-20191007003917-r28346-g55d917fdd1 (Allwinner.Bananapi.arm)
Lakka:~ # systemctl stop retroarch
Lakka:~ # retroarch -v
[INFO] RetroArch 1.7.8 (Git ea7e682253)
[INFO] === Build =======================================
[INFO] Capabilities: NEON VFPv3 VFPv4
[INFO] Built: Oct  7 2019
[INFO] Version: 1.7.8
[INFO] Git: ea7e682253
[INFO] =================================================
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 30000.00 Hz.
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)

Ioalevia, 2019-10-12:
BUT! HDMI does not work. banana shows flower logo only with component out.

Ioalevia, 2019-10-12:
Lakka-Allwinner.Bananapi.arm-nightly-2019-10-08
the same problem
lakka 2.3.1 works with hdmi but the same problem as earlier.

Ioalevia, 2019-10-12:
Lakka-Allwinner.Bananapi.arm-nightly-2019-10-12
no hdmi connection
no PuTTY connection

Ioalevia, 2019-11-05:
2019-11-04
Image does not work.
Maybe this can help you...
https://www.google.com/url?sa=t&source=web&rct=j&url=http://forum.lemaker.org/thread-3082-1-1.html&ved=2ahUKEwjwj5you9LlAhXODewKHYoODWcQFjAAegQICBAB&usg=AOvVaw3p8EDpgNf1WamYWd6bzMO8&cshid=1572936057258
This image of lakka works on bananapi. But it is old and have mistakes

murakas, 2019-12-21:
@ToKe79 
Hi, I also own a banana pi. I can also participate in testing.

Ntemis, 2019-12-21:
Banana Pi is a A20 soc @ToKe79 has mistaken it for H3 :)
We dont have any A20 boards though so is up to @natinusala & @kivutar

khaa-dv, 2020-01-11:
I also have A20 Banana, I can also participate in testing and providing logs. I also want a working Lakka

khaa-dv, 2020-01-13:
@ToKe79 , @kivutar , I am ready to compile the image for the Banana M1 (A20). Tell me how to do this?

kamokar, 2020-01-15:
Hello. 
I can participate in testing the image as well.

WizzardSK, 2020-02-29:
I still had the working version 2.2.1rc5 on my SD card in BPI, so here you are, it is just the first partition, you also have to create another FAT32 partition called LAKKA_DISK. https://drive.google.com/open?id=1c1zy7xjEycS6V3M1V4G3i0t2p7F1_iKd

Ioalevia, 2020-03-01:
Any instructions of creating FAT32 partition called LAKKA_DISK??

WizzardSK, 2020-03-01:
Yes, I just used KDE Partition Manager or GParted in Linux. Created new partition, set FAT32 (maybe ext4 would work too) and apply.

Ioalevia, 2020-03-01:
so. i use image writer and write image on card. than i create fat32 partition with name LAKKA_DISK. Right?

WizzardSK, 2020-03-01:
Almost, but you first need to create an empty 512MB (fat) partition on sd card and write that image to this partition, cause it is just an partition image, not whole drive. The second step is correct. Hope it will work.

Ioalevia, 2020-03-01:
not so easy but it works. but is it normal- White screen with folders from up to down?

Ioalevia, 2020-03-01:
in my variant there no theme...

WizzardSK, 2020-03-01:
It have it with blue xmb theme. Maybe I should make an image of whole sd card

Ioalevia, 2020-03-01:
yes/ on the ext4 partition on my card there a lot of config folders/ it can not work good without them

[47 more replies]

closed 2020-08-18 by Ntemis