libretro/Lakka-LibreELEC #321: HDMI Audio Out not working

mcnutter1, 2017-12-27:
Running Lakka 2.1 GA on a Mac Mini, audio by default plays out of the built in speaker but when trying to direct audio out of the HDMI port no sound is heard. 

Lakka:~ # aplay -L
null
    Discard all samples (playback) or generate zero samples (capture)
default:CARD=NVidia
    HDA NVidia, CS4206 Analog
    Default Audio Device
sysdefault:CARD=NVidia
    HDA NVidia, CS4206 Analog
    Default Audio Device
front:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    Front speakers
surround21:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    4.0 Surround output to Front and Rear speakers
surround41:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Analog
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=NVidia,DEV=0
    HDA NVidia, CS4206 Digital
    IEC958 (S/PDIF) Digital Audio Output
hdmi:CARD=NVidia,DEV=0
    HDA NVidia, HDMI 0
    HDMI Audio Output
hdmi:CARD=NVidia,DEV=1
    HDA NVidia, HDMI 0
    HDMI Audio Output
hdmi:CARD=NVidia,DEV=2
    HDA NVidia, HDMI 0
    HDMI Audio Output
Lakka:~ #

When I try any of the HDMI devices nothing plays.

gouchi, 2017-12-29:
Hi,

What is the output of 

`grep "audio_d.*" /storage/.config/retroarch/retroarch.cfg`

and `dmesg` ?

Thank you.

mcnutter1, 2017-12-29:
Lakka:~ # grep "audio_d.*" /storage/.config/retroarch/retroarch.cfg
audio_driver = "alsathread"
audio_dsp_plugin = ""
audio_device = ""
Lakka:~ # dmesg
[    0.623920] system 00:02: [io  0x0580-0x05ff] has been reserved
[    0.623922] system 00:02: [io  0x0800-0x087f] has been reserved
[    0.623924] system 00:02: [io  0x0880-0x08ff] has been reserved
[    0.623927] system 00:02: [io  0x04d0-0x04d1] has been reserved
[    0.623930] system 00:02: [io  0x0295-0x0296] has been reserved
[    0.623935] system 00:02: Plug and Play ACPI device, IDs PNP0c02 (active)
[    0.623973] pnp 00:03: Plug and Play ACPI device, IDs PNP0b00 (active)
[    0.624545] pnp: PnP ACPI: found 4 devices
[    0.627798] pci 0000:05:00.0: assigning 32 device properties
[    0.627853] pci 0000:02:00.0: assigning 1 device properties
[    0.627883] pci 0000:00:08.0: assigning 4 device properties
[    0.632423] clocksource: acpi_pm: mask: 0xffffff max_cycles: 0xffffff, max_idle_ns: 2085701024 ns
[    0.632518] pci 0000:01:00.0: PCI bridge to [bus 02]
[    0.632525] pci 0000:01:00.0:   bridge window [mem 0xd3400000-0xd34fffff]
[    0.632535] pci 0000:00:0e.0: PCI bridge to [bus 01-02]
[    0.632546] pci 0000:00:0e.0:   bridge window [mem 0xd3400000-0xd34fffff]
[    0.632566] pci 0000:00:15.0: PCI bridge to [bus 03]
[    0.632577] pci 0000:00:15.0:   bridge window [mem 0xd3300000-0xd33fffff]
[    0.632598] pci 0000:00:16.0: PCI bridge to [bus 04]
[    0.632609] pci 0000:00:16.0:   bridge window [mem 0xd3200000-0xd32fffff]
[    0.632616] pci 0000:00:16.0:   bridge window [mem 0xd3100000-0xd31fffff 64bit pref]
[    0.632631] pci 0000:00:17.0: PCI bridge to [bus 05]
[    0.632633] pci 0000:00:17.0:   bridge window [io  0x1000-0x1fff]
[    0.632636] pci 0000:00:17.0:   bridge window [mem 0xd2000000-0xd30fffff]
[    0.632639] pci 0000:00:17.0:   bridge window [mem 0xc0000000-0xd1ffffff 64bit pref]
[    0.632643] pci_bus 0000:00: resource 4 [io  0x0000-0x0cf7 window]
[    0.632644] pci_bus 0000:00: resource 5 [io  0x0d00-0xffff window]
[    0.632646] pci_bus 0000:00: resource 6 [mem 0x000a0000-0x000bffff window]
[    0.632647] pci_bus 0000:00: resource 7 [mem 0x000c0000-0x000c3fff window]
[    0.632649] pci_bus 0000:00: resource 8 [mem 0x000c4000-0x000c7fff window]
[    0.632650] pci_bus 0000:00: resource 9 [mem 0x000c8000-0x000cbfff window]
[    0.632652] pci_bus 0000:00: resource 10 [mem 0x000cc000-0x000cffff window]
[    0.632653] pci_bus 0000:00: resource 11 [mem 0x000d0000-0x000d3fff window]
[    0.632655] pci_bus 0000:00: resource 12 [mem 0x000d4000-0x000d7fff window]
[    0.632656] pci_bus 0000:00: resource 13 [mem 0x000d8000-0x000dbfff window]
[    0.632658] pci_bus 0000:00: resource 14 [mem 0x000dc000-0x000dffff window]
[    0.632659] pci_bus 0000:00: resource 15 [mem 0x000e0000-0x000e3fff window]
[    0.632661] pci_bus 0000:00: resource 16 [mem 0x000e4000-0x000e7fff window]
[    0.632662] pci_bus 0000:00: resource 17 [mem 0x000e8000-0x000ebfff window]
[    0.632664] pci_bus 0000:00: resource 18 [mem 0x000ec000-0x000effff window]
[    0.632665] pci_bus 0000:00: resource 19 [mem 0x000f0000-0x000fffff window]
[    0.632667] pci_bus 0000:00: resource 20 [mem 0xc0000000-0xfebfffff window]
[    0.632668] pci_bus 0000:01: resource 1 [mem 0xd3400000-0xd34fffff]
[    0.632670] pci_bus 0000:02: resource 1 [mem 0xd3400000-0xd34fffff]
[    0.632671] pci_bus 0000:03: resource 1 [mem 0xd3300000-0xd33fffff]
[    0.632673] pci_bus 0000:04: resource 1 [mem 0xd3200000-0xd32fffff]
[    0.632674] pci_bus 0000:04: resource 2 [mem 0xd3100000-0xd31fffff 64bit pref]
[    0.632676] pci_bus 0000:05: resource 0 [io  0x1000-0x1fff]
[    0.632677] pci_bus 0000:05: resource 1 [mem 0xd2000000-0xd30fffff]
[    0.632679] pci_bus 0000:05: resource 2 [mem 0xc0000000-0xd1ffffff 64bit pref]
[    0.632789] NET: Registered protocol family 2
[    0.632997] TCP established hash table entries: 32768 (order: 6, 262144 bytes)
[    0.633349] TCP bind hash table entries: 32768 (order: 8, 1835008 bytes)
[    0.633899] TCP: Hash tables configured (established 32768 bind 32768)
[    0.634012] UDP hash table entries: 2048 (order: 6, 262144 bytes)
[    0.634141] UDP-Lite hash table entries: 2048 (order: 6, 262144 bytes)
[    0.634356] NET: Registered protocol family 1
[    0.634513] RPC: Registered named UNIX socket transport module.
[    0.634514] RPC: Registered udp transport module.
[    0.634516] RPC: Registered tcp transport module.
[    0.634517] RPC: Registered tcp NFSv4.1 backchannel transport module.
[    0.635095] ACPI: PCI Interrupt Link [LUS0] enabled at IRQ 20
[    0.686937] ACPI: PCI Interrupt Link [LUS2] enabled at IRQ 22
[    0.687659] ACPI: PCI Interrupt Link [Z000] enabled at IRQ 20
[    0.739919] ACPI: PCI Interrupt Link [Z001] enabled at IRQ 18
[... 761 lines cut ...]
[31464.952486]  ? wl_event_handler+0x70/0x1e0 [wl]
[31464.952488]  ? kthread+0xf2/0x130
[31464.952502]  ? wl_deinit_priv_mem+0x90/0x90 [wl]
[31464.952504]  ? kthread_create_on_node+0x40/0x40
[31464.952505]  ? do_group_exit+0x42/0xb0
[31464.952508]  ? ret_from_fork+0x25/0x30
[31464.952509] ---[ end trace 000000000000001d ]---
[34729.943320] ------------[ cut here ]------------
[34729.943357] WARNING: CPU: 1 PID: 391 at net/wireless/sme.c:907 wl_notify_roaming_status+0xa3/0x110 [wl]
[34729.943358] Modules linked in: bcma snd_hda_codec_hdmi bnep wl(PO) snd_hda_codec_cirrus snd_hda_codec_generic cfg80211 firewire_ohci firewire_core btusb btrtl crc_itu_t btbcm btintel bluetooth rfkill xpad snd_hda_inte [... cut ...]
[34729.943378]  ? dump_stack+0x5c/0x78
[34729.943381]  ? __warn+0xcd/0xf0
[34729.943395]  ? wl_notify_roaming_status+0xa3/0x110 [wl]
[34729.943408]  ? wl_event_handler+0x70/0x1e0 [wl]
[34729.943411]  ? kthread+0xf2/0x130
[34729.943424]  ? wl_deinit_priv_mem+0x90/0x90 [wl]
[34729.943426]  ? kthread_create_on_node+0x40/0x40
[34729.943428]  ? do_group_exit+0x42/0xb0
[34729.943430]  ? ret_from_fork+0x25/0x30
[34729.943431] ---[ end trace 000000000000001e ]---
[104480.008284] NOHZ: local_softirq_pending 80
Lakka:~ #

gouchi, 2017-12-29:
Thank you for the information.

You have in dmesg

```
[ 2.597230] ALSA device list:
[ 2.597231] No soundcards found.
```

But `aplay -L` display pcm devices...

So you try to edit configuration as described in the [documentation](http://www.lakka.tv/doc/Configuring-Lakka/) and changed with one of those values 

audio_device="hdmi:CARD=NVidia,DEV=0" or 
audio_device="hdmi:CARD=NVidia,DEV=1" or 
audio_device="hdmi:CARD=NVidia,DEV=2"

And none of them provide sound ?

Is it possible for each of the option to provide RetroArch log ? 

Don't forget to stop RetroArch before editing the configuration file.

Thank you.

mcnutter1, 2017-12-29:
I tried typing them manually in the config file and also selecting through the GUI, for either there is no sound played.
Logs below:

HDMI 0:
[INFO] RetroArch 1.6.9 (Git c056726)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 SSSE3 SSE4
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: no
[INFO] [DRM]: Connector 0 has 0 modes.
[INFO] [DRM]: Connector 1 connected: yes
[INFO] [DRM]: Connector 1 has 38 modes.
[INFO] [DRM]: Connector 1 assigned to monitor index: #1.
[INFO] [DRM]: Mode 0: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 1: (1920x1080) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 2: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 3: (1920x1080i) 1920 x 1080, 60 Hz
[INFO] [DRM]: Mode 4: (1920x1080) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 5: (1920x1080i) 1920 x 1080, 50 Hz
[INFO] [DRM]: Mode 6: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 7: (1920x1080) 1920 x 1080, 30 Hz
[INFO] [DRM]: Mode 8: (1920x1080) 1920 x 1080, 25 Hz
[INFO] [DRM]: Mode 9: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 10: (1920x1080) 1920 x 1080, 24 Hz
[INFO] [DRM]: Mode 11: (1680x1050) 1680 x 1050, 60 Hz
[INFO] [DRM]: Mode 12: (1600x900) 1600 x 900, 60 Hz
[INFO] [DRM]: Mode 13: (1280x1024) 1280 x 1024, 75 Hz
[INFO] [DRM]: Mode 14: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 15: (1440x900) 1440 x 900, 60 Hz
[INFO] [DRM]: Mode 16: (1366x768) 1366 x 768, 60 Hz
[INFO] [DRM]: Mode 17: (1280x800) 1280 x 800, 60 Hz
[INFO] [DRM]: Mode 18: (1152x864) 1152 x 864, 75 Hz
[INFO] [DRM]: Mode 19: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 20: (1280x720) 1280 x 720, 60 Hz
[INFO] [DRM]: Mode 21: (1280x720) 1280 x 720, 50 Hz
[INFO] [DRM]: Mode 22: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 23: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 24: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 25: (832x624) 832 x 624, 75 Hz
[INFO] [DRM]: Mode 26: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 27: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 28: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 29: (720x576) 720 x 576, 50 Hz
[INFO] [DRM]: Mode 30: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 31: (720x480) 720 x 480, 60 Hz
[INFO] [DRM]: Mode 32: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 33: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 34: (640x480) 640 x 480, 67 Hz
[INFO] [DRM]: Mode 35: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 36: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 37: (720x400) 720 x 400, 70 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1920x1080.
[... 1432 lines cut ...]
[INFO]  15
[INFO] Environ GET_VARIABLE parallel-n64-gfxplugin-accuracy:
[INFO]  veryhigh
[INFO] Environ GET_VARIABLE parallel-n64-virefresh:
[INFO]  auto
[INFO] Environ GET_VARIABLE parallel-n64-bufferswap:
[INFO]  disabled
[INFO] Environ GET_VARIABLE parallel-n64-framerate:
[INFO]  original
[INFO] Environ GET_VARIABLE parallel-n64-alt-map:
[INFO]  disabled
[INFO] Environ GET_VARIABLE parallel-n64-pak1:
[INFO]  none
[INFO] Environ GET_VARIABLE parallel-n64-pak2:
[INFO]  none
[INFO] Environ GET_VARIABLE parallel-n64-pak3:
[INFO]  none
[INFO] Environ GET_VARIABLE parallel-n64-pak4:
[INFO]  none
[libretro INFO] mupen64plus: Starting R4300 emulator: Dynamic Recompiler
[libretro INFO] Glide64 ucode = 0x3a1cbac3
[libretro INFO] Microcode 0 - RSP SW 2.0X (Super Mario 64)
[libretro INFO] microcheck: old ucode: 2,  new ucode: 0
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).
[INFO] [KMS]: New FB: 1920x1080 (stride: 7680).

gouchi, 2018-01-01:
Thank you for the logs. We don't see any audio issue.

If you changed it from the GUI you need to stop the game and start it again.

Can you check that in Setting tab > Audio, audio mute is not enabled ?

mcnutter1, 2018-01-01:
Mute wasn’t on, I tried all 3 hdmi outputs and neither work, and tried with
the game off before starting, when I change to the internal speaker it
plays sound. I also booted up with a different OS to make sure it’s not a
hardware problem and sound works fine out of the hdmi.

gouchi, 2018-01-01:
And what do you get with [aplay](https://forums.libretro.com/t/lakka-working-fine-but-with-no-sound-ati-radeon-inside/4133/2) ? Please [stop retroarch](http://www.lakka.tv/doc/Troubleshooting-Lakka/) before trying it.

Thank you.

Ntemis, 2018-02-25:
no response since > 30 days

closed 2018-02-25 by Ntemis