libretro/Lakka-LibreELEC #322: Lakka bootloop with any version

RetroGlow, 2017-12-30:
- Which version of Lakka are you using?
Generic.i386-2.1
- What system hardware are you using?
Dell windows xp home. No upgrades, straight out of box hardware.
- What did you do?
We USB booted. It completely wiped xp os.
- What did you expect to happen?
For it to run smoothly without an issue.
- What happened instead?
It flashes the flower icon infinitely. It doesn't let us use ssh or any configurations. Also there are no ways to get the logs. I've tried.

Please help. We have tried other builds and everything and still no luck.

RobLoach, 2017-12-30:
I've had this happen when it was a corrupted install. Try doing an md5 check on the image?

RetroGlow, 2017-12-30:
how would I go about doing it?

ToKe79, 2018-01-31:
Hi,
I have similar issue on old hardware (Compaq 6820s), when Lakka is installed on the internal storage, I cannot boot Lakka from USB. It looks like during the first boot it started installer instead of the live distro and the installation is corrupted.

I suggest, since it already wiped the windows xp from the internal drive, to live boot a linux distro from usb/cd and remove all partitions from the disk and then try to install again. Or instead of live try to boot installer (type `installer` at the `boot:` prompt).

http://www.lakka.tv/get/linux/generic/install/first-boot/

ToKe79, 2018-01-31:
But now I think there might be another issue - RetroArch is segfaulting (maybe incompatible display driver) and systemd restarts the service in a loop - hence the "flashing of the Lakka logo".
To troubleshoot you could do following:
Burn the image to USB thumb drive.
Boot from the image.
At `boot:` prompt type `live tty retroarch=0` and hit `Enter` key.
You will see the Lakka flower, wait for 30-60 seconds and press `Ctrl`+`Alt`+`F3` on the keyboard to switch to the console, you should see the root prompt `Lakka:~#`.
Then you will remount the FAT32 partition to read-write mode, so we can save some logs (if you can read EXT4-FS on your system, you may skip this step):
```
mount -o remount,rw /flash
```
Now you can get some logs by following the [documentation](http://www.lakka.tv/doc/Troubleshooting-Lakka/).
Very usefull is the output of following commands (replace `flash` with `storage` below if you want to save the output to the EXT4 partition):

```
dmesg >/flash/dmesg.txt
lspci -nnk >/flash/lspci.txt
lsusb >/flash/lsusb.txt
```

Also you can get a log from RetroArch:
```
retroarch -v &>/flash/retroarch.txt
```
You will find the logs on the USB drive and provide them here.

NyanProgrammer, 2018-02-21:
I have the same problem on my old machine:

Pentium IV
HD4650 AGP (DVI)

I use nomodeset and acpi=off, and  i cant log some information

ToKe79, 2018-02-21:
@NyanProgrammer Have you tried these steps? https://github.com/libretro/Lakka-LibreELEC/issues/322#issuecomment-362070492

NyanProgrammer, 2018-02-22:
I get the log files: (lsusb ,  not logged )

dmesg.txt:
```[    0.000000] Linux version 4.11.12-rt14 (kivutar@Server01) (gcc version 6.2.0 (GCC) ) #1 SMP PREEMPT RT Sun Nov 19 00:57:45 CST 2017
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e4000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffbffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffc0000-0x000000007ffcffff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007ffd0000-0x000000007fffffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ff780000-0x00000000ffffffff] reserved
[    0.000000] Notice: NX (Execute Disable) protection missing in CPU!
[    0.000000] SMBIOS 2.3 present.
[    0.000000] DMI: System manufacturer System Product Name/P4S800-MX SE, BIOS 1009.002 10/20/2006
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x7ffc0 max_arch_pfn = 0x100000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000 mask F80000000 write-back
[    0.000000]   1 disabled
[    0.000000]   2 disabled
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- UC  
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [c00ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] initial memory mapped: [mem 0x00000000-0x023fffff]
[    0.000000] Base memory trampoline at [c009b000] 9b000 size 16384
[    0.000000] BRK [0x02197000, 0x02197fff] PGTABLE
[    0.000000] 1159MB HIGHMEM available.
[    0.000000] 887MB LOWMEM available.
[    0.000000]   mapped low ram: 0 - 377fe000
[    0.000000]   low ram: 0 - 377fe000
[    0.000000] BRK [0x02198000, 0x02198fff] PGTABLE
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.000000]   Normal   [mem 0x0000000001000000-0x00000000377fdfff]
[    0.000000]   HighMem  [mem 0x00000000377fe000-0x000000007ffbffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.000000]   node   0: [mem 0x0000000000100000-0x000000007ffbffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000001000-0x000000007ffbffff]
[    0.000000] On node 0 totalpages: 524126
[    0.000000] free_area_init_node: node 0, pgdat c1ea1b40, node_mem_map f67fe020
[    0.000000]   DMA zone: 32 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 3998 pages, LIFO batch:0
[    0.000000]   Normal zone: 1744 pages used for memmap
[    0.000000]   Normal zone: 223230 pages, LIFO batch:31
[    0.000000]   HighMem zone: 296898 pages, LIFO batch:31
[    0.000000] Using APIC driver default
[    0.000000] Intel MultiProcessor Specification v1.4
[    0.000000]     Virtual Wire compatibility mode.
[    0.000000] MPTABLE: OEM ID: SiS     
[    0.000000] MPTABLE: Product ID: 
[    0.000000] MPTABLE: APIC at: 0xFEE00000
[    0.000000] Processor #0 (Bootup-CPU)
[    0.000000] IOAPIC[0]: apic_id 2, version 17, address 0xfec00000, GSI 0-23
[    0.000000] Processors: 1
[... 491 lines cut ...]
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[ERROR] [KMS]: Couldn't find a suitable DRM device.
[INFO] [GL]: Found GL context: null
[INFO] [GL]: Detecting screen resolution 320x240.
[INFO] [GL]: Vendor: (null), Renderer: (null).
[INFO] [GL]: Version: (null).
Segmentation fault (core dumped)
```

ToKe79, 2018-02-22:
@NyanProgrammer This error means, that support for your GPU is not available in the kernel.
`[ERROR] [KMS]: Couldn't find a suitable DRM device.`

Your GPU is:
```
01:00.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] RV730 [Radeon HD 4600 AGP Series] [1002:9495]
	Subsystem: PC Partner Limited / Sapphire Technology RV730 [Radeon HD 4600 AGP Series] (Radeon HD 4650 AGP DDR2) [174b:0028]
```
We use kernel 4.11 for Generic, truly I do not know how the support for ATI in this kernel is.

NyanProgrammer, 2018-02-22:
But ubuntu Mate 17.04 has a 4.X kernel and runs fine in this machine with compiz and GPU support (free driver)

ToKe79, 2018-02-22:
That is probably driver for X, not for KMS/DRM which is used by Lakka. I think more support for R700 core was added in 4.13 and later kernel.
```
[    0.488534] [drm] VGACON disable radeon kernel modesetting.
[    0.488544] [drm:radeon_init] *ERROR* No UMS support in radeon module!
[    0.488556] [drm:amdgpu_init] *ERROR* VGACON disables amdgpu kernel modesetting.
```
@gouchi @kivutar - can you shed some light on this? I checked `make menuconfig` and did not find anything else we could change to enable the support, but on the other hand I am not into kernel stuff either. Thank you!

NyanProgrammer, 2018-02-22:
This will sound weird or something.... but I want to help , well I work all day but i think I know enough C/C++ to help or acquire some information... of course I want to learn new "tricks" and Knowledge ... So I'll spent some time to help or start to understand the code

ToKe79, 2018-02-22:
Any help is really appreciated, but in this case I think we would have to update the kernel version for Generic builds. As the underlying OS is based on LibreELEC and we currently use the 8.2 branch, we use the same kernel version. When LE 9.0 gets stable, and there will be some hero who will merge it to Lakka, the kernel will get updated too.

However I could try to make a special build with updated kernel, but cannot make promises. Depends on my free time and skills :-)

gouchi, 2018-02-23:
The GPU is supported by [CONFIG_DRM_RADEON](https://cateee.net/lkddb/web-lkddb/DRM_RADEON.html) but as it using AGP I think we should try to make a test by [enabling AGP support for ATI/AMD](https://github.com/libretro/Lakka-LibreELEC/blob/Lakka-V2.1-dev/projects/Generic/linux/linux.i386.conf#L3098-#L3099).

Ntemis, 2018-02-23:
Yes that would fix this

ToKe79, 2018-02-23:
@NyanProgrammer can you try this image?
http://nightly.builds.lakka.tv/special_builds/agp_ati_amd/

NyanProgrammer, 2018-02-24:
Not worked... TwT, but I get more logs
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1753707/dmesg.txt)
[lspci.txt](https://github.com/libretro/Lakka-LibreELEC/files/1753708/lspci.txt)
[lsusb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1753709/lsusb.txt)
[retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/1753710/retroarch.txt)

gouchi, 2018-02-24:
I didn't check closely enough :)

`00:00.0 Host bridge [0600]: Silicon Integrated Systems [SiS] 661FX/M661FX/M661MX Host [1039:0661] (rev 11)`

So we should try to enable [CONFIG_AGP_SIS](https://cateee.net/lkddb/web-lkddb/AGP_SIS.html)

`00:01.0 PCI bridge [0604]: Silicon Integrated Systems [SiS] AGP Port (virtual PCI-to-PCI bridge) [1039:0003]`

So we should try to enable [CONFIG_HOTPLUG_PCI_SHPC](https://cateee.net/lkddb/web-lkddb/HOTPLUG_PCI_SHPC.html)

ToKe79, 2018-02-24:
@NyanProgrammer I posted udpated image at http://nightly.builds.lakka.tv/special_builds/agp_ati_amd/

NyanProgrammer, 2018-02-25:
Not worked , i get new logs.
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1756412/dmesg.txt)
[lspci.txt](https://github.com/libretro/Lakka-LibreELEC/files/1756413/lspci.txt)
[lsusb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1756414/lsusb.txt)
[retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/1756415/retroarch.txt)

gouchi, 2018-02-25:
By any chance you didn't edit [cmdline](http://www.lakka.tv/doc/The-bootloader/#on-generic-pc) and added nomodeset ?

It seems `00:01.0 PCI bridge [0604]: Silicon Integrated Systems [SiS] AGP Port (virtual PCI-to-PCI bridge) [1039:0003]` is still not detected according to the [lspci log](https://github.com/libretro/Lakka-LibreELEC/files/1756413/lspci.txt).

NyanProgrammer, 2018-02-25:
Without nomodeset option , display turn off (not signal), so... i cant do something

gouchi, 2018-02-25:
Please remove it and provide some logs.

Thank you.

NyanProgrammer, 2018-02-25:
Yes , I tried SSH and tty, but it seems ssh cant start, Im not sure , but after 10~ minutes the machine shutdown, I think something crash or something and call system shutdown, this happen only without nomodeset... (I tried updated image )

Note:  Previous version of lakka run on this machine

gouchi, 2018-02-26:
Please add `ssh` to the cmdline so that you can ssh to the station.

alejandrosenes, 2018-02-26:
Same problem with 32Bits edition, tested on intel atom d510 and mac mini 2.1 1.83 C2Duo

NyanProgrammer, 2018-03-13:
Well , SSH not work properly.... i cant start session (the same screen , and i cant write any command) .... the same after 30 minutes...
[image]

gouchi, 2018-03-17:
Hi,

Is it possible to get the usual logs for the version ([Lakka-Generic.i386-devel-20161221183352-r21729-g274b40f](http://sources.lakka.tv/nightly/Generic.i386/Lakka-Generic.i386-devel-20161221183352-r21729-g274b40f.img.gz)) as it was working at that time.

Thank you.

NyanProgrammer, 2018-03-18:
I get the logs form the old version ,I hope this can be useful(Lakka-Generic.i386-devel-20161221183352-r21729-g274b40f)
[dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1822826/dmesg.txt)
[lspci.txt](https://github.com/libretro/Lakka-LibreELEC/files/1822827/lspci.txt)
[lsusb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1822828/lsusb.txt)

gouchi, 2018-03-18:
lspci shows that radeon driver is used and we don't need agpgart-sis driver.

Can you try to boot with, if you see something ?

`live debugging nosplash break=load_modules`

gouchi, 2018-03-18:
lspci shows that radeon driver is used and we don't need agpgart-sis driver.

Can you try to boot with, if you see something ?

`live debugging nosplash break=load_modules`

Thank you.

mostfurious, 2018-05-09:
Same issue here, (cant post at Lakka forums) here are my logs:
Retroarch.txt
> [INFO] RetroArch 1.6.9 (Git c056726)
[INFO] === Build =======================================
Capabilities: MMX MMXEXT SSE1 SSE2 SSE3 
Built: Nov 21 2017
[INFO] Version: 1.6.9
[INFO] Git: c056726
[INFO] =================================================
[INFO] [Config]: Loading default config.
[INFO] [Config]: loading config from: (null).
[INFO] Looking for config in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Looking for config in: "/storage/.retroarch.cfg".
[WARN] Config: using skeleton config "/etc/retroarch.cfg" as base for a new config file.
[WARN] Config: Created new config file in: "/storage/.config/retroarch/retroarch.cfg".
[INFO] Environ SET_PIXEL_FORMAT: RGB565.
[INFO] Redirecting save file to "/storage/savefiles/.srm".
[INFO] Redirecting savestate to "/storage/savestates/.state".
[INFO] Version of libretro API: 1
[INFO] Compiled against API: 1
[INFO] [Audio]: Set audio input rate to: 29970.03 Hz.
[INFO] Found shader "/tmp/shaders/slang-conversion-template.glsl"
[INFO] Found shader "/tmp/shaders/stock.glsl"
[INFO] [Video]: Video @ fullscreen
[INFO] [Video]: Starting threaded video driver ...
[INFO] [DRM]: Found 2 connectors.
[INFO] [DRM]: Connector 0 connected: yes
[INFO] [DRM]: Connector 0 has 14 modes.
[INFO] [DRM]: Connector 0 assigned to monitor index: #1.
[INFO] [DRM]: Connector 1 connected: no
[INFO] [DRM]: Connector 1 has 0 modes.
[INFO] [DRM]: Mode 0: (1280x1024) 1280 x 1024, 60 Hz
[INFO] [DRM]: Mode 1: (1280x1024) 1280 x 1024, 75 Hz
[INFO] [DRM]: Mode 2: (1152x864) 1152 x 864, 75 Hz
[INFO] [DRM]: Mode 3: (1024x768) 1024 x 768, 75 Hz
[INFO] [DRM]: Mode 4: (1024x768) 1024 x 768, 70 Hz
[INFO] [DRM]: Mode 5: (1024x768) 1024 x 768, 60 Hz
[INFO] [DRM]: Mode 6: (832x624) 832 x 624, 75 Hz
[INFO] [DRM]: Mode 7: (800x600) 800 x 600, 75 Hz
[INFO] [DRM]: Mode 8: (800x600) 800 x 600, 72 Hz
[INFO] [DRM]: Mode 9: (800x600) 800 x 600, 60 Hz
[INFO] [DRM]: Mode 10: (640x480) 640 x 480, 75 Hz
[INFO] [DRM]: Mode 11: (640x480) 640 x 480, 73 Hz
[INFO] [DRM]: Mode 12: (640x480) 640 x 480, 60 Hz
[INFO] [DRM]: Mode 13: (720x400) 720 x 400, 70 Hz
[INFO] [GL]: Found GL context: kms
[INFO] [GL]: Detecting screen resolution 1280x1024.
[INFO] [EGL] Found EGL_EXT_platform_base, trying eglGetPlatformDisplayEXT
[INFO] [EGL]: EGL version: 1.4
[INFO] [EGL]: Current context: 0x7f4b7002f400.
[INFO] [KMS]: New FB: 1280x1024 (stride: 5120).
[INFO] [GL]: Vendor: X.Org R300 Project, Renderer: Gallium 0.4 on ATI RS480.
[INFO] [GL]: Version: 2.1 Mesa 17.1.3.
[INFO] [GL]: Using resolution 1280x1024
[INFO] [GL]: Default shader backend found: glsl.
[INFO] [Shader driver]: Using GLSL shader backend.
[INFO] [GLSL]: Checking GLSL shader support ...
[WARN] [GL]: Stock GLSL shaders will be used.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[INFO] [GLSL]: Found GLSL vertex shader.
[INFO] [GLSL]: Found GLSL fragment shader.
[INFO] [GLSL]: Linking GLSL program.
[... 688 lines cut ...]
	Kernel driver in use: piix4_smbus
00:14.1 IDE interface [0101]: Advanced Micro Devices, Inc. [AMD/ATI] IXP SB4x0 IDE Controller [1002:4376]
	Subsystem: Hewlett-Packard Company IXP SB4x0 IDE Controller [103c:3009]
	Kernel driver in use: pata_atiixp
00:14.3 ISA bridge [0601]: Advanced Micro Devices, Inc. [AMD/ATI] IXP SB4x0 PCI-ISA Bridge [1002:4377]
	Subsystem: Hewlett-Packard Company IXP SB4x0 PCI-ISA Bridge [103c:3009]
00:14.4 PCI bridge [0604]: Advanced Micro Devices, Inc. [AMD/ATI] IXP SB4x0 PCI-PCI Bridge [1002:4371]
00:14.5 Multimedia audio controller [0401]: Advanced Micro Devices, Inc. [AMD/ATI] IXP SB400 AC'97 Audio Controller [1002:4370] (rev 02)
	Subsystem: Hewlett-Packard Company IXP SB400 AC'97 Audio Controller [103c:3009]
	Kernel driver in use: snd_atiixp
	Kernel modules: snd_atiixp
00:18.0 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] HyperTransport Technology Configuration [1022:1100]
00:18.1 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Address Map [1022:1101]
00:18.2 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] DRAM Controller [1022:1102]
00:18.3 Host bridge [0600]: Advanced Micro Devices, Inc. [AMD] K8 [Athlon64/Opteron] Miscellaneous Control [1022:1103]
	Kernel driver in use: k8temp
	Kernel modules: k8temp
01:05.0 VGA compatible controller [0300]: Advanced Micro Devices, Inc. [AMD/ATI] RS480 [Radeon Xpress 200 Series] [1002:5954]
	Subsystem: Hewlett-Packard Company RS480 [Radeon Xpress 200 Series] [103c:3009]
	Kernel driver in use: radeon
01:05.1 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] RS480 [Radeon Xpress 200 Series] (Secondary) [1002:5854]
	Subsystem: Hewlett-Packard Company RS480 [Radeon Xpress 200 Series] (Secondary) [103c:3008]
02:00.0 Ethernet controller [0200]: Broadcom Limited NetXtreme BCM5751 Gigabit Ethernet PCI Express [14e4:1677] (rev 20)
	Subsystem: Hewlett-Packard Company NetXtreme BCM5751 Gigabit Ethernet PCI Express [103c:3009]
	Kernel driver in use: tg3

gouchi, 2018-06-02:
Also try with our [nightly builds](http://nightly.builds.lakka.tv/).

Thank you.

NyanProgrammer, 2018-07-29:
I tried the nightly builds and nothing changed

gouchi, 2018-08-02:
Please provide some [logs](http://www.lakka.tv/doc/Troubleshooting-Lakka/).

Thank you.

gouchi, 2019-03-13:
Hi,

May you try [this command](https://github.com/libretro/Lakka-LibreELEC/issues/618#issuecomment-472534970) to launch RA ?

Thank you.

closed 2019-06-26 by ToKe79