libretro/Lakka-LibreELEC #323: Feature request: menu option to access tty command line.

draperjames, 2017-12-30:
It would be great if there was an option in the menu to access Lakka's tty commandline. Right now the only way to access it requires using ssh from another machine or modifying the boot loader which is pretty difficult for weekend-warriors/noobs like me.

I'd submit a PR myself on this but have no idea where to start. Any advice would be appreciated.

RobLoach, 2017-12-30:
A Terminal Emulator libretro core would be pretty cool.

RobLoach, 2017-12-30:
Haven't tested lately, but use to be able to use CTRL SHIFT F1-6 to switch between screens.

draperjames, 2017-12-30:
@RobLoach by switch screens do you mean change monitor index? I've been able to access the command line by adding tty to the bootloader then when running Lakka pressing escape then Ctrl+F3 quickly and typing `systemctl stop retoarch`. It's not an elegant method but it works.

RobLoach, 2017-12-30:
Correction: CTRL-ALT-F2 and CTRL-ALT-F7 ..... I use that on my Kodi-buntu box, haven't tested on LibreELEC/Lakka.

Ntemis, 2017-12-30:
CTRL+ALT+F2 should land you in terminal

draperjames, 2017-12-31:
@Ntemis I just tried CTRL+ALT+F2 on a fresh install of the most recent build of Lakka for PC on a flash drive and it did not work at all. Granted I didn't mess around with the bootloader or anything. How did you enable that?

Ntemis, 2018-01-01:
I meant pressing CTRL+ALT+F2 keys should land you in command line as all Linux distros normally do, i didn't said it works

ToKe79, 2018-01-31:
Terminal console is not enabled by default. [Here](http://www.lakka.tv/doc/Accessing-Lakka-command-line-interface/) are guides how to access the shell and [here](http://www.lakka.tv/doc/The-bootloader/) are guides how to modify cmdline for individual builds/systems.

kivutar, 2018-04-05:
Even when tty is enabled, retroarch doesn't let you CTRL+ALT+F

closed 2020-08-18 by Ntemis