libretro/Lakka-LibreELEC #324: Tinker Board Wifi not working

shadowed87, 2017-12-31:
When using the tinker board with the image 
Lakka-Rockchip.TinkerBoard.arm-2.1.img.gz | 271M | 22-Nov-2017 12:38

It correctly boots, but I am unable to load any game.
I have not tested the ethernet cable but when i go to select the wifi, no network is showed.

MrTactigles, 2017-12-31:
Ethernet works (on my board at least) with that image, but like you wifi is in-op with the error from this thread https://forums.libretro.com/t/asus-tinker-board-support/9116/62

Ntemis, 2018-01-01:
Wifi doesnt work with rockchip kernel only with asus kernel, we dont know why yet.

space81, 2018-01-01:
Hello all
My wifi works on the Tinker Board with Lakka 2.1 RC5 but none of the newer images. Hope this helps.

MrTactigles, 2018-01-05:
Just wanted to note that Bluetooth IS working, just the WIFI portion of the chip is not.

onesun43, 2018-01-06:
Are you forced to use the rockchip kernel over the asus kernel? If wifi is working with the asus kernel, is there a way for users to switch to that kernel?

MrTactigles, 2018-01-06:
WiFi works on my RPi3, and my RPi Zero W using the images on the main website connecting to a non-hidden SSID with WPA2.

Ntemis, 2018-01-06:
Rockchip generic bsp kernel can be used for many  
Rockchip devices like miqi,rock64 etc

Sam2Hsu, 2018-01-12:
Hello all,
Pull the latest Lakka and change CONFIG_WIFI_LOAD_DRIVER_WHEN_KERNEL_BOOTUP to y can fix Wi-Fi on Tinker board.

Here's my linux.arm.conf:
```
CONFIG_WL_ROCKCHIP=y
# CONFIG_WIFI_BUILD_MODULE is not set
CONFIG_WIFI_LOAD_DRIVER_WHEN_KERNEL_BOOTUP=y
# CONFIG_AP6XXX is not set
CONFIG_RTL_WIRELESS_SOLUTION=y
# CONFIG_RTL8188EU is not set
# CONFIG_RTL8188FU is not set
# CONFIG_RTL8189ES is not set
# CONFIG_RTL8189FS is not set
CONFIG_RTL8723BS=y
# CONFIG_RTL8723BU is not set
# CONFIG_RTL8723CS is not set
# CONFIG_RTL8723DS is not set
```

Ntemis, 2018-01-12:
Ok thank you for the feedback i will test.

closed 2018-01-13 by Ntemis

Tonymac32, 2018-05-27:
Please review that this bug is still active, kernel was changed from MQMaker to the Rockchip one at the end of January, after these reports.

Ntemis, 2018-05-27:
We were using the Rockchip kernel at the time of this bug so looks like it but no hurting to a re-test.

Tonymac32, 2018-05-27:
OK, for Armbian I have configured the driver as a module, and per https://github.com/rockchip-linux/kernel/commit/8286a41a20fe9ff3842842ffc840440066ceeff2#diff-ffbbcd96ee1a57b7026b551d3eca788f

The CONFIG_WIFI_LOAD_DRIVER_WHEN_KERNEL_BOOTUP should be no.  I had to find that, it's only (afaik) documented in that commit.