libretro/Lakka-LibreELEC #325: RPi3 wifi scan does'nt work

djdejawu, 2018-01-03:
When I'm open "Wi-Fi" menu and scan for networks - it's show only "[]" this icon, but i can discover many networks in 2.4 GHz on my phone. It's a bug.
Latest stable release 2.1.

markwkidd, 2018-01-12:
I have the exact same symptom with Lakka 2.1 stable on a Bqeel M9C S905X device.

mstroh76, 2018-01-24:
I had this problem with Raspberry Pi Zero W. But after same time of waiting and reenter of Menu, the Wifi networks were shown. I thought it was because of the slow Hardware. I took also very long to connect to the Wifi network.

kivutar, 2018-01-31:
Works perfectly fine for me, I can even connect.
Can one of you test with the latest http://nightly.builds.lakka.tv/ ?

mstroh76, 2018-02-02:
Lakka-RPi.arm-2.1-devel-r28180-g31ef195 and Pi Zero W:
No Problems with WiFi - all working fine

markwkidd, 2018-02-03:
I would test a nightly with my M905x box since it had the same symptom, but it seems to have suffered some kind of hardware failure. Therefore, seems like this issue is cleared up.

closed 2018-02-03 by kivutar