libretro/Lakka-LibreELEC #326: can't compile latest (940d414529aff7deb1e0fa31b2ecd03182fee86d)

Arnochocobo, 2018-01-04:
There seems to be an issue with mrboom

I am receiving this error:

error: pathspec '7a218ca' did not match any file known to git
recipe for target image failed

I tried a few different things, and I always end up with this error.

Ntemis, 2018-01-04:
Could be that someone hard reset the repo, i will fix it asap

Arnochocobo, 2018-01-05:
Thanks Ntemis. Issue still occurs now. Is it OK on your end??? Is this why you closed it?

Ntemis, 2018-01-05:
No i will push a fix thats why i close it

Arnochocobo, 2018-01-05:
You close an issue before verifying that this is fixed? It makes no sense.

An issue should always be open until this is fix verified, this is the 101 of any software development cycle. How can you keep track of what is really open if you close issues without verifying the fix first? The best is to have the user that found the issue in the first place confirming the fix, of course. I appreciate the time you put into this, but I don't understand at all how you commit/operate.

You keep updating all the emulators all the time. Do you even take the time to check that the update is critical and really necessary? Does it really fix something really bad?Are you checking a few known problematic games to make sure it is working properly? Or do you just update because "latest is always the best"??? That is the perfect recipe for unstable chaos (like Retroarch)

Lakka has never been that broken, compared to when Kivutar was taking care of it alone (during he OpenElec days). First time I updated to LibreElec, FBA didn't work at all, and has been fine for the longest time in OpenElec. Sure, we had a retroarch update every 4-6 weeks, but at least we knew it is stable and more or less checked. Now Lakka became a giant mess that might eventually compile... Not good. Just my two cents, I have just been in large scale software development for more than 15 years :)

Again, I appreciate the time you put into this, but it could be MUCH better with better practices. Thanks.

Ntemis, 2018-01-05:
Hi,
I will gladly tell you how i operate and same time answer your questions :)
- I keep track of issues because am a responsible guy and i did update the cores including the one you reported not working after testing before push so it was broken for you but not for me at the time i did the compile. So its on me to fix whats broken here and i did as you can see.
- I updated all the cores intentionally as i want them to be in synced and i hope you noticed with retroarch v1.7 that was released as stable some days ago so we be able to catch any issues until 1.7.1 is out. Without this move we might not catch all bugs in final lakka 2.1.1 in cores or retroarch it self.
And from Lakka v2.1 stable till v2.1.1 i consider the branch and i have every right to, as a development phase for fixing updating,testing and porting new socs like i did already with S912. 
- Having a core failing to compile even so its not my fault per se is magnitudes away of what you imply here, how is lakka broken in any way? I didn't see you reporting FBA been broken in past issues open or closed. What you are accusing here is at least not fair to anyone.
And you,above all that has been in software development for more than 15 years was the last person i would expect to get this criticism from.
Regards.

Arnochocobo, 2018-01-05:
I could verify your fix, so thanks for that!

As for this:
"I keep track of issues because am a responsible guy and i did update the cores including the one you reported not working after testing before push so it was broken for you but not for me at the time i did the compile. So its on me to fix whats broken here and i did as you can see."

This is how code / bug review works in the real world:
1) I am doing your QA, find an issue and report it
2) You as the dev check the validity, fix if necessary and then get back to QA mentioning the fix can be reviewed
3) QA checks the fix, and close the issue if this is indeed fixed, or keeps it open if it still happens and notifies you, so you can have another look at it.

The dev should never ever be allowed to close their own issue (which they didn't even noticed in the first place) 

What I don't understand at all, and again, I am fine if you want to update every core for no reason 10 times a day, is why you don't check if everything compiles fine and if there is no error after every single commit?? In this case, I don't get how that would compile for you and not for me. (since your fix showed something was clearly broken, and should be broken for everyone)

Well, I am always one of the rare people compiling latest, and actually PLAYING (yes, you can use a controller for something else than scrolling through menus) and checking Lakka.
I didn't bother when I first saw that FBA was broken, and it eventually got fixed along the way with a new version. But I saw those issues popping up quite often, so I decided to start opening tickets and at least raise someone's attention to get it fixed, since it is clear you just update for the sake of it, without actually playing and checking everything is in one piece.

"And you,above all that has been in software development for more than 15 years was the last person i would expect to get this criticism from."
-> Please feel free to ship multiple millions of commercial products like I did, and then sure, we can talk about code review practices.

Ntemis, 2018-01-05:
There is one catch here that you are not aware of, @Kivutar reported the issue on me via our irc channel so i was aware of the issue before you open the issue ticket.
So i knew about the issue and i was going to fix it already thats why i closed it so fast on you.
There was no need for me to go in either QA stage nor check the validity of it, i knew there was an issue and i knew i had to fix it, simple as that.
As i see you keep accusing me for something thats not valid but plain assumptions from your part in open public, if you are so sure about me, why don't you start doing a better job by reporting any issues you can find upstream in Lakka and in retroarch cores? Or even sent PR's reverting my core updates if you find them faulty in any way.

Ntemis, 2018-01-07:
Issue Confirmed fixed

closed 2018-01-07 by Ntemis