libretro/Lakka-LibreELEC #327: Stuck on flower screen, maybe libGL problems

retroalfred, 2018-01-04:
Hi guys,
I am trying to run Lakka-Generic.x86_64-2.1 in live mode on a PC media player (cisco IEC 4610 intel based). On this PC I can easily run ubuntu traditional distributions.
My problem is that it hangs on the flower screen.
I read the forum and tried also changing the following parameter without success:
menu_shader_pipeline = "0"
menu_driver = "rgui"

In attach you can find my logs
Thank you in advanced
Regards
Al
[log-audio.txt](https://github.com/libretro/Lakka-LibreELEC/files/1602941/log-audio.txt)
[log-dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1602942/log-dmesg.txt)
[log-lusb.txt](https://github.com/libretro/Lakka-LibreELEC/files/1602943/log-lusb.txt)
[log-retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/1602944/log-retroarch.txt)
[log-vga.txt](https://github.com/libretro/Lakka-LibreELEC/files/1602945/log-vga.txt)

RobLoach, 2018-01-04:
I've found it can take a while to boot. I've also found sometimes it stalls at boot because of a corrupted install. Make sure the file you downloaded was complete?

retroalfred, 2018-01-04:
I will try to download and install again.
Is the versione x86_64 correct ? or should I use i386 ?

thank you

RobLoach, 2018-01-04:
You can confirm the download was complete by checking the md5:
http://le.builds.lakka.tv/

retroalfred, 2018-01-04:
Hi,
downloaded again and checked that the MD5 was correct.
It stucks again, below you can find the logs.
Thank you again
Regards
Al

[log_dmesg.txt](https://github.com/libretro/Lakka-LibreELEC/files/1604147/log_dmesg.txt)
[log_lspci.txt](https://github.com/libretro/Lakka-LibreELEC/files/1604148/log_lspci.txt)
[log_retroarch.txt](https://github.com/libretro/Lakka-LibreELEC/files/1604149/log_retroarch.txt)

retroalfred, 2018-01-04:
Now I used the version Lakka-Generic.i386-2.1

gouchi, 2018-01-04:
```
[INFO] [DRM]: Connector 1 connected: yes
[INFO] [DRM]: Connector 1 has 11 modes.
[INFO] [DRM]: Connector 1 assigned to monitor index: #2.
```
Can you try to [change the configuration file](http://www.lakka.tv/doc/Configuring-Lakka/)  with
`video_monitor_index = "2"`

Thank you.

P.S = Strange in RA's second long you got 
`Failed to open /tmp/assets/xmb/monochrome/font.ttf: No such file or directory`

retroalfred, 2018-01-05:
Thank you gouchi,
with the correct monitor index the live works !!
Now I installed on the pc and I have the same problem, how can I change the parameter on the hd?
I tried with ctrl+alt+f1 f2 f3 but the prompt does not appear.

Thank you
Regards
Al

retroalfred, 2018-01-05:
Ok I resolved.
Here what I did:
- start from usb with "live ssh"
- connected via ssh and mounted the /dev/sda2 (that's my case) in /tmp/temp
- changed in the /tmp/temp/.config/retroarch/retroarch.cfg the parameters video_monitor_index = "2"

Now I will go on and try to configure.
Thank you for your help guys.
Regards
Al

gouchi, 2018-01-06:
Glad it works.

closed 2018-01-06 by gouchi