libretro/Lakka-LibreELEC #330: Rock64 images freezing after boot

Jturn777, 2018-01-10:
Neither of the available Rock64 images will function more than a few minutes after booting is complete. Tried on multiple boards and SD cards.

Ntemis, 2018-01-23:
No reports of such issue from others users, you sure its not your setup?

Jturn777, 2018-01-23:
I've tried both images available. My Rock64 board works with 3 other
distros without freezing. I've tried running with no peripherals as well.

Ntemis, 2018-02-17:
I would recommend to get a heatsink.Closing issue but for sure my heatsink was on fire when i was testing latest image

closed 2018-02-17 by Ntemis

f0xx0f, 2019-03-31:
Hello. I have the same problem for rock64. Lakka freezing after boot. I have no idea how to solve this problem=( I will be glad to any advice. Thanks!

Jturn777, 2019-03-31:
Havent figured it out yet. Developers were no help. Try recalbox or retropie