libretro/Lakka-LibreELEC #335: Black screen after updating XU4 with cd2ff40be73ae42533197d41ffd415a0442de22b

Arnochocobo, 2018-01-16:
And here we go again...

Compiling -> No problem
However, after updating, system reboots and then black screen, and getting stuck there.

Did you try on an actual XU4? Multiple people on the HK forums were reporting issues with this kernel and even opened a debug party (!!!), which is a very good sign (NOT).

Or did you update just for the sake of it without checking anything on the actual hardware?

Ntemis, 2018-01-16:
Kernel is missing gbm/kms blobs

Arnochocobo, 2018-01-20:
I had no issue to compile it, and now I am stuck with a black screen and can't do anything (except reinstalling an older version). I am stuck there and can't provide logs. Not much more I can do in this situation.

What is the point of committing something unstable, with missing parts apparently, and that has tons of other issues? Did you check on the actual hardware it was not breaking anything before sending your commit?

Ntemis, 2018-01-20:
Stick to 2.1 stable until we announce new rc.

Arnochocobo, 2018-01-21:
This is the whole point of github, check WIP and compile your own version if you feel like it.

Why are you committing broken crap (for something that didn't need to get updated at all) without checking anything on the real hardware? Is that a "responsible" behavior?

Arnochocobo, 2018-01-21:
Kivutar just mentioned that a fresh install on emmc works for him. Updating is definitely not the way to go.
If this is working for Kivutar, then it can be closed, I will try a fresh install as well.

closed 2018-01-21 by Arnochocobo

Ntemis, 2018-02-13:
This should be fixed by now