libretro/Lakka-LibreELEC #336: The README is misleading

carlsmith, 2018-01-17:
The README says...

> Our code is free as in freedom and hosted on Github. We accept external contributions, and...

The project is not free as in freedom, as it includes non-commercial restrictions. The project is only free as in beer.

Technically, it says "our code" is free as in freedom, so if we're being pedantic, the statement is technically true, as anyone can sell Lakka specific code. However, the project as a whole is not free, which is what all but the most careful reading implies.

Furthermore, the project does not meet the Open Source Definition, Section 1 (my emphasis):

> The license **shall not restrict any party from selling** or giving away the software as a component of an aggregate software distribution containing programs from several different sources. The license **shall not require a royalty or other fee for such sale**.

Can you please change the wording to something less likely to mislead? For example...

> Our code is free as in freedom and hosted on Github (though the project uses emulators that forbid commercialisation). We accept external contributions, and...

RetroPie insists that it is open source, even though it does not meet the Open Source Definition, and it is a constant cause of conflict. People use the project as if it is open source, and are then threatened with legal action after their product ships.

twinaphex, 2018-01-17:
There is nobody being mislead here. We can edit the README to exactly what you state if that makes you feel better about it.

closed 2018-01-17 by twinaphex

RobLoach, 2018-01-17:
On a related note, see [Licenses on the libretro docs](http://docs.libretro.com/tech/licenses/) to see which cores are on a non-commercial license. Lakka would meet the Open Source Definition if it weren't for those cores.

Considered putting together a [Lakka-Lite without the cores](https://github.com/libretro/Lakka-LibreELEC/issues/179), but that's another discussion :wink:

carlsmith, 2018-01-18:
Sweet. Thanks guys. On the README, whatever you think is best. I just came across some issues with RetroPie, then read the README here, and thought I should at least mention it. Again, thanks.