libretro/Lakka-LibreELEC #337: PS4 V2 controller starts to behave weirdly when plugged into my PC

GoodDay00, 2018-01-18:
- Which version of Lakka are you using?
Generic.x86_64-2.1
- What system hardware are you using?
PC
- What did you do?
Plugged my PS4 controller into my pc with USB.
- What did you expect to happen?
Work correctly
- What happened instead?
the inputs started to behave weirdly. Like it started to skip, can't go up or down on the menu.

---Input Logs---
Lakka:~ # lsusb
Bus 003 Device 002: ID 0bda:0811 Realtek Semiconductor Corp.
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 002: ID 054c:09cc Sony Corp.
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 002: ID 0bda:0181 Realtek Semiconductor Corp.
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 1038:1384 SteelSeries ApS
Bus 004 Device 002: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

Lakka:~ # dmesg
[    0.000000] Linux version 4.11.12-rt14 (kivutar@Server01) (gcc version 6.2.0                                                                                                                                                              (GCC) ) #1 SMP PREEMPT RT Sat Nov 18 21:12:49 CST 2017
[    0.000000] Command line: BOOT_IMAGE=/KERNEL boot=LABEL=System disk=LABEL=Sto                                                                                                                                                             rage  quiet vt.global_cursor_default=0 loglevel=2
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x0000000000099bff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000099c00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000e2000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x00000000dfe8ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000dfe90000-0x00000000dfe9dfff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000dfe9e000-0x00000000dfedffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000dfee0000-0x00000000dfefffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fff00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000019fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.6 present.
[    0.000000] DMI: Acer Aspire M3400/Aspire M3400, BIOS P01-A2 04/21/2010
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] AGP: No AGP bridge found
[    0.000000] e820: last_pfn = 0x1a0000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-EFFFF uncachable
[    0.000000]   F0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 000000000000 mask FFFF80000000 write-back
[    0.000000]   1 base 000080000000 mask FFFFC0000000 write-back
[    0.000000]   2 base 0000C0000000 mask FFFFE0000000 write-back
[    0.000000]   3 disabled
[    0.000000]   4 disabled
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] TOM2: 00000001a0000000 aka 6656M
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT
[    0.000000] e820: update [mem 0xe0000000-0xffffffff] usable ==> reserved
[    0.000000] e820: last_pfn = 0xdfe90 max_arch_pfn = 0x400000000
[    0.000000] found SMP MP-table at [mem 0x000ff780-0x000ff78f] mapped at [ffff                                                                                                                                                             8800000ff780]
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff880000093000] 93000 size 24576
[    0.000000] Using GB pages for direct mapping
[    0.000000] BRK [0x03cf3000, 0x03cf3fff] PGTABLE
[... 701 lines cut ...]
[    8.502548] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    9.471883] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[    9.592806] RTL871X: start auth
[    9.596232] RTL871X: auth success, start assoc
[    9.606070] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[    9.606074] RTL871X: assoc success
[    9.606167] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   10.006548] RTL871X: recv eapol packet
[   10.006788] RTL871X: send eapol packet
[   10.012967] RTL871X: recv eapol packet
[   10.013046] RTL871X: send eapol packet
[   10.013112] RTL871X: set pairwise key camid:4, addr:00:0c:c3:a1:14:72, kid:0,                                                                                                                                                              type:AES
[   10.015964] RTL871X: set group key camid:5, addr:00:0c:c3:a1:14:72, kid:1, ty                                                                                                                                                             pe:TKIP
[   40.081683] usb 7-2: new full-speed USB device number 2 using ohci-pci
[   40.244493] sony 0003:054C:09CC.0007: Using modified Dualshock 4 report descr                                                                                                                                                             iptor with gyroscope axes
[   40.289513] input: Sony Interactive Entertainment Wireless Controller Touchpa                                                                                                                                                             d as /devices/pci0000:00/0000:00:16.0/usb7/7-2/7-2:1.3/0003:054C:09CC.0007/input                                                                                                                                                             /input21
[   40.342773] input: Sony Interactive Entertainment Wireless Controller as /dev                                                                                                                                                             ices/pci0000:00/0000:00:16.0/usb7/7-2/7-2:1.3/0003:054C:09CC.0007/input/input20
[   40.342913] sony 0003:054C:09CC.0007: input,hidraw6: USB HID v81.11 Gamepad [                                                                                                                                                             Sony Interactive Entertainment Wireless Controller] on usb-0000:00:16.0-2/input3
[   40.438855] usbcore: registered new interface driver snd-usb-audio
[  135.417765] usb 7-2: USB disconnect, device number 2
[  144.191321] usb 7-2: new full-speed USB device number 3 using ohci-pci
[  144.440353] sony 0003:054C:09CC.0008: Using modified Dualshock 4 report descr                                                                                                                                                             iptor with gyroscope axes
[  144.485320] input: Sony Interactive Entertainment Wireless Controller Touchpa                                                                                                                                                             d as /devices/pci0000:00/0000:00:16.0/usb7/7-2/7-2:1.3/0003:054C:09CC.0008/input                                                                                                                                                             /input23
[  144.539371] input: Sony Interactive Entertainment Wireless Controller as /dev                                                                                                                                                             ices/pci0000:00/0000:00:16.0/usb7/7-2/7-2:1.3/0003:054C:09CC.0008/input/input22
[  144.539518] sony 0003:054C:09CC.0008: input,hidraw6: USB HID v81.11 Gamepad [                                                                                                                                                             Sony Interactive Entertainment Wireless Controller] on usb-0000:00:16.0-2/input3

gouchi, 2018-01-18:
Hi,

Can you try [this operation](https://forums.libretro.com/t/dualshock-4-crazy-fast-repeating-controls/14197/6) ?

Thank you.

GoodDay00, 2018-01-19:
That did it! 
Thank you so much!

closed 2018-01-19 by GoodDay00