libretro/Lakka-LibreELEC #338: lakka incompatible with GPD WIN device

rafa1987, 2018-01-19:
HELLO I HAVE PROBLEMS IN MY DEVICE, IN WHICH IT IS INVERTED SCREEN, WIFI DOES NOT WORK AND NOR THE SOUND TBM, PLEASE I WOULD LIKE THOSE PROBLEMS FIXED.

THE CONFIGURATIONS OF IT ARE CPU INTEL ATON X7 Z8700, GPU INTEL GRAPHICS HD CHERRY TRAIL, 4 GB RAM .... PLEASE WAIT AN ANSWER, THANK YOU!

Ntemis, 2018-01-23:
Sorry but that is a custom made console and it demands special image to work without bugs/issues, so don't hold your breath.

closed 2018-01-23 by Ntemis

marleyap, 2018-02-11:
it would not be too much work to carry a version 100% compatible with gpd win, and already has an android version 100% compatible with the device, I hope the developers have an exclusive ISO for it.

Ntemis, 2018-02-12:
No its not but no device to debug lakka on