libretro/Lakka-LibreELEC #342: add FreeIntv to Lakka cores

markwkidd, 2018-01-24:
FreeIntv is a new emulator, written from the ground up as a libretro core. https://github.com/markwkidd/FreeIntv

As of January 24, it is compiling and usable on many, albeit not all, platforms supported by the buildbot. Would you consider adding FreeIntv to Lakka?

Ntemis, 2018-01-24:
Sure will check it out today. Thanks

markwkidd, 2018-02-12:
The repository is now part of the libretro organization, so it has a new link: http://github.com/libretro/FreeIntv

Also if you have access to a RetroArch installation, FreeIntv is available from the buildbot. Windows and Linux builds are generally known to be working via buildbot

Ntemis, 2018-02-12:
Sorry that took so much time, i will prep a package soon thanks for the link

ToKe79, 2018-02-13:
core is merged.

closed 2018-02-13 by ToKe79