libretro/Lakka-LibreELEC #346: Support for 8 controllers

mstroh76, 2018-01-24:
PSX Emulator can support up to 8 controllers (https://en.wikipedia.org/wiki/PlayStation_Multitap).
Lakka input menu shows only 5 controllers even more are connected.
Max 5 controllers are working actual.

Im  using Lakka 2.1 Allwinner Orange Pi PC
Controllers in use:
-  XBox360 Wireless
- Xbox One USB
- 8Bitdo

gouchi, 2018-01-25:
> This issue tracker is NOT for help or support. Please use the Lakka forum (https://forums.libretro.com/c/libretro/lakka-tv-general) or IRC @ #lakka on freenode.net for those kinds of posts.

Anyway, did you enable in the Options entry of the quick menu the support for Multitap ?

mstroh76, 2018-01-25:
It has nothing to do with multitap. I mean that you can't configure more than 5 controllers in the input-menu, although the rom and the emulator supports two multitaps (so in total 8 players).

gouchi, 2018-01-25:
If it is a feature request please add it to [RetroArch issue tracker](https://github.com/libretro/RetroArch/issues).

But first you should try to enable Multitaps to see if you get access to 8 controllers.

mstroh76, 2018-01-25:
I have finally found the needed setting! You habe to activate 'show advanced settings'
(Settings -> User Interface) 
After that you can modify 'Max users' in input settings menu up to 16.

Maybe the default setting should be 8 instead of 5.  The Isse can be closed.

RobLoach, 2018-01-25:
> Maybe the default setting should be 8 instead of 5. 

Perhaps 4 as a default? ala N64?

ToKe79, 2018-01-25:
We are happy you solved it and appreciate your feedback.

closed 2018-01-25 by ToKe79