libretro/Lakka-LibreELEC #355: Lose video when scanning MAME files.

hmlessalky, 2018-01-29:
.:
- Which version of Lakka are you using?
Current, freshly updated

- What system hardware are you using?
PC, Pentium Dual, 4gb memory, GT320 1gb, Dell mb, 500gb HD.

- What did you do?
Scanned my MAME folder, did this before and it happened, wiped the folder and redownloaded.

- What did you expect to happen?
Scan and add my files. On previous attempt it did add about 10 files out of the many.

- What happened instead?
Shut down the video feed to the TV (HDMI). Computer remains on and am still able to connect to it, but there is no video on the main screen.

Sorry, it's the end of the day and I thought to throw this out there before I go to sleep, so no logs as of now. I will try to collect them tomorrow if I have a chance. Just seemed to be an odd occurance and I figured maybe it's an easy fix. Other than this issue the system has been working great. 

Thanks in advance for any assistance.

ToKe79, 2018-01-29:
Hi,
this seems like RetroArch segfaulted, but don't understand why the retroarch.service was not restarted.

I had similar issue when I had sub-folders with .chd files (which are needed by some MAME ROMs), so I worked around it by moving the sub-folders into a different folder, scanned the MAME ROM set without the sub-folders and after scanning finished moved the sub-folders back.

hmlessalky, 2018-01-29:
Thanks for the quick reply. I will look into this possibility next chance I get and report the outcome.

hmlessalky, 2018-03-03:
Sorry it took so long to revisit this issue, been a busy couple of months. You were exactly right, I discovered this when updating a retroarch setup on another computer. The culprit was a file named kinst, which had one file, kinst.chd in it. There was also a similar file kinst2 which I removed as well. Loaded up great, and whatever needed that file, if anything, hasn't needed it or I haven't come across it. Or maybe it was some odd copy-over from a ROM dump at some point. 

Anyway, this forum is the greatest and all us less experienced folks appreciate all the help it provides us.

Thanks ToKe79!

closed 2018-03-03 by hmlessalky